Dilbert66/esphome-vistaECP #2: Using as Zone Expander

residencekedda, 2020-10-05:
Hi,
  Can you share on how this can be used as Zone Expander.  I am planning to buy these WeMos D1 Mini and wire them as you shared or is there a ready to buy item that is available with your design?

D1 Mini link, https://www.amazon.com/IZOKEE-NodeMcu-Internet-Development-Compatible/dp/B076F52NQD/ref=sr_1_3?dchild=1&keywords=d1+wemos+mini&qid=1601913406&sr=8-3

Appreciate your guidance.

TCPMeta, 2020-11-29:
I honestly don't know how it would work as a Zone Expander. I think it listens for a Zone Expander to read Zone statues from but not be used as a Zone Expander. I typically buy Alarm Panel parts from [Alarm liquidators](https://www.alarmliquidators.com/). 
I have Gerber Files for a little [Adapter board](https://drive.google.com/file/d/1L4exDZjM4wccor7hrUE4dZjWjz5OQ7eI/view?usp=sharing) I made for my setup. You can have JLCPCB or PCBWay spin up some boards.

Dilbert66, 2020-11-29:
It emulates a zone expander for the panel.  A zone can be triggered by sending the  set_zone_fault service in home homeassistant
"set_zone_fault",Parameters: "zone","fault" where zone is a zone from 9 - 48 and fault is 0 or 1 (0=ok, 1=open) The zone number will depend on what your expander address is set to.
Or you can hardwire one of the unused esp inputs to be a trigger and have it send the set_zone_fault command internally. Example in the yaml.

closed 2021-01-21 by Dilbert66