Dilbert66/esphome-vistaECP #3: fails to compile

bertiebeez, 2020-11-06:
This is the message I get when I try to upload from esphome.

INFO Reading configuration /config/esphome/VistaAlarm.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/vistaalarm
Processing vistaalarm (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/vista.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
src/vista.cpp: In member function 'bool Vista::handle()':
src/vista.cpp:894:9: error: 'len' was not declared in this scope
         len=4;
         ^
*** [/data/vistaalarm/.pioenvs/vistaalarm/src/vista.cpp.o] Error 1
========================== [FAILED] Took 6.39 seconds ==========================

Dilbert66, 2020-11-06:
whups. I'll fix that. Forgot to push an update.

Dilbert66, 2020-11-06:
Ok, pushed an update. Vista.cpp corrected.

bertiebeez, 2020-11-07:
Thanks for the quick responce.  I am still getting a compile error about the missing define.

Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/ademcoECP.ino.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/main.cpp.o
src/main.cpp:166:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto VistaECP = new vistaECPHome(16);\nVistaECP->accessCode=\"0650\"\
   ^
/config/esphome/vistaalarm/src/ademcoECP.ino: In function 'void loop()':
/config/esphome/vistaalarm/src/ademcoECP.ino:358:50: error: 'statusText' was not declared in this scope
             String lrrString =String(statusText(c));
                                                  ^
*** [/data/vistaalarm/.pioenvs/vistaalarm/src/ademcoECP.ino.cpp.o] Error 1
========================== [FAILED] Took 9.24 seconds ==========================

Dilbert66, 2020-11-07:
remove the ino file. It is not for esphome. That is only a example for  standalone mqtt

bertiebeez, 2020-11-07:
I deleted the .ino file and then got this error upon uploading. This is with the esp8266 connected to the usb port.

Linking /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
Building /data/vistaalarm/.pioenvs/vistaalarm/firmware.bin
Retrieving maximum program size /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
Checking size /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
RAM:   [=====     ]  49.1% (used 40224 bytes from 81920 bytes)
Flash: [====      ]  38.7% (used 404312 bytes from 1044464 bytes)
Creating BIN file "/data/vistaalarm/.pioenvs/vistaalarm/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/vistaalarm/.pioenvs/vistaalarm/firmware.elf"
========================= [SUCCESS] Took 13.35 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of vistaalarm.local
ERROR Error resolving IP address of vistaalarm.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname

Dilbert66, 2020-11-07:
I suggest you read up on this page for the first upload.   

https://esphome.io/guides/getting_started_command_line.html

bertiebeez, 2020-11-07:
I was able to flash via nodemcu flasher and then it seems to work ok, I apologize, this is my first project trying to make all of this work together. I am definitely in the learning curve.  HA has created a entity card with the reported variables from the ESP8266 board. I am now waiting for a logic converter IC before I connect it to the 12v output of the panel.
If I had wanted to use mqtt, what would I have to do? And thank you for your patience with me.

Dilbert66, 2020-11-09:
copy the files vista.cpp,vista.h, ecpsoftwareserial.cpp ,ecpsoftwareserial.h as well as the ademcoecp.ino file in a directory called "ademcoECP"  and compile the ino file using Arduino after you setup your configs in the ino . Ie wifi, mqtt server, etc.

You also need to have an MQTT server running somewhere.  Mosquito for example

bertiebeez, 2020-11-11:
Thank you , I have it flashed onto the esp8266.,I am waiting on a logic level shifter to arrive so I can put it on a breadboard. I may have more questions.  Hope youdon't mind helping me.DavidSent from my Samsung Galaxy Tab®|PRO

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Dilbert66/esphome-vistaECP/issues/3#issuecomment-724251763",
"url": "https://github.com/Dilbert66/esphome-vistaECP/issues/3#issuecomment-724251763",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

bertiebeez, 2020-11-11:
Thank you,  I certainly will. Sent from my Samsung Galaxy Tab®|PRO

https://esphome.io/guides/getting_started_command_line.html

[
{

closed 2020-11-28 by Dilbert66