Dilbert66/esphome-vistaECP #4: Not so much a issue but a question on the accessCode

TCPMeta, 2020-11-27:
I built a breakout board and made the code reflect to my Panel's Zones and Sensors. I also set the ECP Address for it to 18 since I already have a 6160RF programmed as Address 16 and a 5160 programmed as 17.

Anyway, I opted out using the Secrets.yaml and typed in as plain text the accessCode and I ended up getting locked out for 15 minutes from the Panel. I noticed when I compiled the binary it said something about \"1234\"\. Now in the Yaml file it have it as accessCode: "1234" The question is do I put the code in single quotes, double quotes or no quotes in the Yaml file?

Dilbert66, 2020-11-28:
Use double quotes.

TCPMeta, 2020-11-28:
I was testing it yesterday and since the system is monitored from the Company I work for I am able to put the system in test anytime I want. Anyway I sent a Medical Alert signal and that's when I got locked out of the Keypad. After thinking it over I realized the A,B,C,D buttons are set to Zones so after setting those up in the Yaml file Z95 Fire Z96 Police and Z99 as Medical I did a test and works fine now.

closed 2020-11-28 by TCPMeta