Dilbert66/esphome-vistaECP #5: Log Message 0: Unknown Zone 0

TCPMeta, 2020-11-28:
I'm fighting this issue currently **Log Message 0: Unknown Zone 0** Don't know if it see's a zone open somewhere or it thinks a Zone Zero exists. I wonder if it's complaining about the Keyfob I have set as Zones 17, 18 and 19 but not declared in the Yaml.

Also quick question what is the lrrSupervisor?

Dilbert66, 2020-11-28:
Sorry I have no idea.  I've never seen that message.  Where is it showing?  Can you provide logs.

LRR is long range radio used for sending wireless messages to a monitoring station. If you don't have that, the system will emulate one. ie lrrsupervisor so that it can capture more detailed system messages and display them

Dilbert66, 2020-11-28:
Ok, I double checked. That's an lrr message from the system. It doesnt like something in your vista config.

TCPMeta, 2020-11-28:
Yep, I got a GSM. I should of known LRR stood for Long Range Radio. Just use to tossing the term GSM. I wonder if the GSM and lrrSupervisor is what is causing the Zone 0 issue. I'll tweak that and let you know.

TCPMeta, 2020-11-28:
I updated it and turned off the lrrSupervisor and no change for the Zone 0 thing. I guess I should toss a Laptop onto the ESP and let it output a log for a while and see if anything odd pops up. Dunno what in the Panel configuration could cause it. Looking at the Panel Reports from Central Station I only see from newest to oldest.

461 - when it locked me out the first go around
122 panic silent alert (test signal I sent)
339 expansion module reset
333 lost RF (when wiring up the ESP8266)
602 weekly test

Dilbert66, 2020-11-28:
You should be able to show some logs from the esphome console remotely without connecting to the esp. If debug is 1, it should show some packet activity.

Dilbert66, 2020-11-28:
I've added a fix to correct an issue with dealing with a zero lrr message value that might be due to how i use persistent storage to save data.  I've push a new version to master.  The only file that changed is vistaalarm.h
Try it out.

TCPMeta, 2020-11-28:
Yep, that fixed it. Thank you very much.

closed 2020-11-28 by TCPMeta