Dilbert66/esphome-vistaECP #6: Question not an issue

subseaalex, 2020-12-29:
Thank you for the code.
Could you please explain how to get sensor ids? I see in debug, I am reading all my sensors and correct names, however, they are all in hex. 
How do I add them to templates without knowing which z1, z2 etc?
For example, if I have "Living room motion sensor 35" installed, how do I know which id is this? Is it Z35?

Regards,

Dilbert66, 2020-12-29:
The id is the zone that the sensor belongs to.

subseaalex, 2020-12-29:
Thank you for quick reply. And how do I know which sensor belongs to which zone?

Dilbert66, 2020-12-29:
That's part of your panel programming. When you assign your sensors, you assign it to a zone.   You can also just trigger your sensor and see which is showing on your keypad display as open.

subseaalex, 2020-12-29:
We have converted all wired sensors to wireless with receiver SA5881-6NECR. Its wired to vista21iP.
It is not default installed code. I cant check the zones.
For example when I trigger door, the panel displays "Fault 29 Garage Entry" Does it means its zone29?

Dilbert66, 2020-12-29:
Yes, 29 would be the zone

subseaalex, 2020-12-29:
Thank you kindly. I undestand now. I have to add case id with the zone, and use id in template. 
It is working now. Thanks for all your help.

closed 2021-01-21 by Dilbert66