Dilbert66/esphome-vistaECP #7: Device spams open/close when zone is faulted

Codetraptor, 2021-01-18:
Thanks for providing this code. It works great for my needs. One issue I noticed was that if I have a door or window open home assistant flickers the status. The logs in ESPHome show the device is sending these statuses at a quick rate. Is this a known issue or something that could be wrong in my setup?

[image]

Dilbert66, 2021-01-18:
It might be a bug but I would need more logs to know for sure. I have not seen the issue myself so far.
Which branch are you using "dev" or "master". If you are using the master branch, try the "dev" one.

Codetraptor, 2021-01-18:
Thanks. Further review showed it was getting changed here:https://github.com/Dilbert66/esphome-vistaECP/blob/master/src/vistaEcpInterface/vistaAlarm.h#L635

Codetraptor, 2021-01-18:
Hit enter too early. Will try with dev branch

Codetraptor, 2021-01-18:
Uploaded code from dev branch with the same result.

Dilbert66, 2021-01-18:
Also make sure this setting is set correctly: 
 TTL: "30000"  # time  to live  in ms for zone/fire status before expiring;

Dilbert66, 2021-01-19:
That is a timer that will cause the code to wait before it flags a sensor as reset. If it's too short, as soon as it's set, the timer will cound down and reset it again.  The idea is to have that timer longer than the next fault loop which is on average 30000 milliseconds

Codetraptor, 2021-01-19:
It is set. I also had tried raising it to 60000 to see if there was a difference. My current YAML is below:
```
#for documentation see project at https://github.com/Dilbert66/esphome-VistaECP
substitutions:
  systemName: "vistaalarm"
  accessCode: !secret access_code #Only comes into effect if needed for arming and quickarm is not set
  keypadAddr: "16"
  
  # module addresses:
  # 07  4220 zone expander  zones 9-16
  # 08 4229 zone expander zones 17-24
  # 09 4229 zone expander zones 25-32
  # 10 4229 zone expander zones 33-40
  # 11 4229 zone expander zones 41 48
  # 12 4204 relay module  
  # 13 4204 relay module
  # 14 4204 relay module
  # 15 4204 relay module
  expanderAddr1: "0" # 1st zone expander emulator (4229) address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  relayAddr1: "0" # relay module emulation (4204) addresses. Set to 0 to disable
  relayAddr2: "0"
  relayAddr3: "0"
  relayAddr4: "0"
  
  TTL: "30000"  # time  to live  in ms for zone/fire status before expiring;
  quickArm: "true"
  lrrSupervisor: "true" # set to true if we don't have an LRR monitoring supervisor we can emulate one to get the statuses
  
globals:
       #persistent storage variables
    - id: zoneStates  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneAlarms  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneBypass  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneChecks  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes      
    - id: lrrCode  #persistant storage for last lrr message in case of reboot
      type: int
      restore_value: yes      

esphome:
  name: $systemName
  platform: ESP8266
  board: nodemcuv2

  # subdirectory where library *.h and *.cpp are placed 
  includes:
    - vistaEcpInterface/

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "$systemName"
    password: !secret wifi_password

logger:
  baud_rate: 115200
  level: DEBUG
  
api:
[... 208 lines cut ...]
    #system status 
  - platform: template
    id: system_status
    name: "$systemName System Status"
    icon: "mdi:shield"

  - platform: template
    id: m1
    name: "$systemName Lrr Msg"
    icon: "mdi:alert-box"

switch:
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return vista.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectVista();
  - platform: restart
    id: restart_switch```

Dilbert66, 2021-01-19:
Very odd. Is it always the same zone that acts like that?

Dilbert66, 2021-01-19:
If possible can you show a longer sequence of logs as well.

Codetraptor, 2021-01-19:
I have the same results with any door or window zone that I've tried.
Here is a longer log where I opened the door for a few minutes and then closed it.
```
INFO Reading configuration /config/esphome/VistaAlarm.yaml...
INFO Starting log output from vistaalarm.local using esphome API
INFO Connecting to vistaalarm.local:6053 (192.168.2.4)
INFO Successfully connected to vistaalarm.local
[20:59:25][I][app:105]: ESPHome version 1.15.3 compiled on Jan 18 2021, 19:03:10
[20:59:25][C][status_led:019]: Status LED:
[20:59:25][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[20:59:25][C][wifi:443]: WiFi:
[20:59:25][C][wifi:303]:   SSID: [redacted]
[20:59:25][C][wifi:304]:   IP Address: 192.168.2.4
[20:59:25][C][wifi:306]:   BSSID: [redacted]
[20:59:25][C][wifi:307]:   Hostname: 'vistaalarm'
[20:59:25][C][wifi:311]:   Signal strength: -76 dB ▂▄▆█
[20:59:25][C][wifi:315]:   Channel: 11
[20:59:25][C][wifi:316]:   Subnet: 255.255.255.0
[20:59:25][C][wifi:317]:   Gateway: 192.168.2.1
[20:59:25][C][wifi:318]:   DNS1: 192.168.2.1
[20:59:25][C][wifi:319]:   DNS2: (IP unset)
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Trouble'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Bypass'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Away'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Stay'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Instant'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Night'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm AC'
[20:59:25][C][template.binary_sensor:018]:   Device Class: 'plug'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Chime'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Check'
[20:59:25][C][template.binary_sensor:018]:   Device Class: 'problem'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Alarm'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Battery'
[20:59:25][C][template.binary_sensor:018]:   Device Class: 'problem'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Fire'
[20:59:25][C][template.binary_sensor:018]:   Device Class: 'smoke'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Ready'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'r1'
[20:59:25][C][template.binary_sensor:018]: Template Binary Sensor 'r2'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Entry Door'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Bedroom'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Office'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Living Room'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Den Windows'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Kitchen Motion'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Kitchen Door'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Dining Room'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Kitchen Window'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Nursery Smoke'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Nursery C02'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Master Smoke'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Master C02'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Den window far right'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm System Status'
[20:59:25][C][template.text_sensor:020]:   Icon: 'mdi:shield'
[20:59:25][C][template.text_sensor:020]: Template Sensor 'vistaalarm Lrr Msg'
[20:59:25][C][template.text_sensor:020]:   Icon: 'mdi:alert-box'
[20:59:25][C][template.switch:058]: Template Switch 'vistaalarm Connection'
[20:59:25][C][template.switch:058]:   Icon: 'mdi:shield-link-variant'
[20:59:25][C][template.switch:059]:   Restore State: NO
[20:59:25][C][template.switch:060]:   Optimistic: NO
[20:59:25][I][CMD:333]: 9E 02 25 82 5C 00 00 00 00 00 00 00 
[20:59:25][C][logger:185]: Logger:
[20:59:25][C][logger:186]:   Level: DEBUG
[20:59:25][C][logger:187]:   Log Baud Rate: 115200
[20:59:25][C][logger:188]:   Hardware UART: UART0
[20:59:25][C][restart:021]: Restart Switch 'restart_switch'
[20:59:25][C][restart:021]:   Icon: 'mdi:restart'
[20:59:25][C][ota:029]: Over-The-Air Updates:
[... 510 lines cut ...]
[21:02:41][I][CMD:333]: F2 12 06 00 00 00 00 63 6C 02 45 6C 
[21:02:41][I][INFO:521]: Prompt: ****DISARMED****
[21:02:41][I][INFO:522]: Prompt:   Ready to Arm  
[21:02:41][I][INFO:523]: Beeps: 0

[21:02:41][I][CMD:333]: F7 00 00 07 10 08 00 1C 08 02 00 00 
[21:02:41][I][INFO:521]: Prompt: ****DISARMED****
[21:02:41][I][INFO:522]: Prompt:   Ready to Arm  
[21:02:41][I][INFO:523]: Beeps: 0

[21:02:41][I][CMD:333]: F2 16 06 00 00 00 00 64 63 02 45 43 
[21:02:41][I][INFO:521]: Prompt: ****DISARMED****
[21:02:41][I][INFO:522]: Prompt:   Ready to Arm  
[21:02:41][I][INFO:523]: Beeps: 0

[21:02:48][I][CMD:333]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 
[21:02:48][I][EXT:333]: F6 10 00 00 54 87 66 48 80 F7 00 00 
[21:02:50][I][CMD:333]: F7 00 00 07 10 08 00 1C 08 02 00 00 
[21:02:51][I][INFO:521]: Prompt: ****DISARMED****
[21:02:51][I][INFO:522]: Prompt:   Ready to Arm  
[21:02:51][I][INFO:523]: Beeps: 0

[21:02:51][I][CMD:333]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 
[21:02:51][I][EXT:333]: F6 10 00 00 51 87 66 48 00 7A 00 00 
```

Dilbert66, 2021-01-19:
Can you add this line at line 452 in vistaAlarm.h then run it and paste the new log. I want to see what the ready flag is set to. I suspect I know the issue. 
```
 ESP_LOGD("DEBUG","Ready flag: %d",vista.statusFlags.ready); 
```

It should then look like this:
```
    if (debug > 0 && vista.cbuf[0] && vista.newCmd) {  
            printPacket("CMD",vista.cbuf,12);
            ESP_LOGD("DEBUG","Ready flag: %d",vista.statusFlags.ready); // added line above.
            vista.newCmd=false;
       }
```

Codetraptor, 2021-01-19:
I've added the line and here is the new log starting at when the firmware was uploaded. Looks like the ready flag never changed.
```
INFO Reading configuration /config/esphome/VistaAlarm.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/vistaalarm
Processing vistaalarm (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/main.cpp.o
src/main.cpp:170:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto VistaECP = new vistaECPHome(16);\nVistaECP->accessCode=\"\"\
   ^
Linking /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
Building /data/vistaalarm/.pioenvs/vistaalarm/firmware.bin
Creating BIN file "/data/vistaalarm/.pioenvs/vistaalarm/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/vistaalarm/.pioenvs/vistaalarm/firmware.elf"
Retrieving maximum program size /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
Checking size /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
RAM:   [=====     ]  49.2% (used 40320 bytes from 81920 bytes)
Flash: [====      ]  38.8% (used 405172 bytes from 1044464 bytes)
========================= [SUCCESS] Took 14.25 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of vistaalarm.local
INFO  -> 192.168.2.4
INFO Uploading /data/vistaalarm/.pioenvs/vistaalarm/firmware.bin (409328 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from vistaalarm.local using esphome API
INFO Connecting to vistaalarm.local:6053 (192.168.2.4)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.2.4: timed out). Re-Trying in 1 seconds
INFO Connecting to vistaalarm.local:6053 (192.168.2.4)
INFO Successfully connected to vistaalarm.local
[22:33:45][I][app:105]: ESPHome version 1.15.3 compiled on Jan 18 2021, 22:33:10
[22:33:45][C][status_led:019]: Status LED:
[22:33:45][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[22:33:45][C][wifi:443]: WiFi:
[22:33:45][C][wifi:303]:   SSID: [redacted]
[22:33:45][C][wifi:304]:   IP Address: 192.168.2.4
[22:33:45][C][wifi:306]:   BSSID: [redacted]
[22:33:45][C][wifi:307]:   Hostname: 'vistaalarm'
[22:33:45][C][wifi:311]:   Signal strength: -80 dB ▂▄▆█
[22:33:45][C][wifi:315]:   Channel: 11
[22:33:45][C][wifi:316]:   Subnet: 255.255.255.0
[22:33:45][C][wifi:317]:   Gateway: 192.168.2.1
[22:33:45][C][wifi:318]:   DNS1: 192.168.2.1
[22:33:45][C][wifi:319]:   DNS2: (IP unset)
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Trouble'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Bypass'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Away'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Stay'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Instant'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Night'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm AC'
[22:33:45][C][template.binary_sensor:018]:   Device Class: 'plug'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Chime'
[22:33:45][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Check'
[... 311 lines cut ...]
[22:35:22][I][INFO:522]: Prompt: ****DISARMED****
[22:35:22][I][INFO:523]: Prompt:   Ready to Arm  
[22:35:22][I][INFO:524]: Beeps: 0

[22:35:24][I][CMD:333]: 9E 02 25 81 5D 00 00 00 00 00 00 00 
[22:35:24][D][DEBUG:452]: Ready flag: 1
[22:35:24][I][EXT:333]: F9 00 00 00 21 00 DF 05 84 25 05 C9 
[22:35:25][I][CMD:333]: 9E 02 20 82 61 00 00 00 00 00 00 00 
[22:35:25][D][DEBUG:452]: Ready flag: 1
[22:35:25][I][EXT:333]: F9 00 00 00 24 82 5A 05 84 25 05 C9 
[22:35:26][I][CMD:333]: F6 10 00 00 00 00 00 00 00 00 00 00 
[22:35:26][D][DEBUG:452]: Ready flag: 1
[22:35:26][I][EXT:333]: F6 10 00 50 02 0A A4 00 00 00 00 00 
[22:35:26][I][CMD:333]: F2 12 06 00 00 00 00 64 6C 02 45 6C 
[22:35:26][D][DEBUG:452]: Ready flag: 1
[22:35:26][I][INFO:522]: Prompt: ****DISARMED****
[22:35:26][I][INFO:523]: Prompt:   Ready to Arm  
[22:35:26][I][INFO:524]: Beeps: 0

[22:35:26][I][CMD:333]: F7 00 00 07 10 08 00 1C 08 02 00 00 
[22:35:26][D][DEBUG:452]: Ready flag: 1
[22:35:26][I][INFO:522]: Prompt: ****DISARMED****
[22:35:26][I][INFO:523]: Prompt:   Ready to Arm  
[22:35:26][I][INFO:524]: Beeps: 0
```

Dilbert66, 2021-01-19:
Yes, that's the issue I believe. The flag is not being reset. I've push a new update to the DEV branch that should correct the issue.   The only file that changed is vista.cpp.  Try it out. If it works, i'll push it to master.

Codetraptor, 2021-01-19:
It looks like that was the issue. Here is the log from pushing this change. It's changing the ready status as expected and no longer flipping the open/closed value. I only swapped the changed file so the debug line was still in there.
```
INFO Reading configuration /config/esphome/VistaAlarm.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/vistaalarm
Processing vistaalarm (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/vista.cpp.o
Linking /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
Building /data/vistaalarm/.pioenvs/vistaalarm/firmware.bin
Creating BIN file "/data/vistaalarm/.pioenvs/vistaalarm/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/vistaalarm/.pioenvs/vistaalarm/firmware.elf"
Retrieving maximum program size /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
Checking size /data/vistaalarm/.pioenvs/vistaalarm/firmware.elf
RAM:   [=====     ]  49.2% (used 40320 bytes from 81920 bytes)
Flash: [====      ]  38.8% (used 405140 bytes from 1044464 bytes)
========================= [SUCCESS] Took 12.42 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of vistaalarm.local
INFO  -> 192.168.2.4
INFO Uploading /data/vistaalarm/.pioenvs/vistaalarm/firmware.bin (409296 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from vistaalarm.local using esphome API
INFO Connecting to vistaalarm.local:6053 (192.168.2.4)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.2.4: timed out). Re-Trying in 1 seconds
INFO Connecting to vistaalarm.local:6053 (192.168.2.4)
INFO Successfully connected to vistaalarm.local
[08:35:08][I][app:105]: ESPHome version 1.15.3 compiled on Jan 18 2021, 22:33:10
[08:35:08][C][status_led:019]: Status LED:
[08:35:08][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[08:35:08][C][wifi:443]: WiFi:
[08:35:08][C][wifi:303]:   SSID: [redacted]
[08:35:08][C][wifi:304]:   IP Address: 192.168.2.4
[08:35:08][C][wifi:306]:   BSSID: [redacted]
[08:35:08][C][wifi:307]:   Hostname: 'vistaalarm'
[08:35:08][C][wifi:311]:   Signal strength: -81 dB ▂▄▆█
[08:35:08][C][wifi:315]:   Channel: 11
[08:35:08][C][wifi:316]:   Subnet: 255.255.255.0
[08:35:08][C][wifi:317]:   Gateway: 192.168.2.1
[08:35:08][C][wifi:318]:   DNS1: 192.168.2.1
[08:35:08][C][wifi:319]:   DNS2: (IP unset)
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Trouble'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Bypass'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Away'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Stay'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Instant'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Night'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm AC'
[08:35:08][C][template.binary_sensor:018]:   Device Class: 'plug'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Chime'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Check'
[08:35:08][C][template.binary_sensor:018]:   Device Class: 'problem'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Alarm'
[08:35:08][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Battery'
[... 175 lines cut ...]
[08:35:51][D][binary_sensor:036]: 'vistaalarm Ready': Sending state ON
[08:35:51][I][CMD:333]: F2 16 06 00 00 00 00 63 63 02 45 43 
[08:35:51][D][DEBUG:452]: Ready flag: 1
[08:35:51][I][INFO:522]: Prompt: ****DISARMED****
[08:35:51][I][INFO:523]: Prompt:   Ready to Arm  
[08:35:51][I][INFO:524]: Beeps: 0

[08:35:54][I][CMD:333]: F7 00 00 07 10 08 00 1C 08 02 00 00 
[08:35:54][D][DEBUG:452]: Ready flag: 1
[08:35:54][I][INFO:522]: Prompt: ****DISARMED****
[08:35:54][I][INFO:523]: Prompt:   Ready to Arm  
[08:35:54][I][INFO:524]: Beeps: 0

[08:35:57][I][CMD:333]: 9E 02 20 81 62 00 00 00 00 00 00 00 
[08:35:57][D][DEBUG:452]: Ready flag: 1
[08:35:57][I][EXT:333]: 9E 00 20 00 24 00 DC 00 00 00 00 00 
[08:35:58][I][CMD:333]: 9E 02 25 82 5C 00 00 00 00 00 00 00 
[08:35:58][D][DEBUG:452]: Ready flag: 1
[08:35:58][I][EXT:333]: 9E 00 25 00 21 82 5D 00 00 00 00 00 
[08:36:04][I][CMD:333]: F7 00 00 07 10 08 00 1C 08 02 00 00 
[08:36:04][D][DEBUG:452]: Ready flag: 1
[08:36:04][I][INFO:522]: Prompt: ****DISARMED****
[08:36:04][I][INFO:523]: Prompt:   Ready to Arm  
[08:36:04][I][INFO:524]: Beeps: 0
```

Dilbert66, 2021-01-19:
Excellent. Thanks for the  logs.

closed 2021-01-19 by Dilbert66

Dilbert66, 2021-01-19:
FYI, you can try lowering the TTL value to a smaller number such as 10000 or 15000 (10  or 15 seconds) to make it more responsive for detecting fault resets since you don't have many events occuring.  If you have a lot of fault messages, that increases the loop time between fault messages for each zone so you need a larger ttl.