Dilbert66/esphome-vistaECP #8: ESP32 fails to compile

jcuster, 2021-02-06:
Hi,

I'm trying to compile for an ESP32 and it fails. Here are my changes:

````
 esphome:
   name: $systemName
-  platform: ESP8266
-  board: nodemcuv2
+  platform: ESP32
+  board: esp32dev

 status_led:
   pin:
-    number: D4
+    number: 16
     inverted: yes
````

I'd also appreciate knowing if there are any other changes I need to make to get this to work with an ESP32. I've connected SDA and SCL to pins 21 and 22, respectively. And I'm not using the green monitor (not sure if I will yet or not).

Thanks for your help!

Here is the compilation output:
````
INFO Reading configuration VistaAlarm.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d vistaalarm
Processing vistaalarm (board: esp32dev; framework: arduino; platform: espressif32@1.12.4)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Looking for AsyncTCP-esphome library in registry
LibraryManager: Installing id=6798 @ 1.1.1
AsyncTCP-esphome @ 1.1.1 has been successfully installed!
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/vistaalarm/src/ECPSoftwareSerial.cpp.o
Compiling .pioenvs/vistaalarm/src/debug/vistaPrint.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/api/util.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/restart/restart_switch.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/status_led/status_led.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/switch/automation.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/switch/switch.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/template/binary_sensor/template_binary_sensor.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/template/switch/template_switch.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/template/text_sensor/template_text_sensor.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/application.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/component.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/controller.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/esphal.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/log.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/preferences.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/vistaalarm/src/esphome/core/util.cpp.o
Compiling .pioenvs/vistaalarm/src/main.cpp.o
In file included from src/main.cpp:53:0:
src/vistaAlarm.h:18:0: warning: "D1" redefined
 #define D1 18
 ^
src/vistaAlarm.h:13:0: note: this is the location of the previous definition
 #define D1 (5)
 ^
src/vistaAlarm.h:19:0: warning: "D2" redefined
 #define D2 19
 ^
src/vistaAlarm.h:14:0: note: this is the location of the previous definition
 #define D2 (4)
 ^
src/vistaAlarm.h:20:0: warning: "D5" redefined
 #define D5 21
 ^
src/vistaAlarm.h:8:0: note: this is the location of the previous definition
 #define D5 (14)
 ^
src/main.cpp:160:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto VistaECP = new vistaECPHome(16);\nVistaECP->accessCode=\"\"\
   ^
Compiling .pioenvs/vistaalarm/src/vista.cpp.o
Generating partitions .pioenvs/vistaalarm/partitions.bin
Compiling .pioenvs/vistaalarm/libe16/AsyncTCP-esphome/AsyncTCP.cpp.o
src/main.cpp:23:34: error: 'esphome::template_::TemplateBinarySensor* alarm' redeclared as different kind of symbol
 template_::TemplateBinarySensor *alarm;
                                  ^
In file included from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/wchar.h:4:0,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/cwchar:44,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/postypes.h:40,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/char_traits.h:40,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/string:40,
                 from src/esphome/core/component.h:3,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:2,
                 from src/main.cpp:3:
/root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/sys/unistd.h:21:11: note: previous declaration 'unsigned int alarm(unsigned int)'
 unsigned  _EXFUN(alarm, (unsigned __secs ));
           ^
src/main.cpp: In function 'void setup()':
src/main.cpp:248:9: error: assignment of function 'unsigned int alarm(unsigned int)'
   alarm = new template_::TemplateBinarySensor();
         ^
src/main.cpp:248:9: error: cannot convert 'esphome::template_::TemplateBinarySensor*' to 'unsigned int(unsigned int)' in assignment
src/main.cpp:426:35: error: no matching function for call to 'esphome::Application::register_binary_sensor(unsigned int (&)(unsigned int))'
   App.register_binary_sensor(alarm);
                                   ^
In file included from src/esphome/components/api/api_connection.h:4:0,
                 from src/esphome.h:2,
                 from src/main.cpp:3:
src/esphome/core/application.h:47:8: note: candidate: void esphome::Application::register_binary_sensor(esphome::binary_sensor::BinarySensor*)
   void register_binary_sensor(binary_sensor::BinarySensor *binary_sensor) {
        ^
src/esphome/core/application.h:47:8: note:   no known conversion for argument 1 from 'unsigned int(unsigned int)' to 'esphome::binary_sensor::BinarySensor*'
src/main.cpp:427:10: error: request for member 'set_name' in 'alarm', which is of non-class type 'unsigned int(unsigned int)'
   alarm->set_name("vistaalarm Alarm");
          ^
src/main.cpp: In lambda function:
src/main.cpp:507:31: error: request for member 'publish_state' in 'alarm', which is of non-class type 'unsigned int(unsigned int)'
           case salarm: alarm->publish_state(open);break;
                               ^
Compiling .pioenvs/vistaalarm/lib789/WiFi/ETH.cpp.o
In file included from src/esphome/components/api/api_connection.h:4:0,
                 from src/esphome.h:2,
                 from src/main.cpp:3:
src/esphome/core/application.h: In instantiation of 'C* esphome::Application::register_component(C*) [with C = unsigned int(unsigned int)]':
src/main.cpp:249:31:   required from here
src/esphome/core/application.h:82:5: error: static assertion failed: Only Component subclasses can be registered
     static_assert(std::is_base_of<Component, C>::value, "Only Component subclasses can be registered");
     ^
Compiling .pioenvs/vistaalarm/lib789/WiFi/WiFi.cpp.o
*** [.pioenvs/vistaalarm/src/main.cpp.o] Error 1
=============================================================== [FAILED] Took 25.36 seconds ===============================================================
````

jcuster, 2021-02-07:
I was able to get things to compile successfully, but I'm not seeing states accurately reflect what the alarm system is doing (nothing changes when I open/close a door to change to ready -> not ready -> ready). Here are my changes:

````
@@ -103,7 +102,7 @@ custom_component:
     VistaECP->onStatusChange([&](sysState led,bool open) {
       switch(led) {
         case sfire: id(fire).publish_state(open);break;
-        case salarm: id(alarm).publish_state(open);break;
+        case salarm: id(myalarm).publish_state(open);break;
         case strouble: id(trouble).publish_state(open);break;
         case sarmedstay: id(stay).publish_state(open);break;
         case sarmedaway: id(away).publish_state(open);break;
@@ -197,7 +196,7 @@ binary_sensor:
     device_class: problem

   - platform: template
-    id: alarm
+    id: myalarm
     name: "$systemName Alarm"
````

and I changed `src/vistaEcpInterface/vistaAlarm.h` to reflect my pins:
````
 //esp32 use pins 4,13,16-39
 #ifdef ESP32
-#define D1 18
-#define D2 19
-#define D5 21
+#define D1 22
+#define D2 21
+#define D5 -1
 #endif
````

The only wiring differences I have compared to your wiring diagram is that I'm using 22k and 10k Ohm resistors in series to create an effective 32k Ohm resistor (instead of 33k), and instead of a 330 Ohm, I'm using a 470 Ohm resistor. I'm using a LM2596 buck converter to step down to 5V, and I'm also using the 4n26 optoisolator.

Dilbert66, 2021-02-07:
unfortunately, I have not tested the esp32 hardware interface which is why you also got those compile errors since I've done a lot of changes since I tried an esp32 compile.  It's really a work in progress.   Are you getting data input from the debug prints? 
You can post some sample outputs so we can see what messages occur during the transitions.
Please note that due to the lack of info from the panel , there is no accurate way of determining when a zone close/reset event happens as the panel does not send messages for those. Only for opens or alarms basically.   This is why a  time-to-live (ttl) setting is used to reset a zone when the panel stops sending open event messages for that it .  You can adjust the ttl to a smaller value to make the transition happen faster.

Dilbert66, 2021-02-07:
As far as your hardware choices for components, those are fine.   The values are not critical.

jcuster, 2021-02-07:
Thanks for getting back to me and thanks for your help!

A bit more information on what I've done since I left it out of the above. On my main keypad, whose address is 17, I searched for an unused keypad address (19) and activated it. I set `keypadAddr` in the yaml and `KP_ADDR` in `vistaAlarm.h` to 19. The documentation doesn't talk about this, but I thought maybe this is what I am supposed to do instead of setting keypadAddr to the actual physical keypad address?

I commented out the define for `MONITORTX` in `vista.h` and enabled `DEBUG`.

I understand there isn't a way to see when a zone is closed (hence the ttl). I can't even get the status for Ready to report properly (it's always Not Ready, even when the system is Ready), and AC is always reported as disconnected, etc. That is, it seems like nothing is being reported.

Here's a sample log over the span of a couple minutes:
````
11:13:19.627 -> Packet: FE
11:13:39.484 ->
11:13:39.484 -> Packet: E
11:13:49.372 ->
11:13:49.372 -> Packet: FC
11:13:49.372 ->
11:13:49.372 -> Packet: FC
11:13:50.583 -> [0;32m[I][CMD:334]: F8 00 00 00 00 00 00 00 00 00 00 00 [0m
11:13:50.583 -> [0;37m[V][app:081]: A component took a long time in a loop() cycle (1.15 s).[0m
11:13:50.583 -> [0;37m[V][app:082]: Components should block for at most 20-30ms in loop().[0m
11:13:50.966 ->
11:13:50.966 -> Packet: C0
11:14:09.191 ->
11:14:09.191 -> Packet: 1E
11:14:09.229 ->
11:14:09.229 -> Packet: 78
11:14:09.229 ->
11:14:09.229 -> Packet: F0
11:14:43.700 -> [0;32m[I][CMD:334]: F8 00 00 00 00 00 00 00 00 00 00 00 [0m
11:14:43.700 -> [0;37m[V][app:081]: A component took a long time in a loop() cycle (4.82 s).[0m
11:14:43.700 -> [0;37m[V][app:082]: Components should block for at most 20-30ms in loop().[0m
11:15:18.649 ->
11:15:18.649 -> Packet: FC
11:15:18.649 ->
11:15:18.649 -> Packet: 80
11:15:18.687 ->
11:15:18.687 -> Packet: E0
11:15:18.687 ->
11:15:18.687 -> Packet: FC
11:15:18.687 ->
11:15:18.687 -> Packet: FC
````

Dilbert66, 2021-02-07:
Setting keypaddaddr in the yaml is the only needed one. It overrides the vistaalarm default setting.  You did it right. You just need to set it to an unused address.   The "" a component took a long time in a loop" is obviously a warning from the esp32 library. I have not played enough with the esp32 variant to be familiar with it.  The interrupts are time consuming for sure. I've tried to make them as efficient as possible considering what needs to happen.  I will see what else can be optimized.  This still doesnt explain the issue you are having. I will need to whip up a circuit to debug further. 
The cmd data you are showing does not make sense.  Should not be zero's. Looks like the serial interrupt routine is not capturing data correctly.

jcuster, 2021-02-07:
Sorry, I think that last log was just noise. In my haste to convert from powering the esp32 from the panel to using USB, I had wired something wrong. I have everything wired properly now and I don't see any debug prints.

jcuster, 2021-02-07:
Here is an image of my circuit (setup for USB) if it's any help.

[image]

Dilbert66, 2021-02-07:
Well, I just programmed an esp32 and wired it up on a protoboard and it works fine.   I used gpio 18 for d1 and gpio19 for d2 because that's my defaults were. .   The logs should show the same traffic as what is shown on your existing keypad.  You can also try using keypad address 16 which is what I use currently.

jcuster, 2021-02-07:
Thanks. On a whim I decided to try moving the wires to different pins and I started getting data. However, now my physical keypad is reporting Check 100. Any idea what could be causing that?

Dilbert66, 2021-02-07:
it's saying that it can't find the wireless module.   Are you using the virtual zone expander? It might be conflicting with your wireless , so you might want to change your zone expander address if you are using it.  You try clearing the error by entering your mastercode + 1 twice.

jcuster, 2021-02-07:
I'll try disabling the virtual zone expander and see what happens.

jcuster, 2021-02-08:
Thanks!! I disabled all of these just to test and everything is working now:

````
  expanderAddr1: "0" # 1st zone expander emulator (4229) address to use . Set to 0 to disable.
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable.

  relayAddr1: "0" # relay module emulation (4204) addresses. Set to 0 to disable
  relayAddr2: "0"
  relayAddr3: "0"
  relayAddr4: "0"
````

Woohoo! I can start playing around with a config and once I get everything setup, I can possibly add some notes to the documentation.

Dilbert66, 2021-02-08:
Excellent.  I'll polish up the code to fix the esp32 compilation issues and push an update.  
Yes, the documentation could use some work.

closed 2021-03-01 by Dilbert66