Dilbert66/esphome-vistaECP #9: Get status zone 1 to zone 6

morichina123, 2021-02-08:
Hi!
Many thanks for your excellent work !!
Im a problem. System status, TROUBLE, and READY works fine, but I cant read status  for zones. I dont have zone expander, only 16 zones of my vista48LA.
All zones status always are: CLOSED.

Any help ?? 
Many thanks !!

Dilbert66, 2021-02-08:
Unfortunately, the mqtt example is quite old and has not been updated to properly work with the latest library code so I can't say at this point what the issue would be. I will have to program up a chip with that mqtt code to see what needs updating.   The main version is made to support esphome as that is my platform of choice.  The mqtt example was added mostly as an example on using the library for other uses and as such is a work in progress.

morichina123, 2021-02-08:
Thank you very much for answering.
I would appreciate your research on the subject. I understand that the values ​​are obtained but are not published through MQTT.
I will wait impatiently for a solution. Thank you very much !!

Dilbert66, 2021-02-08:
I would suggest that you also post a printout of the debug serial output log from your esp that shows the cmd traffic during an open and close event. That is  extremely helpful for debugging issues.

morichina123, 2021-02-08:
the problem is that I don't have access to the panel with a computer, I can't do it. But it is a good idea. I'll try a laptop.
Thank you very much!

morichina123, 2021-02-11:
Hello Dibert66. This is serial debug output:

CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 06 00 00 08 02 00 00 
Prompt1:FALLO 06        
Prompt2:Distr.1a Planta 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 07 00 00 08 02 00 00 
Prompt1:FALLO 07        
Prompt2:Dorm.Principal  
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 06 00 00 08 02 00 00 
Prompt1:FALLO 06        
Prompt2:Distr.1a Planta 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F7 00 00 01 10 06 00 00 08 02 00 00 
Prompt1:FALLO 06        
Prompt2:Distr.1a Planta 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F7 00 00 01 10 06 00 00 08 02 00 00 
Prompt1:FALLO 06        
Prompt2:Distr.1a Planta 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 01 00 00 08 02 00 00 
Prompt1:FALLO 01        
Prompt2:Entrada         
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: 9E 02 25 81 5D 00 00 00 00 00 00 00 
CMD: F7 00 00 01 10 02 00 00 08 02 00 00 
Prompt1:FALLO 02        
Prompt2:Salon           
CMD: 9E 02 20 82 61 00 00 00 00 00 00 00 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 02 00 00 08 02 00 00 
Prompt1:FALLO 02        
Prompt2:Salon           
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 01 00 00 08 02 00 00 
Prompt1:FALLO 01        
Prompt2:Entrada         
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 03 00 00 08 02 00 00 
Prompt1:FALLO 03        
Prompt2:Pasillo         
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 05 00 00 08 02 00 00 
Prompt1:FALLO 05        
Prompt2:Dorm.Pl.Baja    
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 03 00 00 08 02 00 00 
Prompt1:FALLO 03        
Prompt2:Pasillo         
CMD: F7 00 00 01 10 05 00 00 08 02 00 00 
Prompt1:FALLO 05        
Prompt2:Dorm.Pl.Baja    
CMD: F7 00 00 01 10 05 00 00 08 02 00 00 
Prompt1:FALLO 05        
Prompt2:Dorm.Pl.Baja    
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 03 00 00 08 02 00 00 
Prompt1:FALLO 03        
Prompt2:Pasillo         
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F7 00 00 01 10 01 00 00 08 02 00 00 
Prompt1:FALLO 01        
Prompt2:Entrada         
CMD: F7 00 00 01 10 06 00 00 08 02 00 00 
Prompt1:FALLO 06        
Prompt2:Distr.1a Planta 
CMD: F7 00 00 01 10 06 00 00 08 02 00 00 
Prompt1:FALLO 06        
Prompt2:Distr.1a Planta 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 
CMD: F7 00 00 01 10 08 00 1C 08 02 00 00 
Prompt1:***DESARMADO*** 
Prompt2:Listo Para Armar

Dilbert66, 2021-02-11:
Ok, I see the issue. You have to make a change in the vistaalarm.h file to adjust to your local panel language.  The default currently is English.  The system uses the system prompts to identify the message type.  Change the values here to match the prompts on your panel:
```
top of file vistaAlarm.h file:

  const char* const FAULT="FAULT"; //change these to suit your panel language 
  const char* const BYPAS="BYPAS";
  const char* const ALARM="ALARM";
  const char* const FIRE="FIRE";
  const char* const CHECK="CHECK";
  const char* const CLOSED="CLOSED";
  const char* const OPEN="OPEN";
  const char* const ARMED="ARMED";
```
For example, for your panel, change the fault value to  FAULT="FALLO"; etc.  I don't know the other prompts so you will need to identify them and make the changes. ie, ALARM, FIRE, ARMED

Dilbert66, 2021-02-11:
You can also change the closed and open values and those will show on your home assistant as entered

morichina123, 2021-02-11:
Perfecttt !!!!!!!!!!!!!
Thank you very much for your help. I wouldn't have made it.

closed 2021-02-11 by morichina123