Dilbert66/esphome-vistaECP #10: No output to Vista 20P

rbauman70, 2021-02-26:
Hey, thanks for making this project available.  I can see you put a lot of time into it.

I've assembled the circuitry around a NodeMcu V1.0 and I'm getting status correctly in Home Assistant.  I've been unable to  get the Vista 20P panel to respond to anything sent from Home Assistant, though.  Using the esphome.vistaalarm_alarm_keypress service in the Developer Tools I issue a "#33" string which shows up in the logs on the node.  Here's a snippet:

`[19:25:11][I][CMD:333]: F7 02 00 FF 10 08 00 1C 08 02 00 00
[19:25:11][I][INFO:524]: Prompt: ****DISARMED****
[19:25:11][I][INFO:525]: Prompt:   Ready to Arm
[19:25:11][I][INFO:526]: Beeps: 0
[19:25:11]
[19:25:13][D][Debug:300]: Writing keys: #33
[19:25:19][I][CMD:333]: F6 01 00 00 00 00 00 00 00 00 00 00
[19:25:19][I][EXT:333]: F6 01 00 61 0C 00 6A 62 02 45 43 F5
[19:25:19][I][CMD:333]: F2 0E 02 00 00 00 00 52 FE FE EC 01
[19:25:19][I][INFO:524]: Prompt: ****DISARMED****
[19:25:19][I][INFO:525]: Prompt:   Ready to Arm
[19:25:19][I][INFO:526]: Beeps: 0
[19:25:19]
[19:25:20][I][CMD:333]: F7 02 00 FF 10 08 00 1C 08 02 00 00
[19:25:20][I][INFO:524]: Prompt: ****DISARMED****
[19:25:20][I][INFO:525]: Prompt:   Ready to Arm
[19:25:20][I][INFO:526]: Beeps: 0
[19:25:20]
`
KeypadAddr is set to 16.  I've modified the code to write to the GPIO16 output, the NodeMcu LED, but inverted, whenever the TX_PIN is written.  Here's a little movie of the LED activity.   I gave up my 'scope a few years ago so I can't actually look at the green line to to the panel.  Does this seem normal to you?  Any ideas for further debugging?

Ron 

https://user-images.githubusercontent.com/79720569/109345821-b8488d00-783e-11eb-8c32-605c0eef4ea5.mp4

Dilbert66, 2021-02-26:
Looks like you don't have something hooked up right with the optocoupler circuit going to the green line. Only thing it could be.

rbauman70, 2021-02-26:
Thanks for the quick response.  That's what I thought, too.  I forgot I had a USB logic analyzer, so I hauled that out and put some test points on the board.  Here's a couple of traces.  The TX signal is definitely getting to the green line.  But the NodeMcu seems to be the one generating the frequent 3-burst pattern.  

[image]

[image]

Does this look right?

Dilbert66, 2021-02-26:
There is an oddity, D5 monitorx should be the same as the tx line and the green line. ? Anyhow, the data looks right.     You will see the 3 bursts as the first is an f6 cmd  which asks the requesting device (in this case  address 0x10  decimal 16)   to send, then the cmd is sent, then an acknowledge is sent back.  So this tells me though that the cmd is received fine but ignored by your panel.  Try changing the device address from 16 to something else to see if that makes a difference.  Do other cmds work.

Dilbert66, 2021-02-26:
Actually, looking back at your logs.  The F6 response is not correct for your device. It's responding to device 01 and not to 10 (address 16).  Also it is very delayed so I don't think the panel is responding at all to the esp cmd.

rbauman70, 2021-02-27:
The VistaAlarm.yaml has keypadAddr set to 16.  

Some context: I have a Honeywell VAM attached to the panel.  It's essentially a Tuxedo touchscreen but without the screen.  It is considered an AUI device and defaults to ECP address 1 which accounts for the traffic you were seeing.  I've disconnected it so it won't obfuscate whatever problem we're looking at.  So now it's just the Vista 20P panel (firmware 10.23), one 6160 keypad at address 16 and the NodeMcu device on the ECP bus.   

From debug statements, I can confirm that keypresses are requested to be sent in rxHandleISR but never sent since writeChars is not called.

Dilbert66, 2021-02-27:
Well, it still points to a hardware wiring issue on the tx line since the fact that the software is attempting to send the request to send (in the rxhandleisr) but it is not sent out.  Without this happening, you won't see a corresponding f6 sent back to your keypad address to do the send in writechars.   This is not sofware related.  The tx data is just not going on the green line.

rbauman70, 2021-02-27:
I tried a different address and checked all the wiring again.

Here's a trace of an ECP request from a keypad when a key is pressed.  The keypad is subsquently polled with an F6 command and its address like you described

[image]

This is from a keypress requested from the ESP module.

[image]

This gets repeated at every tri-pulse burst.  Something is wrong with the timing or the state flow on my module.  There's a lot going on on this little device with Wifi running so maybe there's some timing issues.  I'll find it eventually.  This is a log of the build so you can see the toolchain I'm using.

[ESP-build.zip](https://github.com/Dilbert66/esphome-vistaECP/files/6055071/ESP-build.zip)

Dilbert66, 2021-02-27:
The display above is not showing enough of the interaction to know what happens when a cmd sequence is sent. Such as the RTS pulses (which you do show) (on D1)  then  followed by the the f6 10 xxx ( on D2)  then the written module data on (D2) then finally followed by the ack byte (D1).  But it does show the RTS sequence fine.  The odd part is the esp output data is not showing on the d5 monitortx line compared to the trace with the keypad which does show it.     The green and d5 monitortx lines should be the same no matter what device is sending.  Very odd behavior which to me indicates a connection issue or a bad ground connection somewhere.    The other odd aspet is that looking at your esp build zip, there is the initial startup sequence which shows the * being sent by the esp and acknowledge by the panel.  I would need to seem more logs of the interaction when other cmds are sent.  I have not seen this behaviour before.   No matter how busy the device is, it should not matter since the rts is interrupt driven.  You will always see the panel F6 10 respsonse.   If the chip is busy for some reason, the delay would be with the esp response to the f6 after (what you see in the EXT debug print)

```
[18:10:38][D][debug:832]: Vista::handle f6 10
[18:10:38][I][CMD:333]: F6 10 00 00 00 00 00 00 00 00 00 00 // panel cmd f6 to request device 0x10 (16) to go ahead and send
[18:10:38][I][EXT:333]: F6 10 00 10 02 0A E4 00 00 00 00 00  // esp response to f6. Len 02 byte sent: 0A (*) E4 checksum
[18:10:38][I][CMD:333]: F2 12 06 00 00 00 00 65 6C 02 45 6C
[18:10:38][I][INFO:524]: Prompt: ****DISARMED****
[18:10:38][I][INFO:525]: Prompt:   Ready to Arm
[18:10:38][I][INFO:526]: Beeps: 0
```

rbauman70, 2021-02-27:
Here's the keypad keypress traces zoomed out.

[image]
[image]

The ESP request just repeats.  

[image]

Dilbert66, 2021-02-27:
You can see the difference.  The esp request to send is not showing up on the d5 monitor tx output (which is the green line). You have a connection issue somewhere.  If you look at the keypad output you can see it the data is showing up fine there.  This tells me that the panel is not seeing the 3 pulse request to send.

Dilbert66, 2021-02-27:
Also looking at this output from d0 Green line, that one does not make sense to me. Not sure where you're getting that data.

Dilbert66, 2021-02-27:
What optocoupler are you using?

rbauman70, 2021-02-27:
The MONITORTX does show the ESP request if I crank the logic threshold down to .85v.

[image]

I'm using a 4N26.  Could it be not pulling the green line high, like in the active region?  Doesn't seem likely.  I'm more concerned that what should be a single pulse during the third period is showing as two.  It's also shorter than the third one that the keypad is sending.

Dilbert66, 2021-02-27:
The rts is not supposed to be a single pulse, It's a binary address.  0x10 ie  1010  while the keypad address is 0001 which is why they are different.  The fact that you have to crank the threshold for the esp and not for the keypad is odd. It still looks to be an issue with the pulse at the optocoupler to me.   Perhaps try a smaller input resistor such as 220ohm

Dilbert66, 2021-02-27:
ok, I went and I hooked up my analyzer, and looking at my output, the rts to send on mine is a different for address 16.  For address 16 you would see a double wide pulse there to indicate address 0x10 and not the double pulse you show there.  That shows an address above 16.  Are you using address 18? I've attached a sample of mine showing a rts for address 16 and the corresponding f6 and ack
[image]

rbauman70, 2021-02-27:
I'm using address 18 which accounts for the difference in pulses.  The resistor change worked!  The 4N26 is spec'd for 10 mA forward LED current.  (3.3v - .7v)/.01A = 260 ohms, so 330 probably wasn't allowing enough drive.  Thanks for helping me with this.  Now I can get the alarm system finished up and work on other Home Assistant integrations.

Dilbert66, 2021-02-27:
Good stuff. Glad you got that sorted out.  I will change the schematic to lower that value. It works fine for me but that all depends on the tolerances.

closed 2021-02-27 by rbauman70

Dilbert66, 2021-02-27:
Schematic updated.