Dilbert66/esphome-vistaECP #11: Add a wiki?

appleguru, 2021-03-12:
Any way you can add a wiki to this project so we can build up a knowledge base?  (Eg, compatible hardware, examples builds, BOM…)

Dilbert66, 2021-03-13:
I really don't have the resources at this time to maintain it and have not seen any real requests for more info then what is already available. It's a fairly basic circuit.  If there is some confusion , I can always amend the docs.

appleguru, 2021-03-13:
Just set it up and open up permissions so anyone can edit? Happy to contribute with a build guide/some photos/getting started/etc.

I’m still trying to sort out “ideal” part numbers for the voltage regulator (which if I understand correctly should allow it to be powered from the vista system?), for the optocoupler, and for the nodeMCU/ESP module itself. I think I can order everything I need for the build from mouser (if I’m willing to overpay a bit for the esp module).

Dilbert66, 2021-03-13:
https://www.amazon.ca/Converter-Module-Keenso-Step-Down-Conversion/dp/B07PQLNXVC/ref=sr_1_26?dchild=1&keywords=step+down+mini&qid=1615644230&sr=8-26

or any lm2596 based regulator
https://www.amazon.ca/Converter-Ajustable-Step-down-Converters-1-25V-35V/dp/B074J4DLWQ/ref=asc_df_B074J4DLWQ/?tag=googleshopc0c-20&linkCode=df0&hvadid=335365451913&hvpos=&hvnetw=g&hvrand=1334836585404895701&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9000757&hvtargid=pla-899437512675&psc=1

Both of those work extremely well.  None of the parts are critical. Any inexpensive optocoupler will work. Same goes with the esp module.  Any inexpensive one will work. I do suggest though that you power it with 3.3 volts (at the 3.3 volt input of course) instead of with 5 volts to avoid using the weak built in 5volt regulators on some esp modules. This is why i recommend the regulators above (or similar) as they are fully adjustable. 

As to the wiki, I'll think about it.

appleguru, 2021-03-13:
What about for an optocoupler? Will a cheap PC817 work?

Dilbert66, 2021-03-13:
Sure, I think it would work perfectly

appleguru, 2021-03-13:
Thanks. Did an Amazon order (which I generally avoid for electronics due to all the counterfeit parts, but should be ok for this circuit…)

Thanks for putting this together, excited to finally tie my (to date completely useless) existing security system into home assistant!

appleguru, 2021-03-16:
Ok, my build is done and seems to be (mostly?) working. I think the remaining issues are likely due to configuration which I could definitely use some help with.

[image]
[image]
[image]

I disabled the expander, relay, and lrrsupervisor in the config since I *think* I have real devices on the system for all of those at present.

Arming/disarming from home assistant also don’t seem to be working (but this is likely user error? I’m trying to use Developer Tools -> Services for this now but it seems to have no effect. I put my alarm code into a secrets.yaml file as “access_code”

I think I also might be missing some devices? (But am not 100% sure what I have on the vista system). Any help would be greatly appreciated!

Dilbert66, 2021-03-16:
If you can't send keys that means your tx circuit is not working. Looking at your circuit soldering, the optocoupler looks to be miswired.  It looks like your are bridging pins 3 and 4.

appleguru, 2021-03-16:
I don’t think the pins are bridged, but will double check with a meter when I get home. What’s the easiest way to test TX? Arm/disarm?

Dilbert66, 2021-03-16:
Looking at the soldering, they do look soldered together.  only pins 2 and 3 go to ground.
[image]

Dilbert66, 2021-03-16:
Arm /disarm/send a key. You should see a logged response on the esphome log window.

appleguru, 2021-03-16:
The circled pins are actually 2/3 on the options:
[image]

I think this might have my current alarm code in it, so will probably delete or edit after, but this is what I see for debug when I try to arm:
[image]

Dilbert66, 2021-03-16:
If i were you I would double check the pinout on the 817 

https://www.twinschip.com/PC817-Optocoupler

appleguru, 2021-03-16:
> If i were you I would double check the pinout on the 817 . The wiring is all wrong from what I see.
> 
> https://www.twinschip.com/PC817-Optocoupler

I think i have it right? (I misspoke before, I circled 3/4, not 2/3)

Dilbert66, 2021-03-16:
Yes, that's correct but on the picture pin2 is going direct to  the esp module? I don't get it? That should be ground

appleguru, 2021-03-16:
> Yes, that's correct but on the picture pin2 is going direct to the esp module? I don't get it? That should be ground

It’s going to a ground pin on the ESP module. They all should be connected; was just the closest ground to the part.

Dilbert66, 2021-03-16:
ah ok..  I see now.  The solder on the underside threw me off also.

Dilbert66, 2021-03-16:
Either way, if the log is not showing  a key was sent then, you might still have a wiring issue somewhere.

appleguru, 2021-03-16:
> Either way, if the log is not showing a key was sent then, you might still have a wiring issue somewhere.

Can you take a look at the log screenshot above? I’m not sure what I’m looking at/if it shows the key being sent (I think it does?)

Dilbert66, 2021-03-16:
It looks like your sending fine now that I have a closer look at your log image.  The F6 is receiving from device id 0x10 (address 16) which I assume is your esp.

Try using the alarm_keypress service from developer tools and send any keys for example. You can send any keys you want there.

service: esphome.vistaalarmtest_alarm_keypress
data: {"keys":"*1234*1234"}

Dilbert66, 2021-03-16:
This is my test system so I'm using 1234 as an access code

Dilbert66, 2021-03-16:
Those F6 ext cmds (the EXT is a pseudo field to display sent keys from peripherals in response to the previous F6 CMD) don't make sense.  they show a lenght of 1 for the associated. Normally for an arm cmd you will see a lenght of 6 ie. the 4 digit code, the 1 digit arm code and a 1 digit checksum.   That would explain why it's not arming.

adamurban, 2021-03-16:
Interesting.. Any idea what might cause that? Here's the output using the same 12342 example that you posted (I also of course tried with my actual alarm code too).

[image]

Dilbert66, 2021-03-16:
Interesting. This looks to be a software bug somewhere.  I'm running the same version as the master I think.  I'll check to see.

Dilbert66, 2021-03-17:
I recompiled my test version from the master copy and everything works fine on my end.  Well, it's a mystery to me why your version is behaving like that. I assume you are using the latest master code version. I really can't see how the issue you have would occur.  
What panel version do have?

Dilbert66, 2021-03-17:
Can you post a longer copy and paste of the logs so I can view normal interactions.   Also can you enter any random sequence of characters in the physical keypad and post the associated esp logs as well.

Dilbert66, 2021-03-17:
It's possible that's it's conflicting with another device. Can you change your esp address to another address such as 17 instead of the default 16.

adamurban, 2021-03-17:
Trying a different address now; if that doesn't solve it I'll grab some logs.

adamurban, 2021-03-17:
Success! Moved to address 17 and arming/disarming works ok now.

appleguru, 2021-03-17:
Success! Moved to address 17 and arming/disarming works ok now.

appleguru, 2021-03-17:
Panel is version 10.23, which is why I assume address 17 worked for a keypad…
Looking at the hardware, I have:
1) Vista-20p control panel
2) 6160 keypad
3) 5881ENL wireless receiver
4) GSMV4G alarm radio (for remote access??)
5) SRN2000 motion sensor
6) a crap load of contact and bar sensors…
7) WI-MTA1/5817 wireless temp sensor
8) WI-WB200 water sensor
9) ADEMCO-4204 ECP relay module (connected to a system sensor COSMOD2W 2wire fire alarm interface (which connects to COSMO2W smoke/CO detectors)
10) SDM-100 siren driver connected to a AS-SHX51 outdoor siren
11) 4219 wired zone expander
12) MPI35F indoor siren

And probably some other things that I missed. I don’t currently have the programming code for the system, but am hoping to get that tomorrow.

Any tips for identifying everything in HA/making sure I’m not missing any devices?

adamurban, 2021-03-17:
From DIP switches, I see that my real 4219 wired zone expander is on address 7 and my real 4204 relay module is on address 12.

Presumably because I have a GSMV4G, I want lrrSupervisor: "false" right?
And if I want to use relay and/or zone emulation, I would want to change the default relayAddr1: "12" to 13 or something else, but expanderAddr1: "8" is OK. (I have these all set to 0 for now as I don't yet have a need for the emulated inputs).

Is there a good way to figure out what addresses everything else in the alarm system are set to?

appleguru, 2021-03-17:
From DIP switches, I see that my real 4219 wired zone expander is on address 7 and my real 4204 relay module is on address 12.

Presumably because I have a GSMV4G, I want lrrSupervisor: "false" right?
And if I want to use relay and/or zone emulation, I would want to change the default relayAddr1: "12" to 13 or something else, but expanderAddr1: "8" is OK. (I have these all set to 0 for now as I don't yet have a need for the emulated inputs).

Is there a good way to figure out what addresses everything else in the alarm system are set to?

Dilbert66, 2021-03-17:
That's quite a loaded system you have. Yes, using the virtual expander with address 8 is ok and changing the relayaddr1 to 13 is correct also. And yes, since you already have a GSMv4g, you definitively want to set lrrSupervisor to False.  That interface is used to report faults to a remote using cell or pager.  With my system I just use those messages for display as it provides more info.

As to finding out what zones you have, there are two ways I guess.  Open up every zone and record the message displayed on the keypad or the esp.  The other method requires having the installer code and going to program *56, press 1 (to confirm) then use the * key to step through all the programming to see what zones are setup and what type. 

Did you try installer code 4112 or 6321 . Those are the defaults .

You enter 4112800 or 6321800 and see if it gets you the installer prompt.   You should download a vista20p installer manual . Just google it. Lots of copies around.

appleguru, 2021-03-17:
Yeah, the alarm company changed the default installer code; I tried both of those. I called them today and they said they should be able to remotely reset it back to the default code for me tomorrow; we’ll see if it works?

Used your example template to add a keypad, looks nice! I don’t really need/want to have to enter the code each time to disarm, so just hard-coded the “code” field and manually added the “disarm” state, which seems to work. Only problem now is the virtual keypad is useless… any way to make it pass the button presses through to the alarm_keypress service so I can use it like the real keypad on the wall?

Dilbert66, 2021-03-17:
What's needed is a different virtual keypad component. You are right that the existing keypad is very limited as it is now. Not sure what's available but that's a good idea though to fully emulate a keypad. If there isnt one around, I might try my hand at coding one up.  

Once you find out your zones, you'll need to add new sections in the yaml to represent those and edit the existing to match your setup.

appleguru, 2021-03-17:
I don’t think it quite works like this, but being able to conditionally call a service might work… something like:

`
        disarm: >
            {% if {{code}} %}
              - service: esphome.vistaalarm_alarm_keypress
                data_template:
                  keys: '{{code}}'
            {% else %}
              - service: esphome.vistaalarm_alarm_disarm
              data_template:
                code: '1234'
            {% endif %}
        `

Dilbert66, 2021-03-17:
Maybe, I have not really looked too deeply into it since as it is the default panel is limited in it's ability to work as a standalone keypad since it is missing the "#" and "*" keys which are needed in a lot of commands, such as programming, etc.   I'm going to look into adapting an existing custom card to add those

appleguru, 2021-03-17:
Ok, progress.. found some system documentation with zone numbers listed that seems mostly correct, and also got the programming code.

 I updated the yaml template entires and case statement entries like this:

```
   - platform: template
    id: z15
    name: "$systemName Basement Water Sensor"
    

case 15: id(z15).publish_state(open); break;
```

I now have a bunch on unavailable entities that I can’t seem to remove for the zones I no longer have, but the things that are there seem to be working. Will dive into the panel config on the keypad later and double check everything once I figure out how :D

One question: I see a few entries in the log like this:
```
 Got relay address 7 channel 12 = 0
 Got relay address 12 channel 56 = 0
 Got relay address 12 channel 53 = 0
 ```
Does this mean there are input devices I don’t yet have mapped? Or can I ignore these? I guess I still need to wrap my head around how addressing works in the vista system (address vs channel vs zone…)

Dilbert66, 2021-03-18:
You might need to delete the esphome vista alarm entity and re-add it. That's the only I found to remove ghost entries after changes.  Those relay messages are status changes from your panel.  You can ignore those unless you want to see the relay activities.  You can also comment out that log message from vistaalarm.h if you don't want to see it. It's more of a debug message than anything.

Dilbert66, 2021-03-18:
By the way, I'm testing a custom keypad component that will work just like a regular alarm keypad with all keys. I also found an issue with the way my code handles key writes that I need to fix up so I will be pushing an update in a day or so.

Dilbert66, 2021-03-18:
I've pushed a new commit to the "DEV" branch.  This fixes the key send issue and adds a new custom panel card which emulates a regular keypad.   You'll need to use the new yaml or at least add the changes below:
```
    VistaECP->onLine1DisplayChange([&](std::string msg) {
        id(l1).publish_state(msg); 
    });  
    VistaECP->onLine2DisplayChange([&](std::string msg) {
        id(l2).publish_state(msg); 
    });   
``
```
  - platform: template
    id: l1
    name: "$systemName Line1"
 

  - platform: template
    id: l2
    name: "$systemName Line2"
```

Dilbert66, 2021-03-18:
I will add a few more buttons later such as the arm night/day/etc buttons and disarm.

adamurban, 2021-03-18:
Nice work! Giving it a shot now

adamurban, 2021-03-18:
I think the c files are missing a function (unless I copied the wrong files??)

```
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:616:17: error: 'class vistaECPHome' has no member named 'onLine1DisplayChange'
       VistaECP->onLine1DisplayChange([&](std::string msg) {
                 ^
src/main.cpp:619:17: error: 'class vistaECPHome' has no member named 'onLine2DisplayChange'
       VistaECP->onLine2DisplayChange([&](std::string msg) {
                 ^
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
*** [/data/vistaalarm/.pioenvs/vistaalarm/src/main.cpp.o] Error 1
```

appleguru, 2021-03-18:
I think the c files are missing a function (unless I copied the wrong files??)

```
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:616:17: error: 'class vistaECPHome' has no member named 'onLine1DisplayChange'
       VistaECP->onLine1DisplayChange([&](std::string msg) {
                 ^
src/main.cpp:619:17: error: 'class vistaECPHome' has no member named 'onLine2DisplayChange'
       VistaECP->onLine2DisplayChange([&](std::string msg) {
                 ^
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
*** [/data/vistaalarm/.pioenvs/vistaalarm/src/main.cpp.o] Error 1
```

Dilbert66, 2021-03-18:
My copy missed some files. I've pushed them now. Try it again.

Dilbert66, 2021-03-18:
I've updated the card to now allow configuring the 4 function keys to send any sequence of keys.  You will need to update the resource file like so: /local/alarm-keypad-card-js?ver=1 (where 1 will increment everytime you need to refresh changes to it), You will then need to do an F5 on the browser to refresh the page and load the new version.

appleguru, 2021-03-18:
Slick!

I didn’t see your example card yaml anywhere, so hard to type that from your screenshot… here’s mine:

```
type: 'custom:alarm-keypad-card'
title: Vista
unique_id: vista
kpd_line1: sensor.vistaalarm_line1
kpd_line2: sensor.vistaalarm_line2
scale: 1
view_pad: true
kpd_service: sensor.vistaalarm_keypress
button_A: STAY
button_B: AWAY
button_C: DISARM
button_D: BYPASS
cmd_A: '12343'
cmd_B: '12342'
cmd_C: '12341'
cmd_D: '12346#'
```

[image]

appleguru, 2021-03-18:
This virtual keypad is absolute 🔥… I can now do programming right from HA and don’t have to stand in the hallway taking photos of the LCD like a dork!

Dilbert66, 2021-03-18:
I didnt post a yaml since I configured it using the gui. I can post a yaml too

appleguru, 2021-03-18:
> I didnt post a yaml since I configured it using the gui. I can post a yaml too

Maybe I missed the UI option? How/where would I find that? (I didn’t see it in “Add Card”, clicked custom at the bottom and just typed from your example screenshot. It helpfully prompted me for missing fields)

Dilbert66, 2021-03-18:
When you edit a dashboard, you will then see an Edit on each card where you can make changes there.

appleguru, 2021-03-18:
> When you edit a dashboard, you will then see an Edit on each card where you can make changes there.

right, but where does the card come from in the first place?

Dilbert66, 2021-03-18:
from the js code in the resources. When you add a card, it just load the js code and displays it in whatever dashboard you are using.

Dilbert66, 2021-03-18:
I've pushed an update that fixes an issue with the * being sent while in programming mode.  This would cause odd behaviour if your trying to program the system while using the card.

appleguru, 2021-03-20:
Thanks, seems to be working well. I noticed you checked in some mp3 files and made a few changes for "beeps". Do those work? I copied them over but they don't seem to do anything.

Unrelated, I opened a small PR for the js to improve the button size/spacing to make the keypad behave better on mobile (and also match the native themes): https://github.com/Dilbert66/esphome-vistaECP/pull/12

Dilbert66, 2021-03-20:
The beep is still a work in progress.  It needs to have the beep duration better controlled.  It does  beep depending on the beep code sent from the esp (you need to also have the beep section setup in the esp yaml).
Add this line to the card config:
```
beep: sensor.vistaalarmtest_beeps
```

Awesome about the PR, I'll check it out. I also want to add feedback on a keypress to show visually when the button is pressed.  This polymer entity stuff is new to me .

Dilbert66, 2021-03-20:
Tried the changes. Excellent. Exactly what it needed.

appleguru, 2021-03-20:
I had to enable sound "auto play" in my browser too (it worked ok in the HA app out of the box at least).

Those beeps are pretty bad; maybe I'll just make some recordings of the actual beeps..

[8 more replies]

closed 2026-10-17 by Dilbert66