Dilbert66/esphome-vistaECP #13: my system doesn't seem to respond to keypresses

lanefloch, 2021-03-20:
As far as reading info from my Ademco 5881 system goes I'm having no problems. But there seems to be no response from the keypress service calls. I've verified with an oscilloscope that the signals are being sent out. Am I missing something in the configuration?

Dilbert66, 2021-03-20:
Try changing the keypad address to 17 instead of 16. If that does not work, post logs showing the interaction when a key is sent.

lanefloch, 2021-03-20:
I tried changing to 17 and 18, still no luck. I also switched my resistor on my optocoupler to the new lower value. 

`[13:12:25][I][app:105]: ESPHome version 1.16.2 compiled on Mar 20 2021, 13:10:25
[13:12:25][C][wifi:443]: WiFi:
[13:12:25][C][wifi:303]:   SSID: [redacted]
[13:12:25][C][wifi:304]:   IP Address: 192.168.1.252
[13:12:25][C][wifi:306]:   BSSID: [redacted]
[13:12:25][C][wifi:307]:   Hostname: 'vistaalarm'
[13:12:25][C][wifi:311]:   Signal strength: -75 dB ▂▄▆█
[13:12:25][C][wifi:315]:   Channel: 11
[13:12:25][C][wifi:316]:   Subnet: 255.255.255.0
[13:12:25][C][wifi:317]:   Gateway: 192.168.1.1
[13:12:25][C][wifi:318]:   DNS1: 192.168.1.1
[13:12:25][C][wifi:319]:   DNS2: (IP unset)
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Trouble'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Bypass'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Away'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Stay'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Instant'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Night'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm AC'
[13:12:25][C][template.binary_sensor:018]:   Device Class: 'plug'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Chime'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Check'
[13:12:25][C][template.binary_sensor:018]:   Device Class: 'problem'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Alarm'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Battery'
[13:12:25][C][template.binary_sensor:018]:   Device Class: 'problem'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Fire'
[13:12:25][C][template.binary_sensor:018]:   Device Class: 'smoke'
[13:12:25][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Ready'
[13:12:26][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Relay1'
[13:12:26][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Relay2'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Front door'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Garage door'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Back door'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Living room window'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Dining room window'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Family room window LF'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Family room window RF'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Basement windows'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm zone 17'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm System Status'
[13:12:26][C][template.text_sensor:020]:   Icon: 'mdi:shield'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Lrr Msg'
[13:12:26][C][template.text_sensor:020]:   Icon: 'mdi:alert-box'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Line1'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Line2'
[13:12:26][C][template.text_sensor:020]: Template Sensor 'vistaalarm Beeps'
[13:12:26][C][template.switch:058]: Template Switch 'vistaalarm Connection'
[13:12:26][C][template.switch:058]:   Icon: 'mdi:shield-link-variant'
[13:12:26][C][template.switch:059]:   Restore State: NO
[13:12:26][C][template.switch:060]:   Optimistic: NO
[13:12:26][C][logger:185]: Logger:
[13:12:26][C][logger:186]:   Level: DEBUG
[13:12:26][C][logger:187]:   Log Baud Rate: 115200
[13:12:26][C][logger:188]:   Hardware UART: UART0
[13:12:26][C][restart:021]: Restart Switch 'restart_switch'
[13:12:26][C][restart:021]:   Icon: 'mdi:restart'
[13:12:26][C][ota:029]: Over-The-Air Updates:
[13:12:26][C][ota:030]:   Address: vistaalarm.local:8266
[13:12:26][C][ota:032]:   Using Password.
[13:12:26][C][api:095]: API Server:
[13:12:26][C][api:096]:   Address: vistaalarm.local:6053
[13:12:34][I][CMD:346]: F7 00 00 01 10 08 00 0C 08 02 00 00 
[13:12:34][I][EXT:346]: F6 11 00 51 02 0A A3 00 00 00 00 00 
[13:12:34][I][INFO:542]: Prompt: ****DISARMED****
[13:12:34][I][INFO:543]: Prompt: Hit * for faults
[13:12:34][I][INFO:544]: Beeps: 0

[13:12:40][D][Debug:313]: Writing keys: 1
[13:12:41][D][Debug:313]: Writing keys: 2
[13:12:41][D][Debug:313]: Writing keys: 3
[13:12:41][D][Debug:313]: Writing keys: 4
[13:12:42][I][CMD:346]: 9E 02 25 82 5C 00 00 00 00 00 00 00 
[13:12:42][I][EXT:346]: 9E 02 25 82 5C 00 00 00 00 00 00 00 
[13:12:42][I][EXT:346]: 9E 00 25 00 21 83 5C 00 00 00 00 00 
[13:12:44][I][CMD:346]: F7 00 00 01 10 08 00 0C 08 02 00 00 
[13:12:44][I][INFO:542]: Prompt: ****DISARMED****
[13:12:44][I][INFO:543]: Prompt: Hit * for faults
[13:12:44][I][INFO:544]: Beeps: 0

[13:12:53][I][CMD:346]: F7 00 00 01 10 08 00 0C 08 02 00 00 
[13:12:53][I][EXT:346]: F6 11 00 11 02 0A E3 00 00 00 00 00 
[13:12:53][I][INFO:542]: Prompt: ****DISARMED****
[13:12:53][I][INFO:543]: Prompt: Hit * for faults
[13:12:53][I][INFO:544]: Beeps: 0

[13:13:03][I][CMD:346]: F7 00 00 01 10 08 00 0C 08 02 00 00 
[13:13:03][I][EXT:346]: F6 11 00 51 02 0A A3 00 00 00 00 00 
[13:13:03][I][INFO:542]: Prompt: ****DISARMED****
[13:13:03][I][INFO:543]: Prompt: Hit * for faults
[13:13:03][I][INFO:544]: Beeps: 0

`

Dilbert66, 2021-03-20:
It looks like your system is sending since I see an F6 11 response.  Can you comment out the line (460) in vistaalarm.h below and post some logs again.
```
  while(vista.keybusConnected && vista.sendPending()) vista.handle();
```

lanefloch, 2021-03-20:
`INFO Reading configuration /config/esphome/ademco.yaml...
INFO Starting log output from vistaalarm.local using esphome API
INFO Connecting to vistaalarm.local:6053 (192.168.1.252)
INFO Successfully connected to vistaalarm.local
[14:13:17][I][app:105]: ESPHome version 1.16.2 compiled on Mar 20 2021, 14:12:16
[14:13:17][C][wifi:443]: WiFi:
[14:13:17][C][wifi:303]:   SSID: [redacted]
[14:13:17][C][wifi:304]:   IP Address: 192.168.1.252
[14:13:17][C][wifi:306]:   BSSID: [redacted]
[14:13:17][C][wifi:307]:   Hostname: 'vistaalarm'
[14:13:17][C][wifi:311]:   Signal strength: -68 dB ▂▄▆█
[14:13:17][C][wifi:315]:   Channel: 11
[14:13:17][C][wifi:316]:   Subnet: 255.255.255.0
[14:13:17][C][wifi:317]:   Gateway: 192.168.1.1
[14:13:17][C][wifi:318]:   DNS1: 192.168.1.1
[14:13:17][C][wifi:319]:   DNS2: (IP unset)
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Trouble'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Bypass'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Away'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Stay'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Instant'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Night'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm AC'
[14:13:17][C][template.binary_sensor:018]:   Device Class: 'plug'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Chime'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Check'
[14:13:17][C][template.binary_sensor:018]:   Device Class: 'problem'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Alarm'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Battery'
[14:13:17][C][template.binary_sensor:018]:   Device Class: 'problem'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Fire'
[14:13:17][C][template.binary_sensor:018]:   Device Class: 'smoke'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Ready'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Relay1'
[14:13:17][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Relay2'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Front door'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Garage door'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Back door'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Living room window'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Dining room window'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Family room window LF'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Family room window RF'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Basement windows'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm zone 17'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm System Status'
[14:13:17][C][template.text_sensor:020]:   Icon: 'mdi:shield'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Lrr Msg'
[14:13:17][C][template.text_sensor:020]:   Icon: 'mdi:alert-box'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Line1'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Line2'
[14:13:17][C][template.text_sensor:020]: Template Sensor 'vistaalarm Beeps'
[14:13:17][C][template.switch:058]: Template Switch 'vistaalarm Connection'
[14:13:17][C][template.switch:058]:   Icon: 'mdi:shield-link-variant'
[14:13:17][C][template.switch:059]:   Restore State: NO
[14:13:17][C][template.switch:060]:   Optimistic: NO
[14:13:17][C][logger:185]: Logger:
[14:13:17][C][logger:186]:   Level: DEBUG
[14:13:17][C][logger:187]:   Log Baud Rate: 115200
[14:13:17][C][logger:188]:   Hardware UART: UART0
[14:13:17][C][restart:021]: Restart Switch 'restart_switch'
[14:13:17][C][restart:021]:   Icon: 'mdi:restart'
[14:13:17][C][ota:029]: Over-The-Air Updates:
[14:13:17][C][ota:030]:   Address: vistaalarm.local:8266
[14:13:17][C][ota:032]:   Using Password.
[14:13:17][C][api:095]: API Server:
[14:13:17][C][api:096]:   Address: vistaalarm.local:6053
[14:13:24][I][CMD:346]: F7 00 00 01 10 08 00 0C 28 02 00 00 
[14:13:24][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:13:24][I][INFO:543]: Prompt: Hit * for faults
[14:13:24][I][INFO:544]: Beeps: 0
[... 16 lines cut ...]
[14:13:34][I][INFO:543]: Prompt: Hit * for faults
[14:13:34][I][INFO:544]: Beeps: 0

[14:13:34][D][Debug:313]: Writing keys: 4
[14:13:34][I][CMD:346]: F6 12 00 00 00 00 00 00 00 00 00 00 
[14:13:35][I][EXT:346]: F6 12 00 52 03 0A 04 9D 00 00 00 00 
[14:13:38][I][CMD:346]: F6 12 00 00 00 00 00 00 00 00 00 00 
[14:13:38][I][EXT:346]: F6 12 00 92 02 0A 62 00 00 00 00 00 
[14:13:44][I][CMD:346]: F7 00 00 01 10 08 00 0C 28 02 00 00 
[14:13:44][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:13:44][I][INFO:543]: Prompt: Hit * for faults
[14:13:44][I][INFO:544]: Beeps: 0

[14:13:44][I][CMD:346]: F6 12 00 00 00 00 00 00 00 00 00 00 
[14:13:45][I][EXT:346]: F6 12 00 D2 02 0A 22 00 00 00 00 00 
[14:13:54][I][CMD:346]: F7 00 00 01 10 08 00 0C 28 02 00 00 
[14:13:54][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:13:54][I][INFO:543]: Prompt: Hit * for faults
[14:13:54][I][INFO:544]: Beeps: 0

[14:13:54][I][CMD:346]: F6 12 00 00 00 00 00 00 00 00 00 00 
[14:13:54][I][EXT:346]: F6 12 00 12 02 0A E2 00 00 00 00 00 
[14:13:55][I][CMD:346]: F6 12 00 00 00 00 00 00 00 00 00 00 
[14:13:55][I][EXT:346]: F6 12 00 52 02 0A A2 00 00 00 00 00 
`

Dilbert66, 2021-03-20:
I'm not familiar with the 5881? A google of that number tells me it's the wireless receiver module and not the actual system.  From the logs above, the system is definitively receiving the key sent but it's not the data sent.   It could be that your system is not supported by my code or it doesnt like the keypad address used and is ignoring it.  I would assume it's supported since the protocol response looks fine.
Example:the CMD line is from the panel and is asking address 12 hex (18 decimal) to go ahead and send.
The EXT is the response from the ESP which sends 52 03 02 03 E6 (header len [key 02] [key 03] chksum)
```
[14:13:34][D][Debug:313]: Writing keys: 2
[14:13:34][D][Debug:313]: Writing keys: 3
[14:13:34][I][CMD:346]: F6 12 00 00 00 00 00 00 00 00 00 00
[14:13:34][I][EXT:346]: F6 12 00 12 03 02 03 E6 00 00 00 00
```

Dilbert66, 2021-03-20:
Did you try address 16?

lanefloch, 2021-03-20:
yea the 5881 is a wireless receiver. That's really odd, you'd think it wouldn't be too different, yes I've tried 16,17, and 18 before and after commenting out that line.

Dilbert66, 2021-03-20:
At this point I really don't know since it all all looks ok from the logs.  Can you identify the exact panel model you  have?

lanefloch, 2021-03-20:
I’m an idiot.... I’m so sorry I told you the wrong model for my panel. I’m not super familiar with these systems. Just bought a house that had one already installed and I’m trying to make use of it. The 5881 is my wireless receiver. I’m not seeing an exact model number for my panel. I’ll attach a picture of the board and the documentation inside the enclosure. I really appreciate you helping me and if you have a donate button on here somewhere I’ll throw you some money to get a coffee or something. (: 

[image]

[image]

lanefloch, 2021-03-20:
I believe my keypad is referred to as a 6160 as well.

Dilbert66, 2021-03-20:
it's a vista20p version 5.2 so it should work fine.  I did make a lot code changes last few ays related to the keypad entry but it works fine on my system and on another guys system who worked with me on some changes.  You can see the discussion about it in the other open issue.  It's possible that there is a bug but I can't see what it could be at this point.

lanefloch, 2021-03-20:
huh... is there anything as far as settings in the security system that could be affecting it? possibly an issue with the rf module? when you say access code, you mean the one I use to arm the system and not the installer code correct?

Dilbert66, 2021-03-20:
Actually if possible can you also post logs while you use the physical keypad on the system.  Don't use arm or disarm as they will show in the logs, just press the * key a few times.  This will give some comparison code between the virtual keypad and the real one.

Dilbert66, 2021-03-20:
Yes, i mean regular access code.

lanefloch, 2021-03-20:
There's a few presses of the * key on the keypad -

`INFO Reading configuration /config/esphome/ademco.yaml...
INFO Starting log output from vistaalarm.local using esphome API
INFO Connecting to vistaalarm.local:6053 (192.168.1.252)
INFO Successfully connected to vistaalarm.local
[14:56:35][I][app:105]: ESPHome version 1.16.2 compiled on Mar 20 2021, 14:16:47
[14:56:35][C][wifi:443]: WiFi:
[14:56:35][C][wifi:303]:   SSID: [redacted]
[14:56:35][C][wifi:304]:   IP Address: 192.168.1.252
[14:56:35][C][wifi:306]:   BSSID: [redacted]
[14:56:35][C][wifi:307]:   Hostname: 'vistaalarm'
[14:56:35][C][wifi:311]:   Signal strength: -69 dB ▂▄▆█
[14:56:35][C][wifi:315]:   Channel: 11
[14:56:35][C][wifi:316]:   Subnet: 255.255.255.0
[14:56:35][C][wifi:317]:   Gateway: 192.168.1.1
[14:56:35][C][wifi:318]:   DNS1: 192.168.1.1
[14:56:35][C][wifi:319]:   DNS2: (IP unset)
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Trouble'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Bypass'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Away'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Stay'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Instant'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Night'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm AC'
[14:56:35][C][template.binary_sensor:018]:   Device Class: 'plug'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Chime'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Check'
[14:56:35][C][template.binary_sensor:018]:   Device Class: 'problem'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Alarm'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Battery'
[14:56:35][C][template.binary_sensor:018]:   Device Class: 'problem'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Fire'
[14:56:35][C][template.binary_sensor:018]:   Device Class: 'smoke'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Ready'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Relay1'
[14:56:35][C][template.binary_sensor:018]: Template Binary Sensor 'vistaalarm Relay2'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Front door'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Garage door'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Back door'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Living room window'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Dining room window'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Family room window LF'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Family room window RF'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Basement windows'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm zone 17'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm System Status'
[14:56:35][C][template.text_sensor:020]:   Icon: 'mdi:shield'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Lrr Msg'
[14:56:35][C][template.text_sensor:020]:   Icon: 'mdi:alert-box'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Line1'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Line2'
[14:56:35][C][template.text_sensor:020]: Template Sensor 'vistaalarm Beeps'
[14:56:35][C][template.switch:058]: Template Switch 'vistaalarm Connection'
[14:56:35][C][template.switch:058]:   Icon: 'mdi:shield-link-variant'
[14:56:35][C][template.switch:059]:   Restore State: NO
[14:56:35][C][template.switch:060]:   Optimistic: NO
[14:56:35][C][logger:185]: Logger:
[14:56:35][C][logger:186]:   Level: DEBUG
[14:56:35][C][logger:187]:   Log Baud Rate: 115200
[14:56:35][C][logger:188]:   Hardware UART: UART0
[14:56:35][C][restart:021]: Restart Switch 'restart_switch'
[14:56:35][C][restart:021]:   Icon: 'mdi:restart'
[14:56:35][C][ota:029]: Over-The-Air Updates:
[14:56:35][C][ota:030]:   Address: vistaalarm.local:8266
[14:56:35][C][ota:032]:   Using Password.
[14:56:35][C][api:095]: API Server:
[14:56:35][C][api:096]:   Address: vistaalarm.local:6053
[14:56:39][I][CMD:346]: F7 00 00 01 10 08 00 1C 28 02 00 00 
[14:56:39][I][INFO:542]: Prompt:  DISARMED CHIME 
[... 32 lines cut ...]
[14:56:45][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:56:45][I][INFO:543]: Prompt:   Ready to Arm  
[14:56:45][I][INFO:544]: Beeps: 0

[14:56:45][I][CMD:346]: F7 00 00 01 10 08 00 1C 28 02 00 00 
[14:56:45][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:56:45][I][INFO:543]: Prompt:   Ready to Arm  
[14:56:45][I][INFO:544]: Beeps: 0

[14:56:46][I][CMD:346]: F6 10 00 00 00 00 00 00 00 00 00 00 
[14:56:46][I][EXT:346]: F6 10 00 50 02 0A A4 00 00 00 00 00 
[14:56:46][I][CMD:346]: F2 12 06 00 00 00 00 64 6C 02 45 6C 
[14:56:46][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:56:46][I][INFO:543]: Prompt:   Ready to Arm  
[14:56:46][I][INFO:544]: Beeps: 0

[14:56:46][I][CMD:346]: F7 00 00 01 10 08 00 1C 28 02 00 00 
[14:56:46][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:56:46][I][INFO:543]: Prompt:   Ready to Arm  
[14:56:46][I][INFO:544]: Beeps: 0

[14:56:46][I][CMD:346]: F2 16 06 00 00 00 00 65 63 02 45 43 
[14:56:46][I][INFO:542]: Prompt:  DISARMED CHIME 
[14:56:46][I][INFO:543]: Prompt:   Ready to Arm  
[14:56:46][I][INFO:544]: Beeps: 0`

Dilbert66, 2021-03-20:
Ok, your keypad is using address 16. You might need to go into your system programming and enable keypad 17. 
[image]

lanefloch, 2021-03-20:
That worked! somewhat... I now have full control via a home assistant, but a completely blank physical pannel.

lanefloch, 2021-03-20:
Update: got it going, just had to update the *190 parameter to 1 to enable a second keypad. Thank you so much for your help! yet again I'll drop you $5 for a coffee if you have a donation link.

closed 2021-03-20 by lanefloch

Dilbert66, 2021-03-20:
Awsome! No worries about the coffee! I'm glad you got it working.