Dilbert66/esphome-vistaECP #17: ESP01?

knifesk, 2021-05-04:
Hey, first of all.. awesome work.. I'm hooking things up to my Ademco Vista 48LA and things looks promising.. Used the wiring scheme to build my system and instead of an ESP I used my RPi to read from the bus... so, now that I know read is working I was planning to test if the writes does as well.... thing is the only thing I have at hand atm is an ESP01.. I already flashed the MQTT example, and it's connecting to my MQTT broker.. but is not sending anything :(

I know ESP01 is hard to work with because all the available pins have issues if pulled down on boot, but you think is it possible to make it work?

Dilbert66, 2021-05-04:
gpio1 and gpio3 are fine to use but there is a gotcha with the esp8266 and esphome. You need to either disable the logger uart0 access or swap those pins. See https://esphome.io/components/logger.html
You can set baud_rate: 0 and hardware_uart: UART0_SWAP.  The first will disable the logger and the second will switch the uart0 to different pins allowing you to use 1 and 3.  I have never tested this but it should work.

knifesk, 2021-05-05:
Yo're awesome man.. I was definetely not expecting such a fast reply... 

Ok, I'm gonna try what you said, but I will skip the re-wiring since I have gpio 2 available also.. My brain defaulted to TX and RX pins because that's what I was using on my Rpi..

Dilbert66, 2021-05-05:
I suspect you will get the same issue with gpio2 though. Can't hurt to try.

https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

knifesk, 2021-05-06:
Ok, so.. I manged to flash the modified fw into the esp01, changed all my wirings (swapped the anode and tx pins like you said) and now everything is booting up.. but since I haven't configured the panel yet it's showing errors everywhere I can't really tell if it is working... 

I got a NodeMCU and tried it out and it worked (I got tons of mqtt messages), but I couldn't activate the panel by the same reasons, but when I sent the payload A for the arm topic the panel would make a sound which is not doing with the esp01... ..also I knew things were working because I could se the serial ouptut... I don't have that with the 01. Also 
I'm not getting anything into MQTT when it boots...

But I know it must be partially working because I click the button to send the payload and I can see there's activity on the TX line with the multimeter..  

I tried the Swap method like you said (I added it into the setup function) but I don't think it did anything.. I got a lot of activity on the tx line even when I wasn't doing anything.. So I googled a bit and voila:  https://arduino.stackexchange.com/questions/29938/how-to-i-make-the-tx-and-rx-pins-on-an-esp-8266-01-into-gpio-pins

I have to keep testing.. I'm not defeated yet :)

Dilbert66, 2021-05-06:
Don't forget to enable the keypad address in the panel programming.  Make sure it's different then your physical keyapd's address.

knifesk, 2021-05-06:
yeah.. I actually got that sorted out from another issue.. my panel has
address 16 and I changed the esp address to 17.. that's how I got the
NodeMCU working and also that's how I know that even while the panel is
showing "FAULT 01, FAULT 02" with the NodeMCU I would hear a chime when I
tried to activate it... same as I heard when I try to activate it with the
physical keypad.. I feel I'm just a couple steps away from sorting it out..
I just need to go through all the panel's manual and config hehe

Dilbert66, 2021-05-06:
Try sending a bypass all command eg: 12346# where 1234 is your disarm code 6# is the bypass all faults.  If that works should see a ready to arm message or at least a few bypass messages.

knifesk, 2021-05-07:
well... as I was doing some tests I swapped TX for GPIO2 in order to be able to get the output out of the debug console.. Things were booting properly but I think the logic inversion is what's making the esp01 to receive nothing... but just to be sure I tried to measure the voltage on the voltage divider and I messed up the points where I made my measurements and I sent the 12v signal straight into the RX pin and blew it... so, no more ESP01 flashing haha.. I'll just go with the NodeMCU... I might try again someday when I get another esp01.. I don't think is possible anyway because:

if things are working properly without the  logic inversion in the RX pin... if you invert things then you will not get anything.. you will be able to boot properly and transmit data, but you loose the RX side... so.. no go.

I really appreciate your help anyway.. I wouldn't learned so mucho without your help!

knifesk, 2021-05-07:
It actually still works.. I just lost the ability to flash the chip because now the RX pin is dead... I don't really care... I knew that was a possibility and it's just the first time I actually damage the hardware... I've been pretty careful/lucky so far hehe.. At least it wasn't my RPI pin.. that would have hurt haha

Oh, and btw... the <CODE>6# did actually allowed me to arm the system.. I Just reverted the code to what it was, connected the NodeMCU and I was able to arm, disarm, bypass and everything. I'll be working a lot on this on the next weekend probably... I already made some code enhancements (mostly readability).. but once I commit something I'll make a pull request so everyone can grab them..

Thank you man for everything, you're AWESOME!

closed 2021-05-07 by knifesk

Dilbert66, 2021-05-07:
Glad to be of help.  As to the mqtt example, keep in mind that it is not up to date with the latest esphome version so some functions, bugs, etc might show up. I concentrate more on the esphome side of it as this is what I use. I provided the mqtt example to show how to use the library.  

I pushed an updated version of the Mqtt version that adds a small enhancement that fixes some key send timeouts or resends. You might want to add it to your copy.

Dilbert66, 2021-05-08:
FYI, I found a couple issues in the logic that can cause erratic behaviour and crashes in some circumstances.   I will fix those and push it to production.  

Also, this you might like.  I'm adding ArduinoOTA functions to the sketch so you will be able to upload via wifi from the Arduino console.

Dilbert66, 2021-05-08:
I've pushed a new version of the MQTT example that addresses the issues I found as well as added ArduinoOTA update support.