Dilbert66/esphome-vistaECP #21: Reading the bus but no F7s

jn3va, 2021-05-11:
Hi Alain

I connected my ESP to the alarm today and it seems to be decoding messages from the bus but I am not seeing any F7 messages (see attached).  Mostly F8 and 9E.   If I open a door the primary keypad shows a fault on that door, I don't see any F7 messages, just output below:

[20:21:02][I][CMD:347]: 9E 02 20 81 62 00 00 00 00 00 00 00 
[20:21:02][I][CMD:347]: 9E 02 25 60 7E 00 00 00 00 00 00 00 
[20:21:04][I][CMD:347]: F8 98 F8 66 1E 86 86 86 00 00 00 00 
[20:21:04][I][CMD:347]: 9E 98 60 E6 66 00 00 00 00 00 00 00 
[20:21:04][I][CMD:347]: F8 98 66 7E 00 78 00 FE 1E 00 00 00 
[20:21:04][I][CMD:347]: 98 F8 66 00 00 FE 00 00 00 00 00 00 
[20:21:05][I][CMD:347]: F8 F8 F8 F8 00 FE 1E F8 F8 00 00 00 
[20:21:05][I][CMD:347]: F8 F8 00 FE 1E F8 F8 F8 F8 00 00 00 
[20:21:12][I][CMD:347]: F8 F8 F8 F8 00 9E 02 20 F1 00 00 00 
[20:21:12][I][CMD:347]: F2 00 00 00 00 00 00 00 00 00 00 00 
[20:21:12][I][INFO:549]: Prompt: 
[20:21:12][I][INFO:550]: Prompt: 
[20:21:12][I][INFO:551]: Beeps: 0
[20:21:12][I][EXT:347]: 98 07 00 00 00 00 00 00 00 00 00 00 
[20:21:12][I][INFO:549]: Prompt: 
[20:21:12][I][INFO:550]: Prompt: 
[20:21:12][I][INFO:551]: Beeps: 0

[20:21:13][I][CMD:347]: F8 F8 78 98 60 E6 00 F8 00 00 00 00 
[20:21:13][I][CMD:347]: 98 66 7E 00 78 60 00 00 00 00 00 00 
[20:21:13][I][CMD:347]: 9E 86 E6 E0 86 00 00 00 00 00 00 00 
[20:21:14][I][CMD:347]: 9E 78 1E 7E F8 00 00 00 00 00 00 00 
[20:21:14][I][CMD:347]: F8 00 FE 1E 66 00 F8 7E FE 00 00 00 
[20:21:14][I][CMD:347]: F8 F8 80 7E 1E 00 00 FE 9E 00 00 00 
[20:21:14][I][CMD:347]: F8 78 06 7E 00 FE 9E 7E E0 00 00 00 
[20:21:16][I][CMD:347]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 
[20:21:16][I][EXT:347]: 9E 00 25 00 51 86 18 98 90 E9 00 00 
[20:21:17][I][CMD:347]: F8 98 F8 66 1E 86 86 86 00 00 00 00 
[20:21:17][I][CMD:347]: 9E 98 60 E6 66 00 00 00 00 00 00 00 
[20:21:17][I][CMD:347]: F8 98 66 7E 00 78 00 FE 1E 00 00 00 
[20:21:17][I][CMD:347]: 98 F8 66 00 00 FE 00 00 00 00 00 00 
[20:21:18][I][CMD:347]: F8 F8 F8 F8 00 FE 1E F8 F8 00 00 00 
[20:21:18][I][CMD:347]: F8 F8 00 FE 1E F8 F8 F8 F8 00 00 00 
[20:21:26][I][CMD:347]: F8 F8 F8 F8 00 00 00 E0 E0 00 00 00 
[20:21:27][I][CMD:347]: F8 98 F8 66 1E 86 86 86 00 00 00 00 
[20:21:27][I][CMD:347]: 9E 98 60 E6 66 00 00 00 00 00 00 00 
[20:21:27][I][CMD:347]: F8 98 66 7E 00 78 00 FE 1E 00 00 00 
[20:21:28][I][CMD:347]: 98 F8 66 00 00 FE 00 00 00 00 00 00 
[20:21:28][I][CMD:347]: F8 F8 F8 F8 00 FE 1E F8 F8 00 00 00 
[20:21:28][I][CMD:347]: F8 F8 00 FE 1E F8 F8 F8 F8 00 00 00 
[20:21:35][I][CMD:347]: F8 F8 F8 F8 00 9E 02 20 81 00 00 00 
[20:21:35][I][EXT:347]: 98 07 00 00 01 00 00 00 00 00 00 00 

All the sensors in my system are wireless and there is a 5881EN Wireless receiver attached to the keypad bus (that is using keypad ID 17).  I set the ESP to be Keypad ID 19.  I checked and MONITORTX was already uncommented.  I'm assuming the wireless sensors are reporting differently.  

I tried arming from the primary keypad (quick Arm #3) which results and no F7 messages

[20:29:52][I][CMD:347]: 9E 02 20 81 62 00 00 00 00 00 00 00 
[20:30:59][I][EXT:347]: 9E 00 20 00 24 00 DC 00 00 00 00 00 
[20:31:06][I][CMD:347]: F8 98 66 7E 00 78 00 FE 1E 00 00 00 
[20:31:06][I][CMD:347]: 98 F8 66 00 00 FE 00 00 00 00 00 00 
[20:31:06][I][CMD:347]: F8 F8 F8 F8 00 FE 1E F8 F8 00 00 00 
[20:31:06][I][CMD:347]: F8 F8 00 FE 1E F8 F8 F8 F8 00 00 00 

[V20P_log.txt](https://github.com/Dilbert66/esphome-vistaECP/files/6455669/V20P_log.txt)

Dilbert66, 2021-05-11:
What panel and version are you using? The 9E are RF messages which contain the device serial number and a device status byte..
The F8's are unkown to me at this time.  It almost points to AUI data but I don't have that type of setup or wifi so can't test it.  Did you also enable the keypad 19 in the vista panel config (prgram *192 I believe) ? It needs to happen.  What is your main keypad? Is it a touch model?

Dilbert66, 2021-05-11:
Also I see that your monitortx is working fine as I see some EXT response messages in your logs.

Dilbert66, 2021-05-11:
You didnt indicate that there was anything on 16? If there is, then of course, use a different address.  At this point I really don't get why you don't see the F7 but instead see a ton of F8 messages.  Must be because of something in the proprietery programming of that vista20p variant.

jn3va, 2021-05-12:
Awesome - thanks!   I might not be able to get to this again until Thursday but if I can get out of a meeting tomorrow I'll run it and post the output.  Have a great night.

Dilbert66, 2021-05-12:
No problem! Just note that you might get some odd logs as it will only capture about 4 bytes of the initial unknown command sthe rest will show up as another command.  It will skew the output  a bit but thats ok.  Lets see if we can identify this beast.

Dilbert66, 2021-05-12:
> You mentioned that the 9E are wireless sensor state reports. Do you have any reference or info about those? e.g. Which bytes have the address, which the state etc? Maybe I can try and decode some and modify the code to look at the individual info; that way even if I can't arm/disarm etc., I can at least read the sensors.

Just to clarify, the CMD: 9E's are sent from the panel.  They are just requests for status for each device.
The ones to look for are the EXT: 9E which are the responses from the sensors.  Those are the ones with the sensor status

jn3va, 2021-05-12:
Log attached after flashing with the DEF branch.  Let me know if you need more detail or to try anything else.

Joe
[ESPVista_DEV_Capture_20210512.txt](https://github.com/Dilbert66/esphome-vistaECP/files/6469299/ESPVista_DEV_Capture_20210512.txt)

Dilbert66, 2021-05-13:
From the looks of the data, there are a few more unknown commands in there. Perhaps a 1E, FE (but hard to know for sure due to the fact that we are just capturing groups of bytes and without knowing the length of the commands, it's tough to pin down. 
I use an inexpensive logic analyzer such as this one: https://learn.sparkfun.com/tutorials/using-the-usb-logic-analyzer-with-sigrok-pulseview/all with pulse view to identify the protocol.  It's invaluable for this type of stuff.

jn3va, 2021-05-13:
I'm pretty sure this is a newer (within 4-5 years) keypad.  Had to replace the original a few years ago.  The original panel is pretty old.  The keypad will display where the fault is by name that was programmed in by the original installer so I know the panel is sending that info, but I'm assuming it uses an older encoding that the newer keypads still understand.    I'm actually considering upgrading the panel since the wireless sensors all still work and are compatible with the Vista 20 and 21 panels.  Do you know if your interface works with the 21iP?  or just the 20P?  Once I swap it out I'll come back to this.  Also ever used or heard anything about AlarmGrid?  Seems like they have a much better rate for monitoring than I have today.

Dilbert66, 2021-05-13:
That's odd though so there must be an f7 sent somehow since those keypads work with the regular vista 20p.  I would really need to see the bus to know for sure. Anyhow, vista systems are pretty cheap any how especially if they work with the existing hardware you have.  I've never tested or know of anyone using a 21p but looking at the specs, since it works with the same bus hardware as the vista20p, I would assume it would work fine with my code.    I've never used AlarmGrid.  I've been using ADT for many years and they have been ok.  Not the cheapest though.

Dilbert66, 2021-05-13:
I had a another look at the cmds shown in the logs and none really make sense.  The 98's and 9E's are not correct.  The chksums don't even add up so it would seem the protocol is very different on this panel.  Interesting though that the lcd keypad can understand i It could be polarity, baud rate, etc.   it's almost as if the data is inverted? I can't imagine 2 different protocols.

jn3va, 2021-05-13:
I was watching a video about hose keypads.  They come factory set to address 31 - and mine is still set to address 31.  When you add them to a Vista20 you have to change the address to 16, 17, 13 on the keypad and then enable in the panel.  So I suspect there is something in the keypad code like
If ID=31 then
  Decode the data the old way....
else
  Decode the data beginning with "F7" the new way...

I'll check the wiring tonight
Thanks!

jn3va, 2021-05-13:
>> The 98's and 9E's are not correct. The chksums don't even add up so it would seem the protocol is very different on this panel

I am getting a lot of disconnects when viewing the log to capture the output.  I'm running an ESP32 so I figured it would be fast enough but maybe between ESPHome trying to connect and the board trying to listen things are getting dropped.  I don't have a logic analyzer.  Any other thoughts on how to capture the output besides the ESPHome web interface in HaaS?

Dilbert66, 2021-05-13:
That's not normal.  There isnt that much data coming in normally. There can't be that much data missed.. There is definitively something afoot here Watson! lol.   The modded code i gave you should have picked up any unknown packets.  Instead the traffic you show really doesnt make sense. For example.  A 98 cmd should look like this:

```
16:19:17][I][CMD:350]: 98 2E 40 20 F1 B4 00 00 00 00 00 00 
[16:19:18][I][CMD:350]: 98 2E 80 20 F1 74 00 00 00 00 00 00 
```

Yours looks like this.  This is totally wrong. So it's more than just a few lost bytes, it's every byte is incorrect.
```
[20:21:13][I][CMD:347]: 98 66 7E 00 78 60 00 00 00 00 00 00
[20:21:13][I][CMD:347]: 9E 86 E6 E0 86 00 00 00 00 00 00 00
```

Dilbert66, 2021-05-13:
I can't see how the address would be a factor as it's only needed when the keypad needs to write to the bus, otherwise it just sits idly capturing data.  Who knows I guess.  Only way to know is to change the address and see what happens.

Dilbert66, 2021-05-13:
I wonder if it's a baud rate issue... hmmm  You can always try different baud rates to see if it makes any difference?
Here's the line in vista.cpp (line 21)
```
  vistaSerial->begin(4800,SWSERIAL_8E2);
```
You can try other bauds like 1200, 9600, 2400, etc

jn3va, 2021-05-15:
No joy.   Validated the wiring of the board, opto, resistors - all correct.  The only thing I'm using that's different is a 4N35 optocoupler, vs a 4N26.  The 35 has a higher current transfer ratio but otherwise they are the same.

I tried different bauds 300,1200, 2400, 4800, 9600 I even added some additional Parity and stop values to SWSerial and tried 8E1, 7E2, 7N1.  Saw some 7Fs but no F7s.  I decided to buy an updated panel.  Probably take me a couple of weeks to get that up and running.  When I do, I'll circle back and hook the board up and go from there.  Thanks for your help and great work on this!

jn3va, 2021-05-15:
p.s. In a post above you mentioned the bytes look wrong.  Do you have an example of what the bytes should look like?

Dilbert66, 2021-05-16:
That 's so bizarre about that board.  It sure has my curiosity peaked as to why it's different.     As to the wrong bytes, look at the cmd 98 I posted above. It shows typical vs your values.  A 98 in my case always starts with 98 2e and that never changes.  Your 98's were just random bytes the code picked up from the stream of bytes your panel sent.  Which is why this tells me there's a comm discrepency somewhere.   The protocol would need to be close to the same for the keypad to work in my opinion.   

Anyhow, don't throw that board out!

jn3va, 2021-05-31:
New panel, success!  I'm now getting F7s and messages, so must have been something with the old panel.  I am still having one issue though.  I'll open a separate issue b/c I think this might be related to the wireless sensors.

closed 2021-05-31 by jn3va

Dilbert66, 2021-09-08:
Do you still have that older SE board? I would love to analyze it.  someone reported a similar issue with a vista-20SE.  Could you ship it to me for temporary testing? I would send it back to you.  I'll pay for shipping both ways.

jn3va, 2021-09-09:
Hi Alain

Yes I still have it.  Sure you can use it.  I re-used the wireless sensor
module in the new system so its just the alarm panel main board, but you
are welcome to use it.
Let me know where to send it and I can estimate the shipping.  Do you need
the power transformer too?

Joe

Dilbert66, 2021-09-09:
Hi Joe, can you email me at ***@***.***  I will then give you my
details.
Alain