Dilbert66/esphome-vistaECP #44: Looping while in Armed Stay mode since the last update (6/14)

AlexandreUSA, 2021-06-15:
Pretty big issue this morning. I armed my system last night via HA (armed stay) and went to sleep. Did not realized that the system was looping with the timer going from 60sec to 0 then back to 60sec forever. LCD on all the time, and then impossible to disarmed the alarm this morning. I went to the panel, disconnected the ESP32, and then the system armed and I was able to disarmed with my keypad. I later reconnected the ESP32, all good until I "armed stay" again, same issue, looping on the exit timer. I then used HA Vistaalarm disconnect button to be able to disarm. It seems that the system become unavailable multiple times during the night. 

[image]

```
[07:27:14][D][text_sensor:015]: 'vistaalarm Line1': Sending state 'ARMED ***STAY***'
[07:27:14][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  60'
[07:27:14][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:14][I][INFO:586]: Prompt: May Exit Now  60
[07:27:14][I][INFO:587]: Beeps: 3

[07:27:14][D][text_sensor:015]: 'vistaalarm Beeps': Sending state '3'
[07:27:14][D][text_sensor:015]: 'vistaalarm System Status': Sending state 'unavailable'
[07:27:14][D][binary_sensor:036]: 'vistaalarm Ready': Sending state OFF
[07:27:14][I][CMD:351]: F7 00 00 04 00 60 00 80 08 02 00 00 
[07:27:14][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:14][I][INFO:586]: Prompt: May Exit Now  60
[07:27:14][I][INFO:587]: Beeps: 0

[07:27:14][D][text_sensor:015]: 'vistaalarm Beeps': Sending state '0'
[07:27:14][I][CMD:351]: F8 59 12 5A 03 80 00 05 F0 00 00 00 
[07:27:14][I][CMD:351]: F2 17 06 00 00 00 00 61 63 02 45 43 
[07:27:15][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:27:15][I][CMD:351]: F2 12 06 00 00 00 00 62 6C 02 45 6C 
[07:27:15][I][CMD:351]: F2 17 06 00 00 00 00 63 63 02 45 43 
[07:27:15][I][CMD:351]: F7 00 00 0F 10 59 00 80 08 02 00 00 
[07:27:15][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  59'
[07:27:15][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:15][I][INFO:586]: Prompt: May Exit Now  59
[07:27:15][I][INFO:587]: Beeps: 0

[07:27:15][I][CMD:351]: F9 C3 02 53 EF 00 00 00 00 00 00 00 
[07:27:15][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:27:15][I][CMD:351]: F2 12 06 00 00 00 00 64 6C 02 45 6C 
[07:27:16][I][CMD:351]: F7 00 00 0F 10 58 00 80 08 02 00 00 
[07:27:16][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  58'
[07:27:16][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:16][I][INFO:586]: Prompt: May Exit Now  58
[07:27:16][I][INFO:587]: Beeps: 0

[07:27:16][I][CMD:351]: F2 17 06 00 00 00 00 65 63 02 45 43 
[07:27:16][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:27:16][I][CMD:351]: F2 12 06 00 00 00 00 66 6C 02 45 6C 
[07:27:16][I][CMD:351]: F2 17 06 00 00 00 00 67 63 02 45 43 
[07:27:16][I][CMD:351]: F7 00 00 0F 10 58 00 80 08 02 00 00 
[07:27:16][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:16][I][INFO:586]: Prompt: May Exit Now  58
[07:27:16][I][INFO:587]: Beeps: 0

[07:27:17][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:27:17][I][CMD:351]: F2 12 06 00 00 00 00 60 6C 02 45 6C 
[07:27:17][I][CMD:351]: F7 00 00 0F 10 57 00 80 08 02 00 00 
[07:27:17][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  57'
[07:27:17][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:17][I][INFO:586]: Prompt: May Exit Now  57
[07:27:17][I][INFO:587]: Beeps: 0

[07:27:17][I][CMD:351]: F2 17 06 00 00 00 00 61 63 02 45 43 
[07:27:17][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:27:17][I][CMD:351]: F2 12 06 00 00 00 00 62 6C 02 45 6C 
[07:27:18][I][CMD:351]: F2 17 06 00 00 00 00 63 63 02 45 43 
[07:27:18][I][CMD:351]: F7 00 00 0F 10 56 00 80 08 02 00 00 
[07:27:18][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  56'
[07:27:18][I][INFO:585]: Prompt: ARMED ***STAY***
[07:27:18][I][INFO:586]: Prompt: May Exit Now  56
[07:27:18][I][INFO:587]: Beeps: 0

[07:27:18][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:27:18][I][CMD:351]: F2 12 06 00 00 00 00 64 6C 02 45 6C 
[07:27:18][I][CMD:351]: F7 00 00 0F 10 56 00 80 08 02 00 00 
[... 197 lines cut ...]
[07:30:14][I][CMD:351]: F7 00 00 0F 10 08 00 8C 08 02 00 00 
[07:30:14][D][text_sensor:015]: 'vistaalarm Line2': Sending state '                '
[07:30:14][I][INFO:585]: Prompt: ARMED ***STAY***
[07:30:14][I][INFO:586]: Prompt:                 
[07:30:14][I][INFO:587]: Beeps: 0

[07:30:14][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:30:14][I][CMD:351]: F2 12 06 00 00 00 00 67 6C 02 45 6C 
[07:30:14][I][CMD:351]: F7 00 00 0F 10 60 00 80 08 02 00 00 
[07:30:14][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  60'
[07:30:14][I][INFO:585]: Prompt: ARMED ***STAY***
[07:30:14][I][INFO:586]: Prompt: May Exit Now  60
[07:30:14][I][INFO:587]: Beeps: 0

[07:30:14][I][CMD:351]: F2 17 06 00 00 00 00 60 63 02 45 43 
[07:30:14][I][CMD:351]: F6 13 00 00 00 00 00 00 00 00 00 00 
[07:30:14][I][CMD:351]: F2 12 06 00 00 00 00 61 6C 02 45 6C 
[07:30:15][I][CMD:351]: F2 17 06 00 00 00 00 62 63 02 45 43 
[07:30:15][I][CMD:351]: F7 00 00 0F 10 59 00 80 08 02 00 00 
[07:30:15][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'May Exit Now  59'
[07:30:15][I][INFO:585]: Prompt: ARMED ***STAY***
[07:30:15][I][INFO:586]: Prompt: May Exit Now  59
[07:30:15][I][INFO:587]: Beeps: 0

```

Dilbert66, 2021-06-15:
ahh. ok, I know the issue.  A text trigger was  incorrect. Pushed a fix.

AlexandreUSA, 2021-06-15:
That was fast! Updated with your latest push (6/15) and all seems to be back to normal. No more loops. Thank you,

closed 2021-06-21 by Dilbert66