Dilbert66/esphome-vistaECP #45: missing vista.h?

finnishcolumbia, 2021-06-19:
Hi, I tried compiling the ademcoECP.ino sketch but I get a no such file or directory error for "vista.h". I searched the zip file downloaded from your github but I don't see it.

Dilbert66, 2021-06-19:
You need to also copy the *.h and *.cpp files from directory /src/vistaEcpInterface.  That's where the include file is.

finnishcolumbia, 2021-06-20:
Thanks Alain. I see it now.

closed 2021-06-21 by Dilbert66

Dilbert66, 2021-06-22:
FYI, there was a bug in the mqtt exampled that prevented it from compiling. I've uploaded a corrected copy.

finnishcolumbia, 2021-06-22:
Thanks. I'll try it.

finnishcolumbia, 2021-06-24:
Hi,

The code compiles and connects to the panel and I get data on the serial
monitor.

I am using a free mqtt broker and a push client on my phone.

Messages are appearing on my phone but when I send a Stay command, the
panel does not respond.

I see an S on my broker and S appears in the serial monitor but the panel
does not arm and I see a long stream of data instead.

I tried changing the keypad address to 17 and 18 (I have two keypads)

Any ideas?

Thanks

Dilbert66, 2021-06-24:
That's my fault. When I made a change with the mqtt handling, I missed adding some cmds. I've pushed a fixed version to master.

Dilbert66, 2021-06-24:
As to keypads, make sure the keypad id you use is also enabled in your panel and that this keypad id is NOT  the same as an existing keypad.

finnishcolumbia, 2021-06-25:
Hi Alain,

I tried the new code but no luck. I'm sending an "S" to arm the panel. Here
is the message that arrives and the response in the serial monitor.

thanks
Dale

Prompt1:****DISARMED****
Prompt2:  Ready to Arm
mqtt_callback - message arrived - topic [vista/Set/Cmd] payload
[SisplayLine/2  Ready to Arm  Status]
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 5E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: F6 14 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F7 00 00 3F 10 08 00 1C 08 02 00 00 2A 2A
Prompt1:****DISARMED****
Prompt2:  Ready to Arm
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F6 14 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 5E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F6 14 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 5E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 5E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 14 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: F6 14 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: 98 2E 04 20 80 61 00 00 00 00 00 00 00 00
CMD: 98 5E 40 25 F1 AF 00 00 00 00 00 00 00 00
EXT: 98 0C F1 00 00 E3 00 00 00 00 00 00 00 00
CMD: 9E 02 20 80 63 00 00 00 00 00 00 00 00 00
CMD: 98 5E 04 20 F1 F0 00 00 00 00 00 00 00 00
CMD: F6 13 00 00 00 00 00 00 00 00 00 00 00 00
CMD: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 00
CMD: F6 14 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F6 10 00 00 00 00 00 00 00 00 00 00 00 00
CMD: F7 00 00 3F 10 08 00 1C 08 02 00 00 2A 2A
Prompt1:****DISARMED****
Prompt2:  Ready to Arm
CMD: F7 00 00 3F 10 08 00 1C 08 02 00 00 2A 2A
Prompt1:****DISARMED****
Prompt2:  Ready to Arm
CMD: F7 00 00 3F 10 08 00 1C 08 02 00 00 2A 2A
Prompt1:****DISARMED****
Prompt2:  Ready to Arm

Dilbert66, 2021-06-25:
Did you set the accessCode value to a user code that can arm the system? To send single value cmds, the accessCode needs to be set.

Alternatively, you can send direct keypad cmds to the panel using this cmd.  For example to arm in stay mode use cmd sequence: 
!xxxx3 instead of S.  xxxx is your user access code.  The ! is a flag to indicate a direct key sequence.

finnishcolumbia, 2021-07-05:
Hi Alain,

I was using the wrong optocoupler. Everything is good now.

Thanks very much for your work!

Dale