Dilbert66/esphome-vistaECP #49: Full isolation schematic

aselle, 2021-07-14:
First, this project is awesome! Thank you. Second, it looks like you are using the optoisolator as a level shifter, but maybe not only, given there are multiple writers to green on the bus. Shouldn't green only be driven when the microcontroller wants to write, because otherwise if another keypad tries to write to green it will cause a short? In any case, I tested your circuit as is, and it worked, but I was thinking of making a PCB along these lines:

[image]

Likely, just to be safe, I'll make it so you can select with a jumper if you want isolated grounds and populate either the resistor dividers or the optoisolators...something like this:

[image]

Thoughts?

aselle, 2021-07-14:
Maybe adding a diode to prevent high voltages from outside from trying to reverse bias the emitter side would help? 
[image]

Dilbert66, 2021-07-14:
The green line is  already at level zero at rest and gets pulled to 12 volts when active .   I don't get why you think there would be short if another device drives the green line to 12 volts?   If another device drives the green line high, then the transistor just won't turn on which is fine.  In a working system, no two devices drive the bus at the same time.  The only time this would occur if a user set two keypads, devices to the same address by mistake and in those cases, communications with those devices would fail due to data corruption but it would not hurt anything.

aselle, 2021-07-14:
Yes agree that one opto isolator and circuit as is is the simplest option. I was thinking of the point of view that I don't really know how the other devices are interacting with the bus exactly. Like if the other ones used a totem pole output with a npn on high and pnp on low-side, and I did too, then if one drove at 12v and one drove at 0v, then there would be effectively be a short. But common sense would say they couldn't be using a totem pole on the other devices, or else nothing would work. I'm guessing this bus is electrically simpler, with maybe the main panel which is acting as master to poll the keypads having the only low-side driver of the receive pin or just using a passive pull-down resistor to pull things down to zero.  Like in the opto isolator you have, the optoisolators transistors can pull green to 12v, but there is nothing pulling it down to 0v. And it seems like the diode in my 2nd schematic attachment probably could be considered redundant as the transistor is acting as a diode already  (except it maybe isolates my pull-down resistor to not add more pull-down effect to any of the existing devices that may already have their own pulldowns).

aselle, 2021-07-16:
In the spirit of over-engineering to the extreme, I switched side of the output transisters (giving inverted output), ran them through schmidt trigger inverting buffers to cleanup incoming signal. Then on the green output side I added a inverting buffer that has an output enable input. The output enable comes 3.3v out of the esp8266 and then is run through another optoisolator to get it to be 0-12v. Clearly the firmware would have to be changed to provide this output. I haven't tested this yet, and I'm not sure it's worth the extra pain, but I will likely build this circuit, and tune the values of the resistors, etc.

[image]

Dilbert66, 2021-07-16:
Wow, that' s quite a circuit. I don't see why the need for a greenout enable line though?   As to cleaning up the signal, I also have not seen this as an issue since the bus is pretty clean when viewed on a scope. It is very low speed at 4800 baud.  Of course it won't hurt.  For a DIYer though i definitively see this as overkill and overwhelming for most users but if this software was to be used in a commercially available product, of course, it would need to have the type of engineering you suggest to ensure a safe and issue free product. 

Your first circuit showing isolation for all lines is defintively a good alternate diy'er option and I would also definively recommend a separate ground as the best option to provide complete isolation.

Quite a few users are using the basic circuit and I have not heard of any real issues regarding interface so far. Doesnt mean they don't exist of course.  It would definitively have been helpful to have a copy of the panel board schematic but those are not available that I can see.

aselle, 2021-07-17:
Agree that it is overkill. greenout enable allows me to use a buffer that will drive the output to 0v faster. The problem with the optoisolator is it rounds the move from 12v logic high to 0v. But, that feature means that it is safe in the setting of a bus with other keypads. Basically, as you had it the transistor in the optocoupler would only connect to the power source when the mcu was trying to drive a logic high. When it was logic low (also at idol when it wasn't the esp8266's turn to be driving), then other keypads could be driving green high and low. There must be a pull-down somewhere. It has to be not too stiff, or else it can't pull to zero fast enough for 4800 baud, but not too low resistance or else it will load the driver too much (see my attached simulation).  The downside of my simple optocoupler circuit is that the optocoupler moves slow, so the pulldown resistor has to be set properly toward the MCU. Your original circuit doesn't have that downside, because it only uses an optocoupler on the output, and clearly the existing pull-down resistor on the main vista board probably is properly sized. I'll probably build all three and look at it on the scope and report back.
[image]

Dilbert66, 2021-07-17:
Yes, the panel already has a measured passive 1kohm resistance so adding another 1kohm in parallel (as in the emitter of the optocoupler) is not necessary or recommended which is why I don't use one. 

Well, enjoy your experimentation.  That's what it is all about.

aselle, 2021-07-18:
So I built up the circuit, and it works decently well. It uses 4 optocouplers (4n25s) , a cm4503bcn tri-state buffer to drive the 12v green line, a 74hct inverting schmitt triggered buffer to cleanup the 3.3v inputs (green loopback and yellow).  Here's my oscilloscope traces from bench testing of serial output. It cleans it up, but it does require modifying you code to have an enable signal.  I'm going to try it hooked up to my system shortly. Would you be willing to accept a patch to your code that allows an enable pin to be defined (the internals of the ecp soft serial routing already supported this but it just needs to be marshaled through esphome)?

Dilbert66, 2021-07-19:
If you don't mind, I'd rather not add this type of change to the code at thist time as your circuit is not something that i see being used by most users.  Unless I see this becoming a must have change then I will re-think this.  For now, the current circuit works perfectly fine as is and I'd rather not add more complexity for no real added benefit at this point.    I sample at mid pulse and a slower rise time is not an issue as far as I have seen especially at 4800 baud.

aselle, 2021-07-19:
I certainly understand, although I think in for most users, the presence of an enable pin really doesn't hurt them either as it could be safely ignored (unles someone is using the same esp8266 for more stuff, which is possible). Most functionality was already in the underlying libraries (someone added at some point for some reason). Here's my patch: https://github.com/aselle/esphome-vistaECP/commit/9d23408e507a68e380f434bffe5e098a5ca06204

Probably the sweet spot for improvement to the circuit is full opto-isolation, minus the additional logic chips. That would also not require any software change at all.

Dilbert66, 2021-07-19:
I will definitively add a full opto-isolation example schematic as an alternative option for users as  i agree that is the better approach. I will let the users choose the approach they want to use.   I will keep your patch in mind.

aselle, 2021-07-19:
I'm happy to prepare an isolation schematic and submit it as PR w/o the
logic chips.. probably next weekend, once I build and test it.
-A

Dilbert66, 2021-07-19:
> I'm happy to prepare an isolation schematic and submit it as PR w/o the logic chips.. probably next weekend, once I build and test it. -A
> […](#)

Thank you but there is no need as I have one I put one together that I'm testing now.

aselle, 2021-07-19:
Sure, it makes sense. I was also planning to make a PCB design which I'd be
happy to prepare as gerbers so people could order easily on oshpark or
jlcpcb.

Dilbert66, 2021-07-19:
Sure, go ahead.

aselle, 2021-07-29:
My more complicated circuit above has some errors in it. I can post it if anybody is interested (or I will put it in my fork). It mainly changed since I needed to change the pins to avoid the enable pin being manipulated on start (only a few pins don't change when esp8266 turns on, and all of them were already used). It was fairly stable. 

I have been using for a couple weeks and it works well. The final turned out to be more like this (breadboard prototype and pcb prototype):
[image]

However, as we have discussed it is too complicated, so I went back to just adding isolation in both directions to your original schematic. Here is the result:
[image]

I have tested this circuit using the values of the resistors given here, and it seems okish, but the logic highs only go up to about 8.4V when green out is driven.
[image]
(yellow is output from esp8266 and purple is green on 12V side).  The resistance between green and ground on my panel is about 1.1k  ohm (similar to what you had). Simulation suggests 7.5V or so is expected. In fact, I tried a  4n25 instead and it was as low as 4V. So people's mileage may very.  The only thing to do would be to drive the LED on the opto isolator harder by reducing R1 to say 170 ohm or lower, but 210 resistor is already sourcing  10.5mA from esp8266 which is close to the 12mA limit. Maybe since the duty cycle of writing is so low, we could get away with more. Right way is probably to put a transistor  in to drive the optoisolator led.

Anyways, circuit looks like this on breadboard, probably similar to yours. I'll make a PCB version of this soon.
[image]

Dilbert66, 2021-09-27:
I experimented with various optocouplers and found the initial recommendaton of 4n25/4n26 totally inadequate for this application.   The CTR is way too low.    I tried  a couple others I had on hand, the TLP521 and the 4N35.  The TLP521 was the best of the lot.  The 4N35 is a close second.  Both produce plenty of output using minimal forward current so are easily driven from the ESP.  

I also found using too low a resistance on the panel bus led side caused too much load on the bus and drove the signal voltage down.  The minumum that can be used without affecting the bus was 4.7K.    In order to have enough signal on the collector side of the optocoupler, a higher pull down value was needed to be at least 2.2k.  I used 4.7K to simplify the circuit.  

 By using a 4N35 or TLP521, a value of 220 for R1 is plenty sufficient to get full bus signal output. I used 180 in the schematic to account for varying device CTR's and temparature.

Dilbert66, 2021-10-11:
Excellent analysis .  I like the idea of pre-biasing the base of the optocoupler transistor.  I'll try that out.  I couldnt do that with the tlp521 as it does not expose the base but for the 4nxx devices, that works.   I did try using a 10k resistor for the optocoupler led side to further minimize ECP bus load but I found I was not getting optimum output and losing data so I went back to 4.7k and as you noted there that is not perfect either as it induces a voltage sag.  Your design of having a transistor booster stage in front definitively is a good  fix but it goes against my idea of keeping the circuit as simple and workable as possible.  A dilemma.

aselle, 2021-11-07:
I received and assembled a pcb for the latest circuit and built it and made a 3d printed enclosure. Definitely overkill, but it does package it pretty nicely. It could clearly be even smaller if I used surface mount, but I went of ease of assembly. A few problems with the pcb
* Component designators are wrong for some of the resistors.
* The transistor footprint has the pads very close together. It is hand-solderable, but way to easy to get bridges.
* Probably couldn't hurt to have a clearance slot between the 12v and 3.3v sides. I have a pretty large creepage  distance, which "should" be sufficient for two low voltage things, but hey.
I'll probably fix those before sharing it in my fork.

[image]

bobhavlin, 2022-04-06:
@aselle Is there somewhere to order your PCB?  Nice work!

alanlekah, 2022-07-09:
Really clean PCB and case @aselle , would be nice to order

blacktirion, 2022-07-20:
Nice design @aselle ! Any chance we can get a PCB Schematic and parts list for this?

appleguru, 2022-07-20:
+1, if you share the files @aselle I’d be happy to do a run of these (including cases) for anyone interested

lorenzodeveloper, 2022-11-03:
Hi!
Any updated on the schematic? Does anyone have the files to use for ordering a pcb or pcba? Thank you!

aselle, 2023-02-06:
Sorry I missed your requests @blacktirion, @appleguru , @lorenzodeveloper 

@appleguru ,  I have added the files and a short description on my fork master branch
https://github.com/aselle/esphome-vistaECP
It has the gerber files, the case stl files for 3d printing, and a pdf schematic, so you don't need kicad to look at things.

Thanks @Dilbert66 for including an isolated variant as well, and thanks for all your work supporting this project.

blacktirion, 2023-02-06:
@aselle amazing work. I’ll see if I can source parts and put this together.

adamurban, 2023-02-06:
Same, will also take a look at building a batch of these!

appleguru, 2023-02-06:
Thoughts on adding an isolated dc to dc converter to this circuit so that a separate power supply isn't needed? Am thinking something like the ROE-1205S: https://www.digikey.com/en/products/detail/recom-power/ROE-1205S/5684100

Dilbert66, 2023-02-06:
@aselle, I'll add your design and associated files to this repository if you want. It's a good design. A pcb is definitively the best option for this circuit.  The CTR is not as critical due to transistor input stage and I suspect most common optocouplers can be used with no issue.

lorenzodeveloper, 2023-02-06:
Awesome! Thanks.

@Dilbert66 , let's update the readme file in dev branch too.
Some weeks ago I added a link also to a PCB layout made by howardchen3.

```
Thanks to @howardchen3 ( https://community.home-assistant.io/u/howardchen3 ) there is a PCB layout of the ground-isolated version available at oshwlab ( https://oshwlab.com/boobeechen/vistaesphome_copy ) that could be used to send to a manufacturer. Read more on the site.
```

Dilbert66, 2023-02-06:
@lorenzodeveloper ,  I do prefer @aselle's isolated version as it provides very minimal loading on the bus (which is an issue with the current simpler isolated version I provide)  and also provides high gain for the signal to the ESP.   It basically eliminates all the issues at the added cost of slightly more components and complexity for the casual builder.  Having the gerber files and availability of a board will make this a much easier and reliable process. 

@aselle, If you don't mind I'd like to modify your design to use on my DSC implementation as well.  It will need a slight modification though as the DSC tx out is inverted and  requires the green line to be brought to black (ground)  (instead of green to red (12v)) so for that circuit the green line will go to pin 5 of U2 and black to pin4.

Dilbert66, 2023-02-07:
@aselle, FYI, your gerber files are for you older prototype version so you might want to either update them or remove them.  Since you provide the pcb design, it's easy enough to generate new ones.  Nice work though on the design.  I ldo like the breakout pins for expansion.  I've modifed it to work on the DSC system as I've noted above and ordered some pcb's of  for both versions to play with.

FYI, the design will work perfectly well with the ESP32 version of the D1 mini as it has the same footprint available.   For that one, you'll need to use IO22 for Yellow in, IO21 for Green out and IO18 for Green in.

infnorm, 2023-02-08:
@Dilbert66, feel free to include the design in the mainline and modify it as you see fit. 

The gerber files in the circuit/ directory were indeed old, so I removed them and pushed it to my branch. The other ones in the v1-manufacture-files should be correct.

Glad the ESP32 has a d1 mini. I haven't done a ton with those as still have a sizable stash of esp8266's. Hope the prototype boards come back well. I changed my schematic resistor values to the ones that I used (as far as I could read the color codes from my photograph). But I didn't populate the 1M (you can see it empty in my photo above). Let me know how the bring up goes.

infnorm, 2023-02-08:
@appleguru, I chose not to run the esp8266 off the power supply, as I think the vista boards power supply and regulators tend to already run hot. It can be even worse if you have a expander, radio device, etc. Dilbert66 may have further insight.

aselle, 2023-02-08:
@Dilbert66  feel free to include the design in the mainline and modify it as you see fit.

The gerber files in the circuit/ directory were indeed old, so I removed them and pushed it to my branch. The other ones in the v1-manufacture-files should be correct.

Glad the ESP32 has a d1 mini. I haven't done a ton with those as still have a sizable stash of esp8266's. Hope the prototype boards come back well. I changed my schematic resistor values to the ones that I used (as far as I could read the color codes from my photograph). But I didn't populate the 1M (you can see it empty in my photo above). Let me know how the bring up goes.

aselle, 2023-02-08:
@appleguru, I chose not to run the esp8266 off the power supply, as I think the vista boards power supply and regulators tend to already run hot. It can be even worse if you have a expander, radio device, etc. Dilbert66 may have further insight.