Dilbert66/esphome-vistaECP #50: Missing RF messages on smoke alarm trigger

mrbubble62, 2021-07-16:
I am not seeing RFX messages when smoke alarms are triggered with the test button and the panel is monitoring the loop
If the panel is not monitoring loop 1 then the RFX message is sent OK.
E.G
Panel monitoring zone 32 address 0055590 Loop 1;

```
[08:44:06][I][CMD:352]: F7 00 00 03 10 08 00 1C 08 02 00 00 
[08:44:06][I][INFO:581]: Prompt: ****DISARMED****
[08:44:06][I][INFO:582]: Prompt:   Ready to Arm  
[08:44:06][I][INFO:583]: Beeps: 0
[08:44:06]
[08:44:06][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[08:44:12][I][CMD:352]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 
[08:44:12][I][CMD:352]: F9 83 02 53 2F 00 00 00 00 00 00 00 
[08:44:12][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[08:44:12][I][CMD:352]: F7 00 00 03 10 32 00 31 0A 02 00 00 
[08:44:12][D][text_sensor:015]: 'vistaalarm Line1': Sending state 'FIRE 32  BACK   '
[08:44:12][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'BASEMENT SMOKE  '
[08:44:12][I][INFO:581]: Prompt: FIRE 32  BACK   
[08:44:12][I][INFO:582]: Prompt: BASEMENT SMOKE  
[08:44:12][I][INFO:583]: Beeps: 0
[08:44:12]
[08:44:12][D][info:740]: ac=1,batt status = 0,systemflag=0,lightbat status=0,trouble=0
[08:44:12][D][text_sensor:015]: 'vistaalarm System Status': Sending state 'triggered'
[08:44:12][D][binary_sensor:036]: 'vistaalarm Fire': Sending state ON
[08:44:12][I][EXT:352]: F9 00 02 00 54 80 D9 26 B0 7D C3 04 
[08:44:12][D][info:740]: ac=1,batt status = 0,systemflag=0,lightbat status=0,trouble=0
[08:44:12][I][CMD:352]: F2 12 06 00 00 00 00 64 6C 02 45 6C 
[08:44:12][D][info:740]: ac=1,batt status = 0,systemflag=0,lightbat status=0,trouble=0
[08:44:13][I][CMD:352]: F8 59 1C 5A 03 21 00 01 00 00 00 00 

```
Panel monitoring zone 33 address 0855728 Loop 1;
```

[16:10:51]
[16:10:51][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[16:10:51][I][CMD:352]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 
[16:10:51][I][CMD:352]: F9 43 02 53 6F 00 00 00 00 00 00 00 
[16:10:51][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[16:10:52][I][CMD:352]: F7 00 00 03 10 33 00 31 0A 02 00 00 
[16:10:52][D][text_sensor:015]: 'vistaalarm Line1': Sending state 'FIRE 33         '
[16:10:52][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'BASEMENT SMOKE  '
[16:10:52][I][INFO:581]: Prompt: FIRE 33         
[16:10:52][I][INFO:582]: Prompt: BASEMENT SMOKE  
[16:10:52][I][INFO:583]: Beeps: 0
[16:10:52]

```
Panel monitoring zone 32 address 0055590 loop 4 (tamper);
```
[08:36:19]
[08:36:19][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[08:36:20][I][CMD:352]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 
[08:36:20][I][EXT:352]: 9E 04 00 D9 26 B0 00 00 00 00 00 00 
[08:36:20][D][info:553]: RFX: 0055590,B0
[08:36:20][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '0055590,B0'
[08:36:21][I][CMD:352]: F9 43 02 53 6F 00 00 00 00 00 00 00 
[08:36:21][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[08:36:21][I][EXT:352]: F9 00 02 83 04 00 00 00 79 7D 00 00 
[08:36:21][D][info:740]: ac=1,batt status = 0,systemflag=1,lightbat status=0,trouble=0
[08:36:26][I][CMD:352]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 
[08:36:26][I][EXT:352]: 9E 04 00 D9 26 B0 00 00 00 00 00 00 
[08:36:26][D][info:553]: RFX: 0055590,B0
[08:36:26][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '0055590,B0'
[08:36:27][I][CMD:352]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 
[08:36:27][I][EXT:352]: 9E 04 00 D9 26 00 00 00 00 00 00 00 
[08:36:27][D][info:553]: RFX: 0055590,00
[08:36:27][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '0055590,00'
[08:36:28][I][CMD:352]: F7 00 00 03 10 08 00 1C 08 02 00 00 
[08:36:28][I][INFO:581]: Prompt: ****DISARMED****
[08:36:28][I][INFO:582]: Prompt:   Ready to Arm  
[08:36:28][I][INFO:583]: Beeps: 0
[08:36:28]
```

I am wondering why there is no EXT decode for RF when the alarm is triggered.

Panel Vista 15P v10.24, Keypad 6162RF, RF repeater 5800RP, Smoke 5808W3

Dilbert66, 2021-07-16:
It means that the program is not seeing  an expected 9e packet response on the bus. If it captures one you will see a line such as this one below:
[08:36:26][I][EXT:352]: 9E 04 00 D9 26 B0 00 00 00 00 00 00 .  
It's possible that the packet is a different structure then the normal response.  The current code ignores those.  i will post a modified version that will log those as well.  I don't have RF modules to test with so I can't examine the traffic myself.

Dilbert66, 2021-07-16:
i've pushed an updated version to the "DEV" branch.  You can just update the file vista.cpp only as that's the only one changed. Try it out and and post more logs. This one will also display any unknown 9e response packets.

mrbubble62, 2021-07-16:
Thank you, I made the change however I don't see any additional 9E packets
Note; the 9E 02 25 F1 ED is not related, it occurs every ~10 minutes.
```
[11:59:40]
[11:59:42][I][CMD:351]: F9 03 02 53 AF 00 00 00 00 00 00 00 00 
[11:59:42][I][EXT:351]: F9 00 02 43 04 00 00 00 B9 03 00 00 00 
[11:59:50][I][CMD:351]: F7 00 00 03 10 08 00 1C 08 02 00 00 2A 
[11:59:50][I][INFO:585]: Prompt: ****DISARMED****
[11:59:50][I][INFO:586]: Prompt:   Ready to Arm  
[11:59:50][I][INFO:587]: Beeps: 0
[11:59:50]
[11:59:55][I][CMD:351]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 00 
[11:59:56][I][CMD:351]: F9 43 02 53 6F 00 00 00 00 00 00 00 00 
[11:59:56][I][CMD:351]: F7 00 00 03 10 33 00 31 0A 02 00 00 46 
[11:59:56][D][text_sensor:015]: 'vistaalarm Line1': Sending state 'FIRE 33         '
[11:59:56][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'BASEMENT SMOKE  '
[11:59:56][I][INFO:585]: Prompt: FIRE 33         
[11:59:56][I][INFO:586]: Prompt: BASEMENT SMOKE  
[11:59:56][I][INFO:587]: Beeps: 0
[11:59:56]
[11:59:56][D][text_sensor:015]: 'vistaalarm System Status': Sending state 'triggered'
[11:59:56][D][binary_sensor:036]: 'vistaalarm Fire': Sending state ON
[11:59:56][I][EXT:351]: F9 00 02 00 51 8D 0E B0 B0 B4 83 04 00 
[11:59:56][I][CMD:351]: F8 59 1C 5A 03 21 00 02 00 00 00 00 00 

```

Dilbert66, 2021-07-16:
Ok, that's interesting. I would need to see logic analyzer traces of the bus to know for sure what happens then when the RF sensor is triggered.  Obviously, we're missing a cmd response somewhere.   On the assumption that it could be a checksum issue on the cmd, I've pushed an other version to dev that will also display 9e responses that fail a checksum. Otherwise, I can't know for sure.

mrbubble62, 2021-07-16:
I already uncommented the DEBUG define so I'd expect RF checksum failed messages.

Dilbert66, 2021-07-16:
Fyi,You will only see that checksum error msg if you monitor via the serial
interface

mrbubble62, 2021-07-16:
I am monitoring via serial.  I created a logic analyzer trace in Sigrok PulseView  and see a correlated signal on the green wire.
I don't know how to decode it. I added .zip on the end of the export to allow upload here. PulseView has a lot of export formats let me know if you need a different format. 

[smoke_trigger.sr.zip](https://github.com/Dilbert66/esphome-vistaECP/files/6832636/smoke_trigger.sr.zip)

Dilbert66, 2021-07-16:
Great, I do use both sigrok and salae. I'll have a look.

Dilbert66, 2021-07-16:
[image]

This is what I've decoded above.  If possble could you run a trace that also includes the following f7 . Also here' the session setup I use that includes the decoders for the channels.
[vistaecp.pvs.zip](https://github.com/Dilbert66/esphome-vistaECP/files/6832935/vistaecp.pvs.zip)

Dilbert66, 2021-07-16:
Just noticed your channels are opposite of mine. Where it says panel, that's the keypads/rf and whre it says keypad, that's the panel. Reverse the naming.

Dilbert66, 2021-07-16:
So far I don't see anything wrong with that 9E response. It's the right format and the chksum adds up.

Dilbert66, 2021-07-16:
> Thank you, I made the change however I don't see any additional 9E packets
> Note; the 9E 02 25 F1 ED is not related, it occurs every ~10 minutes.

Actually it is related.  The panel can request RF status periodically or if the RF receive has a device that has data to send, it will set it's address on the bus (seen as pulses), then the panel will send a 9E cmd to have the device send it's data update.  Seems in this case for some reason, the firmware is either not capturing or processing the response data even though it looks correct from the trace you sent.   I need more examples of a captured EXT 9E and a missed one so I can compare.   Of course, it would have been easier if I had the same hardware setup but unfortunately, I never picked up some RF hardware for this system.  My main home setup is a DSC 5020.  I wrote this firmware just using an old vista panel and keypad I had lying around.

Dilbert66, 2021-07-16:
Ok, I see the problem. I went back to your previous logs and can see that my receive routine is capturing both an f9 and the FE reponse in the same session in one spot.  That's usually due to having a very short gap between both cmds when receiving. I seen the issue sometime with the f6 cmd.

08:44:12][I][EXT:352]: F9 00 02 00 54 80 D9 26 B0 7D C3 04 

The 00 54 80 d9 26 b0 7D  is an FE cmd response (showing up in an f9).  

Looks like i have to fine tune the receive routine.   If you can post more examples that would be great.

mrbubble62, 2021-07-16:
Correct, the serial for that device is 0855728.
Here's a longer capture. I think the F7 is present.

[image]
I wonder if the FE is actually the fire alarm trigger, the 9E 02 20 F1 F2 maybe to silence other radios.

[smoke_trigger2.sr.zip](https://github.com/Dilbert66/esphome-vistaECP/files/6833531/smoke_trigger2.sr.zip)

Dilbert66, 2021-07-16:
I've pushed a small change to 'vista.cpp' on dev branch. Just prioritized handling of green line rx. Try it out and let me know.

Dilbert66, 2021-07-16:
Thank you. Actually, the 9E is simply the panel's cmd to the RF module to go ahead and send it's data . 

> I wonder if the FE is actually the fire alarm trigger, the 9E 02 20 F1 F2 maybe to silence other radios.

Dilbert66, 2021-07-16:
> Correct, the serial for that device is 0855728.
> Here's a longer capture. I think the F7 is present.
>
> [image]
> I wonder if the FE is actually the fire alarm trigger, the 9E 02 20 F1 F2 maybe to [...]

No, that's bogus data.  That's a sync type pulse the panel sends all the time on which the modules can use to send their requests to send address on.

Dilbert66, 2021-07-16:
> Correct, the serial for that device is 0855728.
> Here's a longer capture. I think the F7 is present.
>
> [image]
> I wonder if the FE is actually the fire alarm trigger, the 9E 02 20 F1 F2 maybe to [...]

Actually, that's the RF module setting it's request to send adress on the panel sync pulse.  RF modules have an address of 0 so basically it's sets one pulse.   The panel, will then use that address to see what is requesting to send then send the appropriate cmd (9e,98, etc) to have the device then go ahead and send it's data.

Dilbert66, 2021-07-16:
Question, are you not seeing F2 cmds? I see it in your trace but I didnt see those in your logs?

mrbubble62, 2021-07-16:
Your last change worked!
```
[17:55:54][I][CMD:351]: 9E 02 25 81 5D 00 00 00 00 00 00 00 00 
[17:55:54][I][EXT:351]: 00 21 00 DF 00 39 E2 00 00 00 00 00 00 
[17:55:55][I][CMD:351]: F7 00 00 03 10 08 00 1C 08 02 00 00 2A 
[17:55:55][I][INFO:585]: Prompt: ****DISARMED****
[17:55:55][I][INFO:586]: Prompt:   Ready to Arm  
[17:55:55][I][INFO:587]: Beeps: 0
[17:55:55]
[17:55:55][I][CMD:351]: 9E 02 20 82 61 00 00 00 00 00 00 00 00 
[17:55:55][I][EXT:351]: 00 24 03 D9 00 39 E2 00 00 00 00 00 00 
[17:55:57][I][CMD:351]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 00 
[17:55:58][I][EXT:351]: 9E 04 0D 0E B0 B0 00 00 00 00 00 00 00 
[17:55:58][D][info:557]: RFX: 0855728,B0
[17:55:58][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '0855728,B0'
[17:55:58][I][CMD:351]: F9 C3 02 53 EF 00 00 00 00 00 00 00 00 
[17:55:58][I][EXT:351]: F9 00 02 03 04 00 00 00 F9 B4 00 00 00 
[17:55:58][I][CMD:351]: F7 00 00 03 10 33 00 31 0A 02 00 00 46 
[17:55:58][D][text_sensor:015]: 'vistaalarm Line1': Sending state 'FIRE 33         '
[17:55:58][D][text_sensor:015]: 'vistaalarm Line2': Sending state 'BASEMENT SMOKE  '
[17:55:58][I][INFO:585]: Prompt: FIRE 33         
[17:55:58][I][INFO:586]: Prompt: BASEMENT SMOKE  
[17:55:58][I][INFO:587]: Beeps: 0
[17:55:58]
```
I had been missing a few other RF messages when things got busy without seeing any checksum errors. They all seem to coming through now even when triggering multiple sensors simultaneously. 
```
[18:04:52][I][CMD:351]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 
[18:04:52][I][EXT:351]: 9E 04 0F FE 20 04 00 00 00 00 00 00 00 
[18:04:52][D][info:557]: RFX: 1048096,04
[18:04:52][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '1048096,04'
[18:04:52][I][CMD:351]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 00 
[18:04:52][I][EXT:351]: 9E 04 0C 5E 75 A0 00 00 00 00 00 00 00 
[18:04:52][D][info:557]: RFX: 0810613,A0
[18:04:52][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '0810613,A0'
[18:04:52][I][CMD:351]: 9E 02 20 F1 F2 00 00 00 00 00 00 00 00 
[18:04:53][I][EXT:351]: 9E 04 00 10 9F A0 00 00 00 00 00 00 00 
[18:04:53][D][info:557]: RFX: 0004255,A0
[18:04:53][D][text_sensor:015]: 'vistaalarm RF Msg': Sending state '0004255,A0'
[18:04:53][I][CMD:351]: 9E 02 25 F1 ED 00 00 00 00 00 00 00 00 
[18:04:53][I][EXT:351]: 9E 04 0F C4 28 A0 00 00 00 00 00 00 00 
[18:04:53][D][info:557]: RFX: 1033256,A0
```

Dilbert66, 2021-07-16:
Awesome. Tks for the help debugging!

mrbubble62, 2021-07-16:
Thank you for sharing and supporting this great project!

closed 2021-07-16 by mrbubble62