Dilbert66/esphome-vistaECP #52: Building, but not booting.

knifesk, 2021-08-20:
Hey guys, I had to rebuild my HomeAssistant instance and finally I decided to give ESPHome a Try... this is like the 10th time I rebuild the YAML, folder, wipe, clean and try again with the same results... 

I have the NodeMCU connected via USB into the home asssitant pc, and I flashed via USB.. while it's building, flashing and starting.. the latter fails 10 times pretty quick and then the chip goes into safe mode. I think the lib is crashing.. maybe it's becase new versions of arduino or ESPHome? I'm a pretty experienced developer, but C++ and low level things are not my thing... so, I can't quite make heads or tails of this.

I tried to do the least modifictaions possible and still couldn't get it to work... Here's the log of the flash/boot process.. any hints are appreciated.

```
INFO Running:  platformio run -d /config/esphome/vistaalarm
Processing vistaalarm (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/ECPSoftwareSerial.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/debug/vistaPrint.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/api_connection.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/api_server.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/list_entities.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/proto.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/user_services.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/api/util.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/logger/logger.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/restart/restart_switch.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/switch/automation.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/switch/switch.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/template/binary_sensor/template_binary_sensor.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/template/switch/template_switch.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/template/text_sensor/template_text_sensor.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/wifi/wifi_component_esp32.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/application.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/application_esp32.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/application_esp8266.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/color.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/component.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/controller.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/esphal.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/helpers.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/log.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/preferences.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/scheduler.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/esphome/core/util.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/main.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/src/vista.cpp.o
Generating LD script /data/vistaalarm/.pioenvs/vistaalarm/ld/local.eagle.app.v6.common.ld
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling /data/vistaalarm/.pioenvs/vistaalarm/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
[... 215 lines cut ...]
[21:02:22]3ffffeb0:  00000000 4021ef98 00000000 00000000  
WARNING Decoded 0x4021ef98: precache
[21:02:23]3ffffec0:  3fff27f4 3fff27f8 3fff27f8 4020e594  
WARNING Decoded 0x4020e594: std::_Function_handler<std::vector<esphome::Component*, std::allocator<esphome::Component*> > (), setup::{lambda()#3}>::_M_invoke(std::_Any_data const&) at main.cpp
[21:02:23]3ffffed0:  3fff2c1c 3fff2c20 3fff2c20 3fffff78  
[21:02:23]3ffffee0:  00000001 feefeffe feefeffe 3fff34d4  
[21:02:23]3ffffef0:  3fff345c 3fff32ac 3fff3204 3fff3154  
[21:02:23]3fffff00:  3fff23d4 3fff2f24 3fff2e9c 3fff2de4  
[21:02:23]3fffff10:  3fff2d2c 3fff2c5c 3fff2b94 3fff2acc  
[21:02:23]3fffff20:  3fff29dc 3fff2924 3fff284c 3fff277c  
[21:02:23]3fffff30:  3fff26d4 3fff262c 3fff2590 3fff24e8  
[21:02:23]3fffff40:  3fff2448 3fff2330 3fff2270 3fff21b8  
[21:02:23]3fffff50:  3fff20f8 3fff2030 3fff1f70 3fff1ec8  
[21:02:23]3fffff60:  3fff1e70 3fff1d80 3fff1cc0 3fff1c18  
[21:02:23]3fffff70:  3fff1b48 3fff1aa0 3fff3574 feefeffe  
[21:02:23]3fffff80:  3fff34d4 00000090 000000a0 feefeffe  
[21:02:23]3fffff90:  feefeffe feefeffe feefeffe 3fff05d4  
[21:02:23]3fffffa0:  3fffdad0 00000000 3fff0594 4021b174  
WARNING Decoded 0x4021b174: loop_wrapper() at core_esp8266_main.cpp
[21:02:23]3fffffb0:  feefeffe feefeffe 3ffe84f4 401009b9  
WARNING Decoded 0x401009b9: cont_wrapper
[21:02:23]<<<stack<<<
[21:02:23]
[21:02:23]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Dilbert66, 2021-08-20:
Interesting.  Might be an effect of the esphome latest version. I'll upgrade and try it out. Your debug code is helpful as it points to the getextbytes  where the error occurs. I'll look into it as well

Dilbert66, 2021-08-20:
I do see one  issue. I have a bug in my bounds checking for an arrary index (not compensating for an increment) so if the receive buffer goes beyond the initialized array, this could cause an error.  Don't think it would create the error above but it all depends on what is stored at the address above the array that is being overwritten.   Anyhow, I will fix that.

This still doesnt explain why you are getting that error since the array is large enough in normal operation not to overflow.  Perhaps you are getting spurious data. Is your wiring correct to the alarm system?   It could also be related to an issue with the 8266 itself. It works fine on the esp32 which I just tried now. I will have to switch chips and try out the esp8266 later.

knifesk, 2021-08-20:
I'm not even reaching the point of connecting the esp to the panel. I
already had everything installed and I need to remove the module from the
socket in order to flash it... I tried connecting it after it went into
safe mode, but I couldn't debug it while it's up in the cabinet. Obviously
I couldn't make it work, so I went back to the computer and tried again...
Shouldn't it boot anyway if it's not connected to the panel?

Dilbert66, 2021-08-21:
It will boot fine without panel connection. I updated my test platform to the latest epshome version and can't see any issues.
At this point I really don't have a solution for your issue.

Dilbert66, 2021-08-21:
Can you post your yaml config?

knifesk, 2021-08-21:
First thing of all: Thanks for your time man! I really appreciate you take time off your own in order to help me.

I'll attach the YAML at the end. At first I added a couple zones (11 and 12) and also removed 1...3 and 5..8 (the wired ones) since my entire setups is wireless and the only wired sensor I have isn't wired yet, but I'm planning to wire it at z4.... as things were not running at all I restored the entire yaml to the original one and tried to change as little as possible, but it still not running... This is the last version I tried before reaching for help. I'm gonna give it another try with a freshly cloned repo... I'll report back in a couple minutes..

As a sidenote: thing was working just fine with the MQTT example before trying the ESP version.. is there any chance that the flash isn't wiping entirely? I've read several times that if the entire flash isn't wiped it might cause issues.

```
#for documentation see project at https://github.com/Dilbert66/esphome-VistaECP
substitutions:
  systemName: "vistaalarm"
  accessCode: !secret access_code #Only comes into effect if needed for arming and quickarm is not set
  keypadAddr: "17" #set this to an unused keypad address slot. Make sure to enable it in  your vista panel programming

  ##esp32
  #rxPin: "22" #GPIO pin to use for data receive (yellow line) 
  #txPin: "21" #GPIO pin to use for data transmit (green line)
  #monitorPin: "18" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable

  ##esp8266
  rxPin: "5" #GPIO pin to use for data receive (yellow line) 
  txPin: "4" #GPIO pin to use for data transmit (green line)
  monitorPin: "-1" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable

  # module addresses:
  # 07 4229 zone expander  zones 9-16
  # 08 4229 zone expander zones 17-24
  # 09 4229 zone expander zones 25-32
  # 10 4229 zone expander zones 33-40
  # 11 4229 zone expander zones 41 48
  # 12 4204 relay module  
  # 13 4204 relay module
  # 14 4204 relay module
  # 15 4204 relay module

  expanderAddr1: "0" # 1st zone expander emulator (4229) address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 

  relayAddr1: "0" # relay module emulation (4204) addresses. Set to 0 to disable
  relayAddr2: "0"
  relayAddr3: "0"
  relayAddr4: "0"

  TTL: "30000"  # time  to live  in ms for zone/fire status before expiring;
  quickArm: "false"
  lrrSupervisor: "false" # set to true if we don't have an LRR monitoring supervisor we can emulate one to get the statuses

globals:
  #persistent storage variables
  - id: zoneStates  #persistant storage for zone states in case of reboot
    type: int
    restore_value: yes
  - id: zoneAlarms  #persistant storage for zone states in case of reboot
    type: int
    restore_value: yes
  - id: zoneBypass  #persistant storage for zone states in case of reboot
    type: int
    restore_value: yes
  - id: zoneChecks  #persistant storage for zone states in case of reboot
    type: int
    restore_value: yes
  - id: lrrCode  #persistant storage for last lrr message in case of reboot
    type: int
    restore_value: yes

esphome:
  name: $systemName
  #platform: ESP32
  #board: nodemcu-32s
  platform: ESP8266
  board: nodemcuv2
[... 260 lines cut ...]

  - platform: template
    id: l2
    name: "$systemName Line2"

  - platform: template
    id: beep1
    name: "$systemName Beeps"

switch:
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return vista.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectVista();
  - platform: restart
    id: restart_switch

```

Dilbert66, 2021-08-21:
Ok, i see the issue.  You have the monitor pin set at -1 to disable but yet, my code is still trying to read an invalid disabled pin.  The serial code should not allow that and technically should also my routine should not be called.   Initically the code used a #DEFINE to  enable or disable the monitor code but I switched to using a yaml configuration. 

I'll fix the code.   For now, you can set a valid pin for the monitor code. (even if you don't have it hooked up) so that the code will at least run.

Dilbert66, 2021-08-21:
Pushed corrected version.

knifesk, 2021-08-21:
Yes sir! that was it! It's booting now! Does it make any difference to have it enabled and not connected? I'm not planning to use it anyway...

I need to do some other stuff rn, but as soon as I finish with my things I'll rebuild all the YAML as I intend to use it and re-connect the module to the panel. I'll be back later! I promise I'll bring a couple pics to show off hehe :)

Thanks for your time man, you're the best. I promised a couple months ago to make some contributions to the project, I'm planning to cleanup the codebase and see if I can improve something :D

Cheers!

Dilbert66, 2021-08-21:
Best to leave it at -1 so you don't pick up errant bits.   The monitor pin is really useful for those that have RF devices as it will pick up various states from those devices.  Also on/off status will be detected immediately since the code will use the device response to decided if the sensor is off or on.

closed 2021-09-08 by Dilbert66