Dilbert66/esphome-vistaECP #53: Invalid state armed_stay for alarm_control_panel

rbauman70, 2021-09-02:
Hi.  I was getting the error:
```
Logger: homeassistant.components.template.alarm_control_panel
Source: components/template/alarm_control_panel.py:210
Integration: Template (documentation, issues)
First occurred: 9:13:22 AM (1 occurrences)
Last logged: 9:13:22 AM

Received invalid alarm panel state: armed_stay. Expected: armed_away, armed_home, armed_night, arming, disarmed, pending, triggered, unavailable
```

As a workaround, I replaced armed_stay with armed_home in the value_template:

`value_template: "{{ states('sensor.vistaalarm_system_status') | replace('armed_stay', 'armed_home')}}"
`
Using HA core-2021.8.8

knifesk, 2021-09-12:
+1 on this. It worked for me

Dilbert66, 2021-09-12:
Alternatively you can change line 50 in vistalarm.h to be armed_home instead of armed_stay 👍 
```
  const char* const STATUS_STAY = "armed_home";
```

closed 2021-10-04 by Dilbert66