Dilbert66/esphome-vistaECP #58: Issue with releases after November 7th, 2021

AlexandreUSA, 2021-11-17:
I can't get my system to work when I upload all the new files for source code released recently. My last update was with the November 7th release and I had no issue. I upload the whole "vistaEcpInterface" folder as I did not change any of the files so the updates are pretty simple. I went back to my old folder from November 7th and the system is back up and running. 

The new code is compiling normally and the vistaalarm log is showing in Home Assistant. The system status is unavailable and nothing is shown in the virtual keypad.

Not sure where to start to find out the issue.

Dilbert66, 2021-11-17:
Interesting.. Ok, let me try in case I messed up the push.

Dilbert66, 2021-11-17:
Are you using an esp32 or 8266?

Dilbert66, 2021-11-17:
Well, I just did a clean install , recompiled using esp32 profile (which I believe you are using) and everything is fine here.
Why don't you try cleaning the build director or doing a clean build files and try again. Sometimes building without clean causes issues.

AlexandreUSA, 2021-11-17:
Let me try the clean install. I am using the esp32 and the Master branch.

Dilbert66, 2021-11-17:
i see the issue. My fault. When I pushed dev changes to master, I messed up a file. I'll correct it. I was using dev to test.

AlexandreUSA, 2021-11-17:
Great. Let me know when I should try again. The clean install did not work so I am looking for your new push.

Dilbert66, 2021-11-17:
ok, update files push to master now.

AlexandreUSA, 2021-11-17:
All good now. Thanks a lot for the quick fix!

FYI, running well on ESP32, and the latest versions of Home Assistant 2021.11.4 and ESPhome 2021.11.1 as of today (11/17/2021)

Dilbert66, 2021-11-17:
Excellent!

closed 2021-11-17 by Dilbert66