Dilbert66/esphome-vistaECP #59: Phantom Alarm Triggers

CodingSquirrel, 2021-12-05:
A couple days ago I pulled the latest from master and noticed that I'm now occasionally getting phantom alarm triggers. I believe the previous version I was using was around c475d71edfc0402689626619a02d215261ed30a3 since I'm sure I had the change that adjusted the `HITSTAR` value. Not sure which change might have caused this but I hadn't seen this issue until now.

The underlying cause appears to be a corrupt message. I looked back at my previous issue (for adding vista SE) and noticed an edit where you think garbage values could be a result of wiring. I'm currently using [this schematic](https://user-images.githubusercontent.com/326106/127437446-710d8afe-08d8-4ab1-b0c0-0023607f4ca5.png) (with a 200 ohm resistor instead of 210), which is a bit different than the ground isolated one you put up. Although I'd hesitate to do any wiring changes since it's soldered now.

I managed to grab the log output the most recent time it happened.

```
[23:39:28][D][text_sensor:067]: 'vistaalarm Zone 13': Sending state 'C'
[23:39:29][D][text_sensor:067]: 'vistaalarm Zone 01': Sending state 'C'
[23:39:30][D][text_sensor:067]: 'vistaalarm Line1': Sending state 'ARMED ***STAY***'
[23:39:32][I][CMD:430]: 02 00 8C 08 00 00 00 00 00 00 00 00 00 
[23:39:34][I][CMD:430]: FE 20 20 20 20 00 00 00 00 00 00 00 00 
[23:39:34][I][INFO:658]: Prompt: ARMED ***STAY***
[23:39:34][I][INFO:659]: Prompt:                 
[23:39:34][I][INFO:660]: Beeps: 0

[23:39:36][D][text_sensor:067]: 'vistaalarm Beeps': Sending state '0'
[23:39:36][D][text_sensor:067]: 'vistaalarm Den window': Sending state 'C'
[23:39:38][D][text_sensor:067]: 'vistaalarm Living room window': Sending state 'C'
[23:39:38][D][text_sensor:067]: 'vistaalarm Side door': Sending state 'C'
[23:39:42][D][text_sensor:067]: 'vistaalarm Zone 09': Sending state 'C'
[23:39:43][D][text_sensor:067]: 'vistaalarm Line1': Sending state 'xMEx~**STAY***'
[23:39:43][I][INFO:658]: Prompt: xMEx~**STAY***
[23:39:43][I][INFO:659]: Prompt:                 
[23:39:43][I][INFO:660]: Beeps: 0

[23:39:43][D][text_sensor:067]: 'vistaalarm System Status': Sending state 'triggered'
[23:39:43][D][binary_sensor:036]: 'vistaalarm Alarm': Sending state ON
[23:39:43][D][binary_sensor:036]: 'vistaalarm Chime': Sending state ON
[23:39:43][D][binary_sensor:036]: 'vistaalarm Away': Sending state ON
[23:39:43][D][binary_sensor:036]: 'vistaalarm Stay': Sending state OFF
[23:39:43][D][binary_sensor:036]: 'vistaalarm Bypass': Sending state ON
[23:39:44][I][EXT:430]: F0 31 00 00 00 00 00 00 00 00 00 00 71 
[23:39:44][I][EXT:430]: F0 00 00 00 DF 00 00 00 00 00 00 00 00 
[23:39:48][D][debug:547]: 30 second time check
[23:39:54][I][CMD:430]: 02 00 8C 08 00 00 00 00 00 00 00 00 00 
[23:39:56][I][CMD:430]: FE 20 20 20 20 00 00 00 00 00 00 00 00 
[23:39:56][D][text_sensor:067]: 'vistaalarm Line1': Sending state 'ARMED ***STAY***'
[23:39:56][I][INFO:658]: Prompt: ARMED ***STAY***
[23:39:56][I][INFO:659]: Prompt:                 
[23:39:56][I][INFO:660]: Beeps: 0

[23:39:56][D][text_sensor:067]: 'vistaalarm System Status': Sending state 'armed_home'
[23:39:56][D][binary_sensor:036]: 'vistaalarm Alarm': Sending state OFF
[23:39:56][D][binary_sensor:036]: 'vistaalarm Chime': Sending state OFF
[23:39:56][D][binary_sensor:036]: 'vistaalarm Away': Sending state OFF
[23:39:56][D][binary_sensor:036]: 'vistaalarm Stay': Sending state ON
[23:39:56][D][binary_sensor:036]: 'vistaalarm Bypass': Sending state OFF
[23:40:05][I][CMD:430]: 02 00 8C 08 00 00 00 00 00 00 00 00 00 
[23:40:07][I][CMD:430]: FE 20 20 20 20 00 00 00 00 00 00 00 00 
[23:40:07][I][INFO:658]: Prompt: ARMED ***STAY***
[23:40:07][I][INFO:659]: Prompt:                 
[23:40:07][I][INFO:660]: Beeps: 0
```

Dilbert66, 2021-12-05:
Yes, I made a few changes on the serial library because i was seeing issues on the Vista20p.  The last changes might have affected your version.  Why don't you try just the file ECPSoftwareSerial.cpp from the version that worked for you. Here's the link to that version:.  That will tell me if that last change is the culprit.
https://github.com/Dilbert66/esphome-vistaECP/tree/c475d71edfc0402689626619a02d215261ed30a3/src/vista20SE

CodingSquirrel, 2021-12-05:
I'll give that a try and update what happens.

CodingSquirrel, 2021-12-13:
After reverting the ECPSoftwareSerial.cpp I haven't encountered the issue.

Dilbert66, 2021-12-14:
Awesome. Tks for letting me know. I've made a small change in the latest master/dev versions to re-add one of the removed checks in the serial library. Hopefully that is the issue.

closed 2022-04-18 by Dilbert66