Dilbert66/esphome-vistaECP #63: Compiler warning at ECPSoftwareSerial.cpp:208

rbauman70, 2022-01-30:
Probably not what you intended.

```$ esphome compile 1320VistaAlarm.yaml

INFO Reading configuration 1320VistaAlarm.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing vistaalarm (board: nodemcuv2; framework: arduino; platform: platformi
o/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.h
tml
PLATFORM: Espressif 8266 (2.6.3) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Building in release mode
Compiling .pioenvs\vistaalarm\src\ECPSoftwareSerial.cpp.o
Compiling .pioenvs\vistaalarm\src\debug\vistaPrint.cpp.o
Compiling .pioenvs\vistaalarm\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\vistaalarm\src\esphome\components\api\api_frame_helper.cpp.o
src\ECPSoftwareSerial.cpp: In member function 'size_t SoftwareSerial::write(uint
8_t, bool, int32_t)':
src\ECPSoftwareSerial.cpp:208:20: warning: statement has no effect [-Wunused-val
ue]
         m_bitCycles==m_4800_bitCycles;
                    ^
Compiling .pioenvs\vistaalarm\src\esphome\components\api\api_pb2.cpp.o
.
.
.

Dilbert66, 2022-01-30:
You are correct! A typo! I will fix it and push an update!

Dilbert66, 2022-01-30:
Correction pushed.

closed 2022-04-10 by Dilbert66