Dilbert66/esphome-vistaECP #64: Zone type 90 alarm does not trigger Alarm status

rbauman70, 2022-01-30:
I'm still digging into this but I thought you might be able to look at the hex and know what's happening.  Let me know if you need more debug info.

I've got my Kidde smoke/CO detectors, hard-wired, connected via relay to a 5816 RF transmitter's loop 1 N.C. input.  In the Vista 20P, I've got its zone 39 configured as type 90, configurable zone, set to alarm but not sound the horn or keypad tones.  There's enough noise going on when the detectors are set off.   The configuration for 90 is ten bytes: 0101010000.  The panel indicates "ALARM 39 SMOKE DETECTOR" but the binary sensors alarm and system_status become unavailable.  Here's the ESPHome device log snippet.  At 18:06:27 I open the RF transmitter's  loop 1.

```[18:06:12][I][CMD:441]: FA 01 02 25 F7 E7 00 00 00 00 00 00 00
[18:06:13][D][text_sensor:067]: 'vistaalarm 1st bedroom 2': Sending state 'C'
[18:06:15][D][text_sensor:067]: 'vistaalarm 2nd bedroom 3': Sending state 'C'
[18:06:16][D][text_sensor:067]: 'vistaalarm Line1': Sending state '****DISARMED*
***'
[18:06:19][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[18:06:19][I][INFO:687]: Prompt: ****DISARMED****
[18:06:19][I][INFO:688]: Prompt:   Ready to Arm
[18:06:19][I][INFO:689]: Beeps: 0

[18:06:19][D][text_sensor:067]: 'vistaalarm Patio motion sensor': Sending state
'C'
[18:06:23][D][text_sensor:067]: 'vistaalarm Den window': Sending state 'C'
[18:06:23][D][text_sensor:067]: 'vistaalarm 2nd bedroom 2': Sending state 'C'
[18:06:24][D][text_sensor:067]: 'vistaalarm 3rd bathroom window': Sending state
'C'
[18:06:26][D][text_sensor:067]: 'vistaalarm Garage inside door': Sending state '
C'
[18:06:27][I][CMD:441]: FB 02 20 F1 F2 00 00 00 00 00 00 00 00
[18:06:27][I][EXT:441]: FB 04 07 01 D7 A0 00 00 00 00 00 00 00
[18:06:27][D][info:659]: RFX: 0459223,A0
[18:06:27][D][text_sensor:067]: 'vistaalarm RF Msg': Sending state '0459223,A0'
[18:06:27][I][CMD:441]: F8 59 1C 5A 03 00 00 02 00 00 00 00 77
[18:06:27][I][CMD:441]: F2 12 06 00 00 00 00 65 6C 02 45 6C F5
[18:06:27][I][CMD:441]: F2 16 06 00 00 00 00 66 63 02 45 43 F5
[18:06:27][I][CMD:441]: F8 59 10 5A 00 03 0A 0A 00 00 00 00 77
[18:06:28][D][text_sensor:067]: 'vistaalarm 1st bedroom 1': Sending state 'C'
[18:06:28][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[18:06:28][D][text_sensor:067]: 'vistaalarm Line1': Sending state 'ALARM 39 SMOK
E  '
[18:06:28][D][text_sensor:067]: 'vistaalarm Line2': Sending state 'DETECTOR
   '
[18:06:28][I][INFO:687]: Prompt: ALARM 39 SMOKE
[18:06:28][I][INFO:688]: Prompt: DETECTOR
[18:06:28][I][INFO:689]: Beeps: 0

[18:06:28][D][text_sensor:067]: 'vistaalarm Smoke/CO detectors': Sending state '
A'
[18:06:28][D][text_sensor:067]: 'vistaalarm System Status': Sending state 'unava
ilable'
[18:06:28][D][binary_sensor:036]: 'vistaalarm Ready': Sending state OFF
[18:06:28][D][text_sensor:067]: 'vistaalarm Garage overhead door': Sending state
 'C'
[18:06:28][D][text_sensor:067]: 'vistaalarm Smoke/CO detectors': Sending state '
A'
[18:06:31][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[18:06:31][I][INFO:687]: Prompt: ALARM 39 SMOKE
[18:06:31][I][INFO:688]: Prompt: DETECTOR
[18:06:31][I][INFO:689]: Beeps: 0

[18:06:32][D][text_sensor:067]: 'vistaalarm Line2': Sending state 'DETECTOR
   '
[18:06:34][D][text_sensor:067]: 'vistaalarm 2nd bedroom 4': Sending state 'C'
[18:06:34][D][text_sensor:067]: 'vistaalarm Right sliding door': Sending state '
C'
[18:06:35][D][text_sensor:067]: 'vistaalarm 3rd bedroom 2': Sending state 'C'
[18:06:35][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[18:06:35][I][INFO:687]: Prompt: ALARM 39 SMOKE
[18:06:35][I][INFO:688]: Prompt: DETECTOR
[18:06:35][I][INFO:689]: Beeps: 0

[18:06:36][I][CMD:441]: FB 02 25 81 5D 00 00 00 00 00 00 00 00
[18:06:36][I][EXT:441]: 00 21 00 DF 00 00 00 00 00 00 00 00 74
[18:06:37][I][CMD:441]: FB 02 20 82 61 00 00 00 00 00 00 00 00
[18:06:37][I][EXT:441]: 00 24 05 D7 00 00 00 00 00 00 00 00 74
[18:06:39][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[18:06:39][I][INFO:687]: Prompt: ALARM 39 SMOKE
[18:06:39][I][INFO:688]: Prompt: DETECTOR
[18:06:39][I][INFO:689]: Beeps: 0

[18:06:43][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[18:06:43][I][INFO:687]: Prompt: ALARM 39 SMOKE
[18:06:43][I][INFO:688]: Prompt: DETECTOR
[18:06:43][I][INFO:689]: Beeps: 0

[18:06:44][I][CMD:441]: FA 01 02 20 F7 EC 00 00 00 00 00 00 00
[18:06:47][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[18:06:47][I][INFO:687]: Prompt: ALARM 39 SMOKE
[18:06:47][I][INFO:688]: Prompt: DETECTOR
[18:06:47][I][INFO:689]: Beeps: 0

Dilbert66, 2022-01-30:
The system will be unavailable because it's not ready anymore since you have an active sensor.  Also I see the smoke text sensor sending out an "A" which means zone in Alarm so I'm not sure what you see as the problem? Looks to me like it's sending out the correct sense data.

[18:06:28][D][text_sensor:067]: 'vistaalarm Smoke/CO detectors': Sending state 'A'

rbauman70, 2022-01-30:
Maybe I'm not understanding what constitutes an alarm.  If I set the zone type to 08 (auxiliary alarm) and trigger the sensor, the same message, "ALARM 39 SMOKE DETECTOR" is displayed but also an alternating audio alarm is played on the keypad (not desired).  However the binary sensor alarm (binary_sensor.vistaalarm_alarm) changes to on.  I use that binary sensor to trigger my home automation actions.   If I set a zone type of 90 with the behavior set to alarm, I don't get the the binary sensor alarm set to on.  It goes to unavailable.

Dilbert66, 2022-01-30:
Can you display a log with the zone type set to 08 so I can compare the status bits. It's possible I'm not interpreting a bit correctly. I never tested a smoke detector config that thoroughly.

Dilbert66, 2022-01-30:
FYI, the vistaalarm_alarm is triggered by a specific bit in the status from the panel.  So if you set it to not alarm audibly, that bit wont be set so the only way to know the zone is triggered is by the zone status that goes to A (there are separate bits for zone status and system status.  The vistaalarm alarm status is system wide.  What type of status would you want to see?

rbauman70, 2022-01-30:
Here's a log with the zone type set to 08.  I'm not using any smoke detector-specific options with the Vista 20P.  The RF transmitter's zone entry determines the panel's behavior.  For zone type 90 I can set the abnormal condition to normal, alarm, trouble or fault.

```[21:47:14][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[21:47:14][I][INFO:687]: Prompt: ****DISARMED****
[21:47:14][I][INFO:688]: Prompt:   Ready to Arm
[21:47:14][I][INFO:689]: Beeps: 0

[21:47:17][I][CMD:441]: FB 02 20 81 62 00 00 00 00 00 00 00 00
[21:47:17][I][EXT:441]: 00 24 00 DC 00 00 00 00 00 00 00 00 74
[21:47:18][I][EXT:441]: 00 24 00 00 21 05 DA 00 00 00 00 00 74
[21:47:21][D][text_sensor:067]: 'vistaalarm 3rd bedroom 2': Sending state 'C'
[21:47:22][D][text_sensor:067]: 'vistaalarm Laundry window': Sending state 'C'
[21:47:23][D][text_sensor:067]: 'vistaalarm Line1': Sending state '****DISARMED*
***'
[21:47:24][D][text_sensor:067]: 'vistaalarm Garage inside door': Sending state '
C'
[21:47:24][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[21:47:24][I][INFO:687]: Prompt: ****DISARMED****
[21:47:24][I][INFO:688]: Prompt:   Ready to Arm
[21:47:24][I][INFO:689]: Beeps: 0

[21:47:24][D][text_sensor:067]: 'vistaalarm Left sliding door': Sending state 'C
'
[21:47:25][I][CMD:441]: FA 01 02 20 F7 EC 00 00 00 00 00 00 00
[21:47:25][D][text_sensor:067]: 'vistaalarm Den window': Sending state 'C'
[21:47:25][D][text_sensor:067]: 'vistaalarm 1st bedroom 3': Sending state 'C'
[21:47:29][D][text_sensor:067]: 'vistaalarm Smoke/CO detectors': Sending state '
C'
[21:47:29][D][text_sensor:067]: 'vistaalarm Guest bathroom window': Sending stat
e 'C'
[21:47:31][D][text_sensor:067]: 'vistaalarm Right sliding door': Sending state '
C'
[21:47:33][I][CMD:441]: F2 00 00 00 00 00 00 00 00 00 00 00 77
[21:47:33][I][EXT:441]: 00 54 87 01 D7 A0 AD 00 00 00 00 00 71
[21:47:33][I][CMD:441]: F2 12 06 00 00 00 00 66 6C 02 45 6C F5
[21:47:33][D][text_sensor:067]: 'vistaalarm RF Msg': Sending state '0022603,04'
[21:47:33][I][CMD:441]: F7 00 00 FF 10 39 00 00 0B 02 00 00 41
[21:47:33][D][text_sensor:067]: 'vistaalarm Line1': Sending state 'ALARM 39 SMOK
E  '
[21:47:33][D][text_sensor:067]: 'vistaalarm Line2': Sending state 'DETECTOR
   '
[21:47:33][I][INFO:687]: Prompt: ALARM 39 SMOKE
[21:47:33][I][INFO:688]: Prompt: DETECTOR
[21:47:33][I][INFO:689]: Beeps: 0

[21:47:33][D][text_sensor:067]: 'vistaalarm Smoke/CO detectors': Sending state '
A'
[21:47:33][D][text_sensor:067]: 'vistaalarm System Status': Sending state 'trigg
ered'
[21:47:33][D][binary_sensor:036]: 'vistaalarm Alarm': Sending state ON
[21:47:33][D][binary_sensor:036]: 'vistaalarm Ready': Sending state OFF
[21:47:33][I][CMD:441]: F8 59 1C 5A 03 00 03 02 00 00 00 00 77
[21:47:33][I][CMD:441]: F2 16 06 00 00 00 00 67 63 02 45 43 F5
[21:47:33][I][CMD:441]: F8 59 10 5A 00 03 0A 0A 00 00 00 00 77
[21:47:35][D][text_sensor:067]: 'vistaalarm 3rd bathroom window': Sending state
'C'
[21:47:35][I][CMD:441]: FB 02 25 F1 ED 00 00 00 00 00 00 00 00
[21:47:35][I][EXT:441]: FB 04 07 01 D7 20 00 00 00 00 00 00 00
[21:47:35][D][info:659]: RFX: 0459223,20
[21:47:35][D][text_sensor:067]: 'vistaalarm RF Msg': Sending state '0459223,20'
[21:47:35][I][CMD:441]: F8 59 10 5A 00 03 0A 0A 00 00 00 00 77
[21:47:36][I][CMD:441]: F2 12 06 00 00 00 00 60 6C 02 45 6C F5
[21:47:36][I][CMD:441]: F7 00 00 FF 10 39 00 00 0B 02 00 00 41
[21:47:36][I][INFO:687]: Prompt: ALARM 39 SMOKE
[21:47:36][I][INFO:688]: Prompt: DETECTOR
[21:47:36][I][INFO:689]: Beeps: 0

[21:47:36][I][CMD:441]: F2 16 06 00 00 00 00 61 63 02 45 43 F5
[21:47:36][D][text_sensor:067]: 'vistaalarm 1st bathroom window': Sending state
'C'
[21:47:36][D][text_sensor:067]: 'vistaalarm Garage overhead door': Sending state
 'C'
[21:47:38][D][Debug:408]: Writing keys: 1
[21:47:38][D][text_sensor:067]: 'vistaalarm Beeps': Sending state '0'
[21:47:38][I][CMD:441]: F6 12 00 00 00 00 00 00 00 00 00 00 00
[21:47:38][D][text_sensor:067]: 'vistaalarm 2nd bedroom 2': Sending state 'C'
[21:47:38][I][CMD:441]: F7 00 00 FF 10 39 00 00 0B 02 00 00 41
[21:47:38][I][INFO:687]: Prompt: ALARM 39 SMOKE
[21:47:38][I][INFO:688]: Prompt: DETECTOR
[21:47:38][I][INFO:689]: Beeps: 0

Dilbert66, 2022-01-30:
Ok, i see the issue. it's the alarm bit logic.  There are 2 bits. I was looking at the system bit for the alarm trigger and ignoring the zone status bit.   Can you tell me if your panel ready light is on when the smoke detector is triggered either way?

Dilbert66, 2022-01-30:
Ok, I pushed an updated. You only need to update vistaalarm.h.

rbauman70, 2022-01-31:
I can confirm that the panel ready light goes off when the smoke alarm zone is triggered, either with zone type 08 or 90.
The binary_sensor.vistaalarm_alarm is toggling quite a few times now with this new fix and remains in the on state after clearing the sensor and the panel notifications.  
```[14:43:37][I][EXT:441]: FA 07 F7 00 DF 00 00 00 00 00 00 00 00
[14:43:38][I][CMD:441]: FB 02 20 60 83 00 00 00 00 00 00 00 00
[14:43:38][I][EXT:441]: 00 24 00 DC 00 00 00 00 00 00 00 00 74
[14:43:40][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[14:43:40][I][INFO:687]: Prompt: ****DISARMED****
[14:43:40][I][INFO:688]: Prompt:   Ready to Arm
[14:43:40][I][INFO:689]: Beeps: 0

[14:43:45][I][CMD:441]: FA 01 02 25 F7 E7 00 00 00 00 00 00 00
[14:43:50][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[14:43:50][I][INFO:687]: Prompt: ****DISARMED****
[14:43:50][I][INFO:688]: Prompt:   Ready to Arm
[14:43:50][I][INFO:689]: Beeps: 0

[14:43:58][D][text_sensor:067]: 'vistaalarm Smoke/CO detectors': Sending state '
C'
[14:44:00][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[14:44:00][I][INFO:687]: Prompt: ****DISARMED****
[14:44:00][I][INFO:688]: Prompt:   Ready to Arm
[14:44:00][I][INFO:689]: Beeps: 0

[14:44:00][D][text_sensor:067]: 'vistaalarm 1st bedroom 2': Sending state 'C'
[14:44:03][D][text_sensor:067]: 'vistaalarm Laundry window': Sending state 'C'
[14:44:05][D][text_sensor:067]: 'vistaalarm 2nd bathroom window': Sending state
'C'
[14:44:05][D][text_sensor:067]: 'vistaalarm Left sliding door': Sending state 'C
'
[14:44:06][D][text_sensor:067]: 'vistaalarm 3rd bathroom window': Sending state
'C'
[14:44:07][D][text_sensor:067]: 'vistaalarm 2nd bedroom 4': Sending state 'C'
[14:44:07][D][text_sensor:067]: 'vistaalarm 1st bathroom window': Sending state
'C'
[14:44:09][D][text_sensor:067]: 'vistaalarm 1st bedroom 1': Sending state 'C'
[14:44:09][I][CMD:441]: FB 02 25 81 5D 00 00 00 00 00 00 00 00
[14:44:09][I][EXT:441]: 00 21 00 DF 00 00 00 00 00 00 00 00 74
[14:44:10][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[14:44:10][I][INFO:687]: Prompt: ****DISARMED****
[14:44:10][I][INFO:688]: Prompt:   Ready to Arm
[14:44:10][I][INFO:689]: Beeps: 0

[14:44:10][I][CMD:441]: FB 02 20 82 61 00 00 00 00 00 00 00 00
[14:44:10][I][EXT:441]: 00 24 05 D7 00 00 00 00 00 00 00 00 74
[14:44:11][D][text_sensor:067]: 'vistaalarm 3rd bedroom 2': Sending state 'C'
[14:44:11][D][text_sensor:067]: 'vistaalarm 3rd bedroom 1': Sending state 'C'
[14:44:11][D][text_sensor:067]: 'vistaalarm Beeps': Sending state '0'
[14:44:11][D][text_sensor:067]: 'vistaalarm 1st bedroom 4': Sending state 'C'
[14:44:12][D][text_sensor:067]: 'vistaalarm 2nd bedroom 3': Sending state 'C'
[14:44:12][D][text_sensor:067]: 'vistaalarm Line2': Sending state '  Ready to Ar
m  '
[14:44:13][D][text_sensor:067]: 'vistaalarm RF Msg': Sending state '0459223,20'
[14:44:13][D][text_sensor:067]: 'vistaalarm Guest bathroom window': Sending stat
e 'C'
[14:44:13][D][text_sensor:067]: 'vistaalarm Right sliding door': Sending state '
C'
[14:44:13][D][text_sensor:067]: 'vistaalarm Patio motion sensor': Sending state
'C'
[14:44:15][I][EXT:441]: 00 24 05 00 51 87 01 D7 A0 B0 00 00 74
[14:44:15][I][CMD:441]: F8 59 1C 5A 03 00 00 02 00 00 00 00 77
[14:44:15][D][text_sensor:067]: 'vistaalarm Garage overhead door': Sending state
 'C'
[14:44:15][I][CMD:441]: F2 12 06 00 00 00 00 61 6C 02 45 6C F5
[14:44:15][I][CMD:441]: F2 16 06 00 00 00 00 62 63 02 45 43 F5
[14:44:15][D][binary_sensor:036]: 'vistaalarm Alarm': Sending state OFF
[14:44:15][I][CMD:441]: F7 00 00 FF 10 39 00 00 0A 02 00 00 41
[14:44:15][D][text_sensor:067]: 'vistaalarm Line1': Sending state 'ALARM 39 SMOK
E  '
[14:44:15][D][text_sensor:067]: 'vistaalarm Line2': Sending state 'DETECTOR
   '
[... 191 lines cut ...]
[14:45:14][I][CMD:441]: FB 02 25 82 5C 00 00 00 00 00 00 00 00
[14:45:14][I][EXT:441]: 00 21 05 DA 00 00 00 00 00 00 00 00 74
[14:45:15][D][text_sensor:067]: 'vistaalarm Garage overhead door': Sending state
 'C'
[14:45:19][D][text_sensor:067]: 'vistaalarm 1st bedroom 3': Sending state 'C'
[14:45:21][I][CMD:441]: FA 01 02 20 F7 EC 00 00 00 00 00 00 00
[14:45:21][D][text_sensor:067]: 'vistaalarm 2nd bedroom 2': Sending state 'C'
[14:45:22][D][text_sensor:067]: 'vistaalarm System Status': Sending state 'disar
med'
[14:45:23][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[14:45:23][I][INFO:687]: Prompt: ****DISARMED****
[14:45:23][I][INFO:688]: Prompt:   Ready to Arm
[14:45:23][I][INFO:689]: Beeps: 0

[14:45:23][D][text_sensor:067]: 'vistaalarm Front door': Sending state 'C'
[14:45:26][D][text_sensor:067]: 'vistaalarm Den window': Sending state 'C'
[14:45:26][D][text_sensor:067]: 'vistaalarm 2nd bedroom 1': Sending state 'C'
[14:45:26][D][text_sensor:067]: 'vistaalarm Line1': Sending state '****DISARMED*
***'
[14:45:27][D][text_sensor:067]: 'vistaalarm Garage inside door': Sending state '
C'
[14:45:33][I][CMD:441]: F7 00 00 FF 10 08 00 1C 08 02 00 00 2A
[14:45:33][I][INFO:687]: Prompt: ****DISARMED****
[14:45:33][I][INFO:688]: Prompt:   Ready to Arm
[14:45:33][I][INFO:689]: Beeps: 0

Dilbert66, 2022-01-31:
Yes, I have to redo the alarm processing logic. The problem here is that once you disable the alarm, the panel no longer sends out the alarm status so there is no alarm off flag, so I will need use the TTL logic to timeout the zone alarm status when it no longer sees the smoke zone update.  The flag for the audible alarm is a system flag and is always present but the zone status only get sent on a zone update.  The vista panel is strange that way.

Dilbert66, 2022-01-31:
ok, i've added the timeout. Havent had time to test it though..

rbauman70, 2022-02-01:
Wow, that was fast.  I was going to say to not worry about coding for this rather corner case where the audible alarm is disabled.  For the hopefully rare instances where the smoke detectors trigger, another audible alarm is not a big deal.  I can use the zone type 08 and all works fine.  The day-to-day functions of vistaECP work great.  Thanks for all the hard work.

closed 2022-04-10 by Dilbert66