Dilbert66/esphome-vistaECP #69: Battery, Alarm, Fire showing as unknown

AlexandreUSA, 2022-04-07:
Did something change in the messaging for the status of the Battery, Alarm, or Fire states? I have the "unknown" status now and I can't find a way to get back to the regular message. Everything else seems fine. Thanks.

Dilbert66, 2022-04-08:
No, the messaging did not change but the yaml configuration did. Check your config.

Dilbert66, 2022-04-08:
Ah, yes. You must have been using the binary sensors yaml. I looked and realized I had removed the extended statuses section. I re-add and update the branch.

AlexandreUSA, 2022-04-11:
Hi, I am not using the binary sensor.yaml. I am still on the stable branch with the normal vistaalarm.yaml file. I do not think there is an extended statuses section there.

Dilbert66, 2022-04-11:
Only thing that changed in master is that the system will only display system status data for the partition assigned to the script.  Any other partition system statuses will be ignored (except for zone data which will always show)

AlexandreUSA, 2022-04-18:
Updated to the 4/18/2022 version and all is working now. I also updated my vistaalam.yaml as this file had not been changed for a while on my side. Thank you.

closed 2022-04-18 by Dilbert66