Dilbert66/esphome-vistaECP #73: Keypad 6139 Macro keys

giukio, 2022-05-09:
Hello, I am using the multi-partition dev branch and I'm quite happy with it!
My system has two 6139 keypads on two partitions, and we are using its 4 macro keys to arm/disarm both partitions at once.

Is it possible to call the macro commands from home assistant?

thanks

Dilbert66, 2022-05-09:
You can't call a macro but you can send any key sequence to one are all partitions using the alarm_keypress_partition service.
The cmd to arm all is <code> + 0 + the arm code#.  Note the addition of the 0 in from of the cmd which indicates its a multi partition cmd.
To disarm all it's <code> + 0 + 1.  You can send the code the partition 1 and it should send to all.   The macros keys just do the same thing.  You can then program the virtual keypad spare buttons to do the same thing.

To bypass all open zones on all partitions, the cmd is <code> + 0 + 6 + #

Dilbert66, 2022-05-09:
Ah ok, I see what you  are doing. i had forgotten about the function key trigger buttons. You are correct.   I am developing for 2 different systems. DSC and Vista and sometimes some details get astay!    As to the custom component it's designed to be totally customizable by users so they can modify as they choose.

closed 2024-05-17 by Dilbert66