Dilbert66/esphome-vistaECP #80: Issues with sensors and keypad presses

travisrroy, 2022-09-14:
Hello Alain! Thanks for creating this project to help us smartify our older alarm systems.

I'm having some odd issues with my setup that I haven't been able to figure out. I've read through many of the other issues people have had and still haven't been able to figure it out. I've tried to attach as much information and logs about my system as possible. Don't mind the Telco Fault, I had the system disconnected at this time to hopefully prevent a false alarm. I'm sure my issues are just a mistake (or mistakes) I've made trying to set this up. 

## System

  - Monitored by ADT through POTS
  - System: Vista 20p
  - Version: 9.18 D1 (I think)
  - Model: SAVS20P5
  [image]

### Devices

  - 6150RF (Address 16. With 4 wireless contact sensors and 1 motion sensor, Zones 9-12 and 13)
  - 6160 (Address 17)
  - MH ET LIVE ESP32DevKIT running this project (Address 18)
  [image]
  
### Config
```yaml
#for documentation see project at https://github.com/Dilbert66/esphome-VistaECP
substitutions:
  systemName: "vistaalarm"
  accessCode: !secret access_code #Only comes into effect if needed for arming and quickarm is not set
  keypadAddr: "18" #set this to an unused keypad address slot. Make sure to enable it in  your vista panel programming
 
  ##esp32
  rxPin: "22" #GPIO pin to use for data receive (yellow line) 
  txPin: "21" #GPIO pin to use for data transmit (green line)
  monitorPin: "18" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  ##esp8266
  # rxPin: "5" #GPIO pin to use for data receive (yellow line) 
  # txPin: "4" #GPIO pin to use for data transmit (green line)
  # monitorPin: "14" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  # module addresses:
  # 07 4229 zone expander  zones 9-16
  # 08 4229 zone expander zones 17-24
  # 09 4229 zone expander zones 25-32
  # 10 4229 zone expander zones 33-40
  # 11 4229 zone expander zones 41 48
  # 12 4204 relay module  
  # 13 4204 relay module
  # 14 4204 relay module
  # 15 4204 relay module
  
  expanderAddr1: "07" # 1st zone expander emulator (4229) address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  relayAddr1: "0" # relay module emulation (4204) addresses. Set to 0 to disable
  relayAddr2: "0"
  relayAddr3: "0"
  relayAddr4: "0"
  
  TTL: "30000"  # time  to live  in ms for zone/fire status before expiring;
  quickArm: "false"
  lrrSupervisor: "true" # set to true if we don't have an LRR monitoring supervisor we can emulate one to get the statuses
  
globals:
       #persistent storage variables
    - id: zoneStates  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneAlarms  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneBypass  #persistant storage for zone states in case of reboot
      type: int
[... 432 lines cut ...]
```
[02:45:41][D][Debug:412]: Writing keys: 12341
[02:45:42][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:42][I][EXT:445]: F6 12 00 22 0A 01 02 03 04 01 DD 00 00 
[02:45:42][I][EXT:445]: F6 12 00 92 01 6D 02 03 04 01 DD 00 00 
[02:45:42][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:42][I][EXT:445]: F6 12 00 92 0A 01 E0 C0 70 9D 00 00 00 
[02:45:43][I][EXT:445]: F6 12 00 D2 01 2D E0 C0 70 9D 00 00 00 
[02:45:43][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:43][I][EXT:445]: F6 12 00 92 01 01 02 03 04 01 5D 00 00 
[02:45:43][I][EXT:445]: F6 12 00 12 01 ED 02 03 04 01 5D 00 00 
[02:45:44][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:44][I][EXT:445]: F6 12 00 92 0A 04 03 04 01 1D 00 00 00 
[02:45:44][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:44][I][EXT:445]: F6 12 00 12 04 40 E0 C0 70 DD 00 00 00 
[02:45:44][I][EXT:445]: F6 12 00 52 01 AD E0 C0 70 DD 00 00 00 
[02:45:45][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:45][I][EXT:445]: F6 12 00 12 02 C0 E0 C0 70 9D 00 00 00 
[02:45:45][I][EXT:445]: F6 12 00 92 01 6D E0 C0 70 9D 00 00 00 
[02:45:45][D][text_sensor:067]: 'vistaalarm Garage Door': Sending state 'C'
[02:45:45][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00 
[02:45:45][I][EXT:445]: F6 12 00 D2 01 2D 00 00 00 00 00 00 00 
[02:45:47][I][EXT:445]: F6 12 00 00 21 00 DF 00 00 00 00 00 00 
[02:45:48][I][CMD:445]: F7 00 00 07 10 94 00 10 08 02 00 00 54
```

Dilbert66, 2022-09-14:
From the look of the logs, there is something very weird going with the tx line.  Those F6 are send cmds to the panel. Looking at what you are sending, the format is incorrect.   From what I see, your esp is sending bad data to the panel continuously.   I have no idea what is causing that. 

```
[02:45:41][D][Debug:412]: Writing keys: 12341
[02:45:42][I][CMD:445]: F6 12 00 00 00 00 00 00 00 00 00 00 00  #this is the panel response to a request to send. It's saying , ok device address 18 (hex 0x12) go ahead and send your keys
[02:45:42][I][EXT:445]: F6 12 00 22 0A 01 02 03 04 01 DD 00 00  #this is your ESP sending it's keys.  The first 3 bytes are just to show that it's a response to the F6 12.  The following bytes are 22 0a 01 02 03 04 01 dd .  22 is the message id, 0a should be the message length but it is incorrect.  It should be 06 , then your 5 keys then the last byte is a checksum. I didnt validate the checksum to see if that is correct.

[02:45:42][I][EXT:445]: F6 12 00 92 01 6D 02 03 04 01 DD 00 00 # this is the following response. I have no idea where that is coming from.   I

Honestly, to me it looks like you have 2 different devices sending on address 18?  

Try changing your ESP address to another address.

```

travisrroy, 2022-09-15:
Ah! That's what it was. Enabled address 19 and changed to 19 and now it's working! 

I am still unsure about lrrSupervisor. Would you be able to clarify what that is for?

Dilbert66, 2022-09-15:
If you are using a long range radio (IP or GSM) interface and monitored by a central monitoring station , then set it to false. If you don't have this interface, then set it to true and the firmware will emulate one. It enables extra status messages sent from the panel to be captured by the firmware for display.

travisrroy, 2022-09-15:
Ok, good to know. 

Would you know how to fix all these values being set to Unknown after an OTA update? 
[image]

Dilbert66, 2022-09-15:
Try using the "dev" branch.  It has been updated to support multi partitions as well as other bug fixes.  Please note that the yaml file has changed so you will need to update it with your own config.

travisrroy, 2022-09-16:
Ah! That definitely helped. After a couple minutes, it updated all the binary sensors to their proper values.

Although, I did have to make a few changes to the single partition config to get the keypad working. I'm sure you know what needs to be changed for that or would you rather I open a pull request with those changes?

closed 2022-09-26 by travisrroy