Dilbert66/esphome-vistaECP #82: Random disconnects, from multiple times per minute to a few times per day

jakejakejakejakejakejake, 2022-10-09:
Random disconnects, from multiple times per minute to a few times per day.

`INFO Reading configuration /config/esphome/esphome-web-2af211.yaml...
INFO Detected timezone 'America/New_York'
INFO Starting log output from vistapanel-2af211.local using esphome API
INFO Successfully connected to vistapanel-2af211.local
[00:09:24][I][app:102]: ESPHome version 2022.9.4 compiled on Oct  8 2022, 23:39:47
[00:09:24][C][status_led:019]: Status LED:
[00:09:24][C][wifi:372]:   Subnet: 255.255.255.0
[00:09:24][C][wifi:374]:   DNS1: 172.16.20.1
[00:09:24][C][logger:275]: Logger:
[00:09:24][C][logger:278]:   Hardware UART: UART0
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Trouble'
[00:09:24][C][template.binary_sensor:018]:   Device Class: 'problem'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Bypass'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Away'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Armed'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Stay'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Instant'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Night'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 AC'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Chime'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Alarm'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Battery'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Fire'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Ready'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Trouble 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Bypass 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Away 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Armed 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Stay 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Instant 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Night 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 AC 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Chime 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Alarm 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Battery 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Fire 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Ready 2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Relay1'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Relay2'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Relay3'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'vistapanel-2af211 Relay4'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r5'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r6'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r7'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r8'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r9'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r10'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r11'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r12'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r13'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r14'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r15'
[00:09:24][C][template.binary_sensor:018]: Template Binary Sensor 'r16'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 1'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Side Door'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 3'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 4'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 5'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 6'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 7'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 8'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Front Door'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Back Door'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 11'
[00:09:24][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 12'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 13'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Zone 14'
[00:09:25][C][template.text_sensor:021]:   Icon: 'mdi:alert-box'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Line1'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Line2'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Line1 2'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Line2 2'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Beeps'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'vistapanel-2af211 Beeps 2'
[00:09:25][C][template.text_sensor:021]: Template Sensor 'zoneExtended'
[00:09:25][C][template.switch:058]: Template Switch 'vistapanel-2af211 Connection'
[00:09:25][C][template.switch:059]:   Restore State: NO
[00:09:25][C][restart:022]: Restart Switch 'restart_switch'
[00:09:25][C][restart:022]:   Icon: 'mdi:restart'
[00:09:25][C][mdns:100]: mDNS:
[00:09:25][C][mdns:101]:   Hostname: vistapanel-2af211
[00:09:25][C][api:138]: API Server:
[00:09:25][C][api:143]:   Using noise encryption: NO
[00:09:25][C][wifi_signal.sensor:009]: WiFi Signal 'vistapanel-2af211 WiFi'
[00:09:25][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[00:09:25][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[00:09:25][C][sntp:050]: SNTP Time:
[00:09:25][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[00:09:25][I][CMD:505]: 2022-10-09 00:09  F7 00 00 07 10 08 00 1C 28 02 00 00 20 
[00:09:25][I][INFO:813]: Display to partition: 01
[00:09:25][I][INFO:834]: Prompt:   Ready to Arm  
[00:09:25][I][CMD:505]: 2022-10-09 00:09  F7 00 00 08 00 08 00 1C 08 02 00 00 2A 
[00:09:25][I][INFO:813]: Display to partition: 02
[00:09:25][I][INFO:834]: Prompt:   Ready to Arm  
[00:09:26][I][CMD:505]: 2022-10-09 00:09  F7 00 00 10 00 08 00 1C 08 02 00 00 2A 
[00:09:26][I][INFO:813]: Display to partition: 03
[00:09:26][I][INFO:834]: Prompt:   Ready to Arm  
[00:09:26][D][text_sensor:067]: 'vistapanel-2af211 Beeps': Sending state '0'
[00:09:28][D][text_sensor:067]: 'vistapanel-2af211 Zone 13': Sending state 'C'
[00:09:30][D][text_sensor:067]: 'vistapanel-2af211 Zone 8': Sending state 'C'
INFO vistapanel-2af211.local: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 1 expected bytes
INFO vistapanel-2af211.local: Ping Failed: Error while reading data: 0 bytes read on a total of 1 expected bytes
INFO Disconnected from ESPHome API for vistapanel-2af211.local
WARNING Disconnected from API
INFO Successfully connected to vistapanel-2af211.local
[00:09:45][I][CMD:505]: 2022-10-09 00:09  F7 00 00 08 00 08 00 1C 08 02 00 00 2A 
[00:09:45][I][INFO:813]: Display to partition: 02
[00:09:45][I][CMD:505]: 2022-10-09 00:09  F7 00 00 10 00 08 00 1C 08 02 00 00 2A 
[00:09:45][I][INFO:813]: Display to partition: 03
[00:09:45][I][INFO:834]: Prompt:   Ready to Arm  
[00:09:46][D][text_sensor:067]: 'vistapanel-2af211 Line2 2': Sending state '  Ready to Arm  '
`

jakejakejakejakejakejake, 2022-10-09:
attaching yaml
```
#for documentation see project at https://github.com/Dilbert66/esphome-VistaECP
substitutions:
  systemName: "vistapanel-2af211"
  accessCode: !secret access_code #Only comes into effect if needed for arming and quickarm is not set
  
  #assign a new virtual keypad address to each active partition using programs *190 - *196
  #and enter it below.  For unused partitions, use 0 as the keypad address.
  keypadAddr1: "18" #partition 1 virtual keypad
  keypadAddr2: "19" #partition 2 virtual keypad. Set to 0 to disable
  keypadAddr3: "20" #partition 3 virtual keypad. Set to 0 to disable
 
  ##esp32
  #rxPin: "22" #GPIO pin to use for data receive (yellow line) 
  #txPin: "21" #GPIO pin to use for data transmit (green line)
  #monitorPin: "18" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  ##esp8266
  rxPin: "5" #GPIO pin to use for data receive (yellow line) 
  txPin: "4" #GPIO pin to use for data transmit (green line)
  monitorPin: "14" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  # module addresses:
  # 07 4229 zone expander  zones 9-16
  # 08 4229 zone expander zones 17-24
  # 09 4229 zone expander zones 25-32
  # 10 4229 zone expander zones 33-40
  # 11 4229 zone expander zones 41 48
  # 12 4204 relay module  
  # 13 4204 relay module
  # 14 4204 relay module
  # 15 4204 relay module
  
  expanderAddr1: "0" # 1st zone expander emulator (4229) address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 

#
# TESTED IF TOO MANY RELAY EMULATIONS ARE TOO MUCH
#
#  relayAddr1: "12" # relay module emulation (4204) addresses. Set to 0 to disable
#  relayAddr2: "13"
#  relayAddr3: "14"
#  relayAddr4: "15"
  
  relayAddr1: "12" # relay module emulation (4204) addresses. Set to 0 to disable
  relayAddr2: "0"
  relayAddr3: "0"
  relayAddr4: "0"

  TTL: "10000"  # time  to live  in ms for zone/fire status before expiring;
  quickArm: "true"
  lrrSupervisor: "false" # set to true if we don't have an LRR monitoring supervisor we can emulate one to get the statuses
  
globals:
       #persistent storage variables
    - id: zoneStates  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneAlarms  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneBypass  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes
    - id: zoneChecks  #persistant storage for zone states in case of reboot
      type: int
      restore_value: yes      
    - id: lrrCode  #persistant storage for last lrr message in case of reboot
      type: int
[... 497 lines cut ...]

  - platform: template
    id: beep1_2
    name: "$systemName Beeps 2"   

  - platform: template
    id: zoneExtended
    #name: "$systemName Zone Status" 

switch:
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return vista.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectVista();
  - platform: restart
    id: restart_switch

```

Dilbert66, 2022-10-09:
The best way really to debug what is going on  is if you have a laptop or computer that you can connect to the esp32 usb port. You can then run epshome dashboard, select log, select connected to this computer. You can then  capture the serial logs.  The esp is possibly rebooting due to a bug ssomewhere. That should give idea where it fails.  

At this point its hard to know what the issue could be.  Sounds like a firmware bug but I need more info. I don't see any issues with your config.

Dilbert66, 2022-10-09:
I'll have to setup an esp8266 to test the code again.  I usually develop with an esp32 nowadays.

Dilbert66, 2022-10-09:
Ok, looks like your getting out of memory errors.  The dev branch uses a bit more ram and the esp8266 is tight on that.   I'll try and see if I can optimise the code better.  This is one reason I've moved on to the esp32. It's a much more powerfull and versatile platform then the esp8266.

jakejakejakejakejakejake, 2022-10-09:
Acknowledged, I'll see if I can source a esp32 and move to that. Really great work!

jakejakejakejakejakejake, 2023-07-30:
Changed to esp32 a while ago, have not run out memory since.

closed 2023-07-30 by jakejakejakejakejakejake