Dilbert66/esphome-vistaECP #86: Bypassed Sensor Fault

rcmurphy, 2022-12-08:
As a user of a Vista system, I would like to be able to know when a motion sensor faults.

My understanding was that this was possible from snooping the serial number of the sensor off the ECP bus, but I could be wrong.

Dilbert66, 2022-12-08:
yes, from the FB responses (EXT lines after the FB cmds)  for the small vista system.  The problem is that if your RF system is using a vplex polling  loop, then it won't show up on the ECP bus so there is no way to get that data.

Dilbert66, 2022-12-08:
if you can post logs when such sensors activate, we can see if that data is showing or not on your system.

rcmurphy, 2022-12-08:
Sure thing! Will do so later tonight when I'm not "at work" 😄

rcmurphy, 2022-12-08:
Grabbed some during lunch.

This is of me opening and closing zone 10, (a contact sensor, because it's easier to verify, but the output should be similar):

```
[14:12:23][I][CMD:518]: 2022-12-08 14:12  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[14:12:23][I][CMD:518]: 2022-12-08 14:12  FA 01 01 20 30 F7 00 00 00 00 00 00 77 
[14:12:24][I][EXT:518]: 2022-12-08 14:12  82 87 00 00 04 04 00 00 00 00 00 00 71 
[14:12:24][I][CMD:518]: 2022-12-08 14:12  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[14:12:24][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:25][I][CMD:518]: 2022-12-08 14:12  F7 00 D5 00 F0 03 00 43 00 00 00 00 77 
[14:12:25][I][EXT:518]: 2022-12-08 14:12  F9 00 43 87 00 00 04 04 00 00 2E 00 00 
[14:12:25][I][CMD:518]: 2022-12-08 14:12  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[14:12:25][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:25][I][CMD:518]: 2022-12-08 14:12  FA 01 01 20 30 F7 00 00 00 00 00 00 77 
[14:12:26][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:26][I][CMD:518]: 2022-12-08 14:12  F7 00 9D 00 00 00 00 00 00 00 00 00 77 
[14:12:26][I][EXT:518]: 2022-12-08 14:12  F9 00 07 78 00 50 04 05 00 00 28 00 00 
[14:12:26][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:27][I][CMD:518]: 2022-12-08 14:12  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[14:12:28][I][CMD:518]: 2022-12-08 14:12  F7 00 BD 00 F0 10 00 F7 0F 00 10 00 77 
[14:12:28][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:29][I][CMD:518]: 2022-12-08 14:12  F7 00 9D 00 F0 10 00 00 00 00 00 00 77 
[14:12:29][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:29][I][CMD:518]: 2022-12-08 14:12  F7 00 D5 00 F9 55 02 52 5E 00 00 98 77 
[14:12:30][I][CMD:518]: 2022-12-08 14:12  FA 01 01 20 30 F7 00 00 00 00 00 00 77 
[14:12:31][I][CMD:518]: 2022-12-08 14:12  F7 00 9D 00 F0 11 00 F9 55 02 52 5E 77 
[14:12:31][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:32][I][CMD:518]: 2022-12-08 14:12  F7 00 CD 00 F0 11 00 F9 55 02 52 5E 77 
[14:12:32][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:33][I][CMD:518]: 2022-12-08 14:12  F7 00 9D 00 00 3E 98 00 F9 55 02 52 77 
[14:12:34][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:34][I][CMD:518]: 2022-12-08 14:12  F7 00 CD 00 F0 12 00 F9 55 02 52 5E 77 
[14:12:34][I][CMD:518]: 2022-12-08 14:12  F7 00 BD 00 00 00 00 00 00 00 00 00 77 
[14:12:35][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:35][I][CMD:518]: 2022-12-08 14:12  F7 00 9D 00 F0 12 00 F9 55 02 52 5E 77 
[14:12:35][I][CMD:518]: 2022-12-08 14:12  F7 00 D5 00 F7 0F 00 10 00 10 00 1C 77 
[14:12:36][I][CMD:518]: 2022-12-08 14:12  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[14:12:36][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:37][I][EXT:518]: 2022-12-08 14:12  F9 00 07 54 80 4E 1A 80 3D 00 00 00 00 
[14:12:37][I][CMD:518]: 2022-12-08 14:12  F7 00 BD 00 9E 01 02 20 F1 F1 00 F9 77 
[14:12:37][I][CMD:518]: 2022-12-08 14:12  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[14:12:37][I][CMD:518]: 2022-12-08 14:12  F7 0F 00 10 00 10 00 1C 38 02 00 00 48 
[14:12:37][I][INFO:826]: Display to partition: 01
[14:12:37][I][INFO:846]: Prompt: HOME DISARMED    
[14:12:37][I][INFO:847]: Prompt: BYPAS-RDY TO ARM
[14:12:37][I][INFO:848]: Beeps: 0

[14:12:38][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:38][I][CMD:518]: 2022-12-08 14:12  F7 00 D5 00 F0 13 00 00 00 00 00 00 77 
[14:12:39][I][CMD:518]: 2022-12-08 14:12  F7 00 CD 00 F9 55 02 52 5E 00 98 BE 77 
[14:12:40][I][CMD:518]: 2022-12-08 14:12  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[14:12:40][I][CMD:518]: 2022-12-08 14:12  FB B1 25 F1 EC 00 00 00 00 00 00 00 77 
[14:12:40][I][EXT:518]: 2022-12-08 14:12  07 51 80 4E 1A 00 C0 00 00 00 00 00 73 
[14:12:40][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:40][I][CMD:518]: 2022-12-08 14:12  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[14:12:41][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:41][I][CMD:518]: 2022-12-08 14:12  F7 00 CD 00 00 00 00 00 00 00 00 00 77 
[14:12:41][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:41][I][CMD:518]: 2022-12-08 14:12  F7 00 BD 00 00 3E 61 03 00 00 00 00 77 
[14:12:42][I][CMD:518]: 2022-12-08 14:12  FB 01 02 20 F1 00 00 00 00 00 00 00 77 
[14:12:42][I][EXT:518]: 2022-12-08 14:12  07 54 84 F3 5C 80 52 00 00 00 00 00 73 
[14:12:42][I][CMD:518]: 2022-12-08 14:12  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[14:12:42][I][CMD:518]: 2022-12-08 14:12  F7 0F 00 10 00 10 03 00 38 02 00 00 46 
[14:12:42][I][INFO:826]: Display to partition: 01
[14:12:42][I][INFO:846]: Prompt: FAULT 010 FRONT  
[14:12:42][I][INFO:847]: Prompt: DOOR            
[14:12:42][I][INFO:848]: Beeps: 3

[14:12:43][I][CMD:518]: 2022-12-08 14:12  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[... 35 lines cut ...]
[14:12:57][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:58][I][CMD:518]: 2022-12-08 14:12  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[14:12:58][I][CMD:518]: 2022-12-08 14:12  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:12:59][I][CMD:518]: 2022-12-08 14:12  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[14:12:59][I][CMD:518]: 2022-12-08 14:12  F7 00 CD 00 00 00 00 00 00 00 00 00 77 
[... cut ...]
[14:13:03][I][CMD:518]: 2022-12-08 14:13  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[14:13:03][I][CMD:518]: 2022-12-08 14:13  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[14:13:03][I][CMD:518]: 2022-12-08 14:13  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[14:13:04][I][CMD:518]: 2022-12-08 14:13  FB 01 02 20 40 00 00 00 00 00 00 00 77 
[14:13:04][I][CMD:518]: 2022-12-08 14:13  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[14:13:04][I][EXT:518]: 2022-12-08 14:13  FA 11 F7 00 D5 00 00 00 00 00 00 00 00 
[14:13:04][I][CMD:518]: 2022-12-08 14:13  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
```

Dilbert66, 2022-12-08:
Big issue I see here is that I'm seeing a lot of checksum errors in your cmd packets (the 77 in byte 12).  That means that your circuit needs some tweeking.  You might be better off with the non-isolated version.   You should have prompt messages after every F7 but you are not seeing those as they fail the checksum.

rcmurphy, 2022-12-08:
No zone expanders actually; Motion is on the RF expander

rcmurphy, 2022-12-08:
Let me try supplying external power to the bus

Dilbert66, 2022-12-08:
Very odd then. I've never seen a signal that low.  Can you put your scode on the ESP side and see what the signal looks like. You want the signal to be between 2 and 3 volts peak. If it's too low, you can change the resistor ratio to increase the voltage out.

rcmurphy, 2022-12-08:
You may also be seeing messages for other partitions?

Dilbert66, 2022-12-08:
Doesnt matter, they should all be captured fine with no checksum errors, even if not displayed.  The 0x77 is the telltale.

rcmurphy, 2022-12-08:
Weird. If it was a bad connection one would think it would be consistently bad. I also couldn't resolve it by isolating any part of the ECP bus other than the system itself.

Dilbert66, 2022-12-08:
Only way to know would be to look at the signal.  I've seen issues like that when the signal is hovering at the 0/1 detect threshold of the ESP input pin.

rcmurphy, 2022-12-08:
Well, I replaced the 10K resistors with 15K, and am still getting tons of checksum errors, so no idea:
```
[18:58:00][I][CMD:518]: 2022-12-08 18:58  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[18:58:01][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:01][I][CMD:518]: 2022-12-08 18:58  F7 00 D5 00 00 00 00 00 00 00 00 00 77 
[18:58:01][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:02][I][CMD:518]: 2022-12-08 18:58  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[18:58:02][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:02][I][CMD:518]: 2022-12-08 18:58  F7 00 BD 00 00 00 00 00 00 00 00 00 77 
[18:58:03][I][CMD:518]: 2022-12-08 18:58  F7 00 9D 00 F9 55 02 52 5E 00 00 98 77 
[18:58:04][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:04][I][CMD:518]: 2022-12-08 18:58  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[18:58:04][I][CMD:518]: 2022-12-08 18:58  F7 00 BD 00 00 00 00 00 00 00 00 00 77 
[18:58:05][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:05][I][CMD:518]: 2022-12-08 18:58  F7 00 9D 00 F9 55 02 52 5E 00 98 BE 77 
[18:58:06][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:06][I][CMD:518]: 2022-12-08 18:58  F7 00 CD 00 F7 0F 00 10 00 10 00 1C 77 
[18:58:06][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:07][I][CMD:518]: 2022-12-08 18:58  F7 00 BD 00 00 00 00 00 00 00 00 00 77 
[18:58:07][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:07][I][CMD:518]: 2022-12-08 18:58  F7 00 9D 00 00 00 00 00 00 00 00 00 77 
[18:58:07][I][CMD:518]: 2022-12-08 18:58  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[18:58:08][I][CMD:518]: 2022-12-08 18:58  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[18:58:08][I][CMD:518]: 2022-12-08 18:58  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
```
I also checked, and the ECP bus is peaking between 11.1 and 12 V.

Dilbert66, 2022-12-09:
Very odd.  What about the esp side? Maybe try a different input pin on the esp for the yellow line? Something  is noisy or the output is too low.  Can you compare the input to the output using dual trace on a scope to see if you are not picking up noise somewhere? Is your ground solid?

rcmurphy, 2022-12-09:
Reflowed all the joints, and switched the RX pin.
```
[21:27:33][I][CMD:518]: 2022-12-08 21:27  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[21:27:33][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:33][I][CMD:518]: 2022-12-08 21:27  F7 00 CD 00 F7 0F 00 10 00 21 00 1C 77 
[21:27:34][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:34][I][CMD:518]: 2022-12-08 21:27  FA 01 01 20 30 F7 00 00 00 00 00 00 77 
[21:27:34][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:35][I][CMD:518]: 2022-12-08 21:27  F7 00 9D 00 00 00 00 00 00 00 00 00 77 
[21:27:35][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:35][I][CMD:518]: 2022-12-08 21:27  F7 00 D5 00 00 3E 61 03 00 00 00 00 77 
[21:27:35][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:36][I][CMD:518]: 2022-12-08 21:27  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[21:27:36][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:36][I][CMD:518]: 2022-12-08 21:27  FA 01 01 20 30 F7 00 00 00 00 00 00 77 
[21:27:36][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:37][I][CMD:518]: 2022-12-08 21:27  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[21:27:37][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:37][I][CMD:518]: 2022-12-08 21:27  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[21:27:37][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:38][I][CMD:518]: 2022-12-08 21:27  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[21:27:38][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:38][I][CMD:518]: 2022-12-08 21:27  F7 00 BD 00 00 BE 71 FE 00 00 00 00 77 
[21:27:39][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:39][I][CMD:518]: 2022-12-08 21:27  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[21:27:39][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:39][I][CMD:518]: 2022-12-08 21:27  FA 01 01 08 30 F7 00 00 00 00 00 00 77 
[21:27:40][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:40][I][CMD:518]: 2022-12-08 21:27  FA 01 01 10 30 F7 00 00 00 00 00 00 77 
[21:27:40][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:40][I][CMD:518]: 2022-12-08 21:27  FA 01 01 20 30 F7 00 00 00 00 00 00 77 
[21:27:41][I][CMD:518]: 2022-12-08 21:27  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[21:27:41][I][CMD:518]: 2022-12-08 21:27  FA 01 01 40 30 F7 00 00 00 00 00 00 77 
[21:27:41][I][CMD:518]: 2022-12-08 21:27  F7 0F 00 10 00 21 00 1C 38 02 00 00 48 
```

rcmurphy, 2022-12-09:
I'll have to get the scope and probes out tomorrow.

Dilbert66, 2022-12-09:
ok, had a better look at the FA cmd and looks like my logic is not correct for your system.  There are two variations for the packet.  One has a longer byte depending on it's type.  I was using byte 2 for the flag to determine that, but that logic doesnt work on your system.   I wonder if that is also the case with your system for the F7.  Let me add some debug code to try and determine what is going on to see if that's the case.

rcmurphy, 2022-12-09:
Could this be related to the missing hundreds digit in the zone number?

rcmurphy, 2022-12-09:
I don't want you to feel like you have to support the 250, but if you do, I would be happy to give you ngrok style remote access to logs

rcmurphy, 2022-12-09:
Any idea where those messages are coming from? That way I can hook Ch1 up
to the source directly and Ch2 up to the ESP.

The other thing I could do is add something like an opamp to really make
sure that we're getting a solid 0 or 1

Dilbert66, 2022-12-09:
The FA is a normal cmd. It's the status request message from the panel to the zone and relay expanders on your system.   The issue is that it's a very basic cmd of 6 or 7 bytes. The issue is that the checksum is missing in your logs.   The F9 cmd (LRR status) checksum adds up fine.   This might also be the same issue with your F7, the chksum might be missing.  Might be a timing issue somewhere that your panel differs a bit and the serial routines are not capturing it.   No way for me to know unless I'd see the data via a logic analyzer trace.  

An op amp is overkill.  There is something else at play here.  The resistor divider is more than adquate.

rcmurphy, 2022-12-09:
> This is what your system shows for a FA:
> FA 01 01 08 30 F7 00
> FA 01 01 10 30 F7 00
> FA 01 01 20 30 F7 00
> FA 01 01 40 30 F7 00
>
> This is on my system:
> FA 01 01 02 20 F1 F1
> FA 01 01 04 25 F1 EA
>
> Yours has 00 where the checksum should be.

Ok, ordered a logic analyzer.Best,Rebecca

Dilbert66, 2022-12-09:
There is something odd though.   Your logs from timelords's issue  looked ok, but these new ones are showing the issue? What changed?

rcmurphy, 2022-12-09:
Ah. I was playing with virtual relays and forgot to disable the 4 ECP devices when I was done.

```
[11:01:05][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:06][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:07][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:07][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:09][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:10][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:10][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:11][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:11][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:12][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:12][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:13][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:13][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:14][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:15][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:15][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:16][I][CMD:518]: 2022-12-09 11:01  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[11:01:16][I][CMD:518]: 2022-12-09 11:01  F7 47 00 10 00 00 00 1C 38 02 00 00 48 
[11:01:16][I][INFO:826]: Display to partition: 01
[11:01:16][I][INFO:846]: Prompt: HOME DISARMED    
[11:01:16][I][INFO:847]: Prompt: BYPAS-RDY TO ARM
[11:01:16][I][INFO:848]: Beeps: 0
```

Dilbert66, 2022-12-09:
ah, ok.  Still doesnt explain why your chksums were not valid though...  I use virtual relays here and works fine.

rcmurphy, 2022-12-09:
Well, I had then configured in the panel, but not in the device.

rcmurphy, 2022-12-09:
```
[12:51:31][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:31][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:32][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:32][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:33][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:33][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:34][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:34][I][CMD:518]: 2022-12-09 12:51  F7 47 00 10 00 10 03 00 38 02 00 00 46 
[12:51:34][I][INFO:826]: Display to partition: 01
[12:51:34][I][INFO:846]: Prompt: FAULT 010 FRONT  
[12:51:34][I][INFO:847]: Prompt: DOOR            
[12:51:34][I][INFO:848]: Beeps: 3

[12:51:35][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:35][I][CMD:518]: 2022-12-09 12:51  F7 47 00 10 00 10 00 0C 38 02 00 00 48 
[12:51:35][I][INFO:826]: Display to partition: 01
[12:51:35][I][INFO:846]: Prompt: HOME DISARMED    
[12:51:35][I][INFO:847]: Prompt:   ZONES FAULTED 
[12:51:35][I][INFO:848]: Beeps: 0

[12:51:35][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:36][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:37][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:38][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:38][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:39][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:39][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:39][I][CMD:518]: 2022-12-09 12:51  F7 47 00 10 00 10 00 00 38 02 00 00 46 
[12:51:39][I][INFO:826]: Display to partition: 01
[12:51:39][I][INFO:846]: Prompt: FAULT 010 FRONT  
[12:51:39][I][INFO:847]: Prompt: DOOR            
[12:51:39][I][INFO:848]: Beeps: 0

[12:51:40][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:40][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:41][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:41][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:42][I][CMD:518]: 2022-12-09 12:51  F7 47 00 10 00 10 00 1C 38 02 00 00 48 
[12:51:42][I][INFO:826]: Display to partition: 01
[12:51:42][I][INFO:846]: Prompt: HOME DISARMED    
[12:51:42][I][INFO:847]: Prompt: BYPAS-RDY TO ARM
[12:51:42][I][INFO:848]: Beeps: 0

[12:51:42][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:42][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:43][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:44][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:44][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:45][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:45][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:46][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:46][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:47][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
[12:51:47][I][CMD:518]: 2022-12-09 12:51  F9 55 02 52 5E 00 00 00 00 00 00 00 00 
```

Let's try this for a clean sample of the RF logs.

Dilbert66, 2022-12-09:
Interesting. I only see one RF supervision cmd and one response.  You do have a checksum error with the cmd though:
```
[13:20:14][I][CMD:518]: 2022-12-09 13:20  FB 01 02 20 40 00 00 00 00 00 00 00 77 
[13:20:14][I][EXT:518]: 2022-12-09 13:20  07 24 00 D5 00 00 00 00 00 00 00 00 74 
```

The response EXT is an unknown though (flag 74).

You must have long range radio enabled in your panel because I see a lot of F9 status requests with no response.

rcmurphy, 2022-12-11:
Yeah, it's because the LRR isn't responding back normally. Or wasn't. Should be less chatty now.

Would hooking the logic analyzer up and getting a dump from that still be useful in troubleshooting?

I'm also wondering if the 4232CBM would provide the data I'm looking for if nothing else.

Dilbert66, 2022-12-12:
Logic traces of your FB and FA cmds and any following EXT responses would be good to see as well.  The FB cmds are the ones that represent your RF traffic.  The FA are for zone and relay expanders.

rcmurphy, 2022-12-17:
@Dilbert66 do you have a way I can send you the dump privately? My address is in the keypad default message; otherwise I need to reprogram that.

Dilbert66, 2022-12-17:
Sure, you can send to aturbide99 at gmail.com

rcmurphy, 2022-12-17:
Sent!

Dilbert66, 2022-12-17:
Ok, I found a bug with the RF EXT packet calc.  That does not explain some of the other odd chksum failures you have with the F7 and FA commands, but this should correct the RF zone serial and status decoding. I've pushed an update to dev.

Dilbert66, 2022-12-17:
One other issue I found is related to odd noise pulses periodically in your traffic.  They would cause data corruption.  No idea of what the source would be.
[image]

Dilbert66, 2022-12-18:
Ok, your logs were very helpful. Found a new cmd F0 that is not present in the hardware I have. Excellent. That would explain some of the invalid data you see.  My code didnt know how to handle it. I will add support and publish an update shortly.

Dilbert66, 2022-12-18:
Pushed new update to dev with F0 command processing added. 
The command is a polling cmd where it sends F0 + byte (most likely a device address).  Devices on the bus will then respond with a 9 byte string.  The panel will then acknowledge with the header byte of the received string.

rcmurphy, 2022-12-18:
Hmm, now I'm not seeing any `EXT` entries in the logs:
```
INFO Reading configuration /config/esphome/alarm-system.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'America/New_York'
INFO Starting log output from alarm-system.local using esphome API
INFO Successfully connected to alarm-system.local
[11:34:51][I][app:102]: ESPHome version 2022.12.0-dev compiled on Dec 18 2022, 11:32:27
[11:34:51][I][i2c.arduino:068]: Results from i2c bus scan:
[11:34:51][I][i2c.arduino:074]: Found i2c device at address 0x3F
[11:34:53][I][CMD:518]: 2022-12-18 11:34  F7 73 00 00 00 20 00 0C 28 02 00 00 48 
[11:34:53][I][INFO:826]: Display to partition: 01
[11:34:53][I][INFO:846]: Prompt: HOME DISARMED    
[11:34:53][I][INFO:847]: Prompt:  ZONES IN CHECK 
[11:34:53][I][INFO:848]: Beeps: 0

[11:35:00][I][CMD:518]: 2022-12-18 11:35  9E 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:00][I][CMD:518]: 2022-12-18 11:35  08 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:00][I][CMD:518]: 2022-12-18 11:35  25 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:00][I][CMD:518]: 2022-12-18 11:35  82 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:00][I][CMD:518]: 2022-12-18 11:35  56 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  9E 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  08 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  20 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  F1 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  EC 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  F9 C3 02 53 EF 00 00 00 00 00 00 00 00 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  43 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  98 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  BE 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  10 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  30 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  F7 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  CE 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  F0 00 00 00 00 00 00 00 00 00 00 00 00 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  80 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:01][I][CMD:518]: 2022-12-18 11:35  F7 73 00 00 00 12 03 00 28 02 00 00 46 
[11:35:01][I][INFO:826]: Display to partition: 01
[11:35:01][I][INFO:846]: Prompt: FAULT 012 PATIO  
[11:35:01][I][INFO:847]: Prompt: DOOR            
[11:35:01][I][INFO:848]: Beeps: 3

[11:35:02][I][CMD:518]: 2022-12-18 11:35  F0 01 00 00 00 00 00 00 00 00 00 00 00 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  F0 01 00 00 00 00 00 00 00 00 00 00 00 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  0F 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  9E 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  08 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  25 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  F1 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:03][I][CMD:518]: 2022-12-18 11:35  E7 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:04][I][CMD:518]: 2022-12-18 11:35  F0 01 00 00 00 00 00 00 00 00 00 00 00 
[11:35:04][I][CMD:518]: 2022-12-18 11:35  23 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:04][I][CMD:518]: 2022-12-18 11:35  F7 73 00 00 00 20 00 00 28 02 00 00 46 
[11:35:04][I][INFO:826]: Display to partition: 01
[11:35:04][I][INFO:846]: Prompt: FAULT 020 OWNERS 
[11:35:04][I][INFO:847]: Prompt: BEDROOM DOOR    
[11:35:04][I][INFO:848]: Beeps: 0

[11:35:04][I][CMD:518]: 2022-12-18 11:35  F0 02 00 00 00 00 00 00 00 00 00 00 00 
[11:35:04][I][CMD:518]: 2022-12-18 11:35  0E 00 00 00 00 00 00 00 00 00 00 00 91 
[11:35:05][I][CMD:518]: 2022-12-18 11:35  F0 03 00 00 00 00 00 00 00 00 00 00 00 
[11:35:05][I][CMD:518]: 2022-12-18 11:35  83 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:06][I][CMD:518]: 2022-12-18 11:35  F0 03 00 00 00 00 00 00 00 00 00 00 00 
[11:35:06][I][CMD:518]: 2022-12-18 11:35  C3 00 00 00 00 00 00 00 00 00 00 00 90 
[11:35:07][I][CMD:518]: 2022-12-18 11:35  F0 04 00 00 00 00 00 00 00 00 00 00 00 
[... 116 lines cut ...]
[11:36:30][I][CMD:518]: 2022-12-18 11:36  9E 00 00 00 00 00 00 00 00 00 00 00 90 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  08 00 00 00 00 00 00 00 00 00 00 00 91 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  25 00 00 00 00 00 00 00 00 00 00 00 91 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  82 00 00 00 00 00 00 00 00 00 00 00 91 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  56 00 00 00 00 00 00 00 00 00 00 00 91 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  F9 43 02 53 6F 00 00 00 00 00 00 00 00 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  C3 00 00 00 00 00 00 00 00 00 00 00 90 
[11:36:30][I][CMD:518]: 2022-12-18 11:36  FA 01 10 30 F7 00 00 00 00 00 00 00 77 
[... cut ...]
[11:36:34][I][CMD:518]: 2022-12-18 11:36  0E 00 00 00 00 00 00 00 00 00 00 00 91 
[11:36:35][I][CMD:518]: 2022-12-18 11:36  F0 03 00 00 00 00 00 00 00 00 00 00 00 
[11:36:35][I][CMD:518]: 2022-12-18 11:36  03 00 00 00 00 00 00 00 00 00 00 00 90 
[11:36:36][I][CMD:518]: 2022-12-18 11:36  F0 03 00 00 00 00 00 00 00 00 00 00 00 
[11:36:36][I][CMD:518]: 2022-12-18 11:36  43 00 00 00 00 00 00 00 00 00 00 00 90 
[11:36:37][I][CMD:518]: 2022-12-18 11:36  F0 04 00 00 00 00 00 00 00 00 00 00 00 
[11:36:37][I][CMD:518]: 2022-12-18 11:36  04 00 00 00 00 00 00 00 00 00 00 00 90 

```

Dilbert66, 2022-12-18:
Are you using the last push I did not long ago? I also don't see any EXT
responses which means it's not seeing the green line data.
Very odd. Works fine here.

Dilbert66, 2022-12-18:
The ext missing normally would mean an issue with your line connection.
The code I worked on should not affect that.

rcmurphy, 2022-12-18:
Weird. I haven't even entered the room with the ESP. I'll try reverting and see what happens...

rcmurphy, 2022-12-18:
Oh, it overwrote my patch to enable using the GPIs as monitoring pins. I use pin 39.

Dilbert66, 2022-12-18:
What patch is that? I can add that if needed. 
There is still an issue I need to address. You are getting some missed FA cmds. I see those here as well. I need to tweak a couple things.

rcmurphy, 2022-12-18:
Oh, I just tweaked the isValidGPIOPin function to allow me to use pins up to GPI39. Obviously that won't work for outputs though.

Up to date logs:
[logs_alarm-system_upload.txt](https://github.com/Dilbert66/esphome-vistaECP/files/10254336/logs_alarm-system_upload.txt)

rcmurphy, 2022-12-18:
Out of curiosity, how do you associate the RF serial number with the zone after you intercept it?

Dilbert66, 2022-12-18:
oh ya.... I see that. I can adjust that to allow inputs.  

As to the serial numbers, I don't associate as the serial number comes from the RF module traffic and it has no zone identity there.   It's only the panel that has that data.  It then uses the RF data to send an F7 with the updated status.    I suppose we could create a lookup table in the ESP to do that just like the panel does.

rcmurphy, 2022-12-18:
The advantage of associating serial numbers would be that you could overcome the limitation of the vista panels and get motion detection in interior zones 😄 
You could even make the user add the LUT.

Dilbert66, 2022-12-18:
The FB cmds (RF serial data) are messed up in you logs. I have to fix my code. It's tough testing those as I don't have an RF module to work with. I depend on you folks!

rcmurphy, 2022-12-18:
Shoot me your address and I'll send you a 5881ENH that needs a good home!

Dilbert66, 2022-12-18:
I'll add the max_zones config in the yaml as well as the serial lookups. Easy enough to do. Never needed it before so didnt bother. 
FYI, my main home system is a DSC panel where I have RF there.  The vista was a side project to toy with :)

Dilbert66, 2022-12-18:
> Shoot me your address and I'll send you a 5881ENH that needs a good home!

Hmm, very tempting!  If you're sure you don't need it.  I'll pay shipping.

rcmurphy, 2022-12-18:
Ok, I emailed you.

Dilbert66, 2022-12-18:
> The advantage of associating serial numbers would be that you could overcome the limitation of the vista panels and get motion detection in interior zones 😄 You could even make the user add the LUT. It would be lovely if there was a way to pass in MAX_ZONES so I don't have to keep patching that too, but that might be tougher as its a constant.

I do like that idea. I have something similar on my DSC.  Internal RF is monitored but doesnt trigger the panel.   I'lll put the LUT config in the yaml .

Dilbert66, 2022-12-18:
I've pushed a small tweak to "vista.cpp".  See if that helps with your FB (rf cmd processing).

rcmurphy, 2022-12-18:
whoops.

Dilbert66, 2022-12-18:
Well, another tweak. See if that helps with the FB cmds and serial decode.

rcmurphy, 2022-12-18:
[logs_alarm-system_upload (1).txt](https://github.com/Dilbert66/esphome-vistaECP/files/10254474/logs_alarm-system_upload.1.txt)

Dilbert66, 2022-12-18:
Can you post some logs with the list pushed version of vista.cpp? Tks!

rcmurphy, 2022-12-18:
Oh, That was updated I thought. I think I updated the wrong file though; is that second copy vestigial?
[logs_alarm-system_logs (1).txt](https://github.com/Dilbert66/esphome-vistaECP/files/10254632/logs_alarm-system_logs.1.txt)

Dilbert66, 2022-12-18:
If possible, can i see a sample of a logic analyzer trace when sending keys via your physical keypad? There differences in the sync pulse for your system compared to a vista20 and this will impact keypad presses. I need to figure out how yours does it.
Your system:
[image]

Vista20p:
[image]

[31 more replies]

closed 2022-12-20 by rcmurphy