Dilbert66/esphome-vistaECP #91: where do I find the the esphome directory?

finnishcolumbia, 2023-01-12:
"To use this software you simply place the vistaAlarm.yaml file in your main esphome directory, then copy the *.h and *.cpp files from the vistaEcpInterface directory to a similarly named subdirectory (case sensitive) in your esphome main directory and then compile the yaml as usual. The directory name is in the "includes:" option of the yaml."

Where can I find this directory? I am accessing esphome through  home assistant from within my browser. Is there another way?

lorenzodeveloper, 2023-01-12:
Ciao @finnishcolumbia ,

You can check the dev branch, it contains some updates to the readme part that guides you more on the installation part, but it's still under revision by @Dilbert66 so it might be not 100% accurate or complete.

However, you can access to the esphome directory by either connecting to home assistant via samba (e.g. Samba share plugin) or by using a simple file explorer plugin on homeassistant (e.g. File editor plugin).

The esphome folder is located in "<HA-IP>/config/esphome".

Lorenzo

Dilbert66, 2023-01-12:
You can also use the Terminal & SSH addon then use winscp as the client (or scp if using linux).   I find File Editor to be a bit tedious to upload many files as the developers only support uploading 1 file at a time.  Using Samba is probably your best bet. as Lorenzo noted.

closed 2024-11-01 by Dilbert66