Dilbert66/esphome-vistaECP #93: [dev] Compile Problems on Case Sensitive FS

rcmurphy, 2023-01-24:
`vistaAlarm.h` is referenced as `vistaalarm.h`; I was going to submit a PR to just change the include, but I think a better solution would be to switch to lower case filenames for *.cpp / *.h.

Dilbert66, 2023-01-24:
Yes, I was switching the includes to lower case for a few includes (except for the serial library for now).
Not sure where you got the reference error though. I had not pushed the update yet. I'll do that now.

Dilbert66, 2023-01-24:
Interesting. i just did a push and it did not change the case on the pushed version but my copy is fine.  I guess, git doesnt change the case.  I have to read up on this and correct it.

Dilbert66, 2023-01-24:
Was a git setting to tell it not to ignore case. I repushed again with corrected file names.

closed 2023-01-24 by rcmurphy