Dilbert66/esphome-vistaECP #94: failing to compile after uncommenting "define MONITORTX"

freaker2112, 2023-01-26:
when i tried to compile to push new code to the esp module, it errored out.
my system is also displaying a bf error, and everything except ac power and system status shows up as 'unknown' or 'C', did i miss a step?

compile log is attached
[logs_vista20p_upload.txt](https://github.com/Dilbert66/esphome-vistaECP/files/10504800/logs_vista20p_upload.txt)

Dilbert66, 2023-01-26:
You need to have the # in there as in #define. That's a preprocessor directive. A comment is two slashes // in front.  No slashes then it's not commented out.

freaker2112, 2023-01-26:
that worked! its compiling again but still shows the same status as before in HA. i may have wired it wrong. is the esp supposed to connect directly to the big vista20p board? i used the wires that used to go to the adt board. 

on a side note, how do i clear the 'bf' error on the keypad? the only ways i saw used an envisalink

Dilbert66, 2023-01-26:
Check if you get panel cmd packets If not, then your connection is bad.  I have no idea what you mean by an adt board.  As long as you connect to the green/yellow and black wires of the bus.  The error you show indicates a connection problem with LRR I believe.  For initial testing make sure lrrSupervisor is false and all the expander and relay addresses are 0  in your yaml. Otherwise I have no clue what your issue is.  You can try entering your access code + 1 a couple times to clear errors.

closed 2023-02-03 by freaker2112