Dilbert66/esphome-vistaECP #96: RF door sensors reporting open at regular intervals

travisrroy, 2023-02-17:
### Problem
I updated to the latest dev branch since I had some issues with the ESP32 disconnecting randomly after unexpected power loss and now I have a new problem. Basically, each door sensor will report to home assistant that it has been opened when it has not every 72 minutes or so since the last time it falsely reported it. Maybe something to do with `rfSerialLookup`? I do like this new feature though.

### Platform
Vista-20P, 6150RF, 6x 5816WMWH, ESP32, Home Assistant 2023.2.3

### Config

```
#for documentation see project at https://github.com/Dilbert66/esphome-VistaECP
substitutions:
  systemName: "vistaalarm"
  accessCode: !secret access_code #Only comes into effect if needed for arming and quickarm is not set

  maxZones: "48" #maximum amount of zones that your panel supports
  maxPartitions: "3" #maximum amount of partitions that your panel supports

  rfSerialLookup: "0397567:09:80,0931570:10:80,0037117:11:80,0117340:12:80,0136928:13:80,0798385:14:80,0199136:15:80,0474834:16:80" # list of RF serial numbers with associated zone and bitmask. 
  #Format: "serial1#:zone1:mask1,serial2#:zone2:mask2"
  #Mask: hex value used to mask out open/close bit from RF returned value
  
  defaultPartition: "1"  

  #assign a new virtual keypad address to each active partition using programs *190 - *196
  #and enter it below.  For unused partitions, use 0 as the keypad address.
  keypadAddr1: "19" #partition 1 virtual keyapd
  keypadAddr2: "0" #partition 2 virtual keypad. set to 0 to disable
  keypadAddr3: "0" #partition 3 virtual keypad. set to 0 to disable

  ##esp32
  rxPin: "22" #GPIO pin to use for data receive (yellow line) 
  txPin: "21" #GPIO pin to use for data transmit (green line)
  monitorPin: "18" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  ##esp8266
  # rxPin: "5" #GPIO pin to use for data receive (yellow line) 
  # txPin: "4" #GPIO pin to use for data transmit (green line)
  # monitorPin: "14" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable
  
  # module addresses:
  # 07 4229 zone expander  zones 9-16
  # 08 4229 zone expander zones 17-24
  # 09 4229 zone expander zones 25-32
  # 10 4229 zone expander zones 33-40
  # 11 4229 zone expander zones 41 48
  # 12 4204 relay module  
  # 13 4204 relay module
  # 14 4204 relay module
  # 15 4204 relay module
  
  expanderAddr1: "8" # 1st zone expander emulator (4229) address to use . Set to 0 to disable. 
  expanderAddr2: "0" # 2nd expander emulator address to use . Set to 0 to disable. 
  
  relayAddr1: "0" # relay module emulation (4204) addresses. Set to 0 to disable
  relayAddr2: "0"
  relayAddr3: "0"
  relayAddr4: "0"
  
  TTL: "15000"  # time  to live  in ms for zone/fire status before expiring;
  quickArm: "false"
  lrrSupervisor: "true" # set to true if we don't have an LRR monitoring supervisor we can emulate one to get the statuses

esphome:
  name: $systemName
  platform: ESP32
  #board: nodemcu-32s
  board: mhetesp32devkit
  # platform: ESP8266
  # board: nodemcuv2

[... 304 lines cut ...]
    name: "$systemName Beeps"  

  - platform: template
    id: zoneExtended
    name: "$systemName Zone Status"    

switch:
  - platform: template
    name: "$systemName Connection"
    id: connection_status_switch
    lambda: |-
      return vista.keybusConnected;
    icon: "mdi:shield-link-variant"
    turn_on_action:
      - switch.toggle: restart_switch
    turn_off_action:
      - lambda: |-
          disconnectVista();
  - platform: restart
    id: restart_switch
```

### Logs
I managed to catch these 3 logs from ESPHome.  
[image]

Dilbert66, 2023-02-17:
Well, I can see that the panel is sending RF device statuses with the status of 0x84 and since your bit mask is set to 0x80, it sees it as an open.  Do you have a status with a close event so we can determine what the correct bit mask should be.

Dilbert66, 2023-02-17:
I think the heartbeat that rf devices send every 60 to 90 minutes is causing the issue. I'll check my code in how it handles the mask.

Dilbert66, 2023-02-17:
I've pushed a small update on dev. Just an updat to vistalarm.h that has it ignoring RF heartbeat status messages for zone updates.  See if that addresses your issue. I'd still like to see a close RF status as well to see if there isnt anything else at play here.

travisrroy, 2023-02-18:
Ok, so I haven't updated yet, but I collected these logs. I'm not sure if they have any specific close events in them. 

**Door 1**
Event 1
[image]  
Event 2 (I believe this is me actually opening the door)
[image]
Event 3
[image]

Door 2
[image]
[image]

Dilbert66, 2023-02-18:
Ok, I see the issue. You should be using 0x20 as the mask and not 0x80.  With rf sensors there are 4 possible loops that can be used and the corresponding masks would be 0x80, 0x40, 0x20 and 0x10.  Your's has two loops defined.  0x80 is always on and 0x40 is what determine the open/close status. 0x  I don't know the logic at this time as to how the loops are selected.  The rf serial lookup is still experimental at this time. So change the mask value (3rd entry of each record) of each sensor to 0x20.

travisrroy, 2023-02-18:
Ok. It looks like changing the mask fixed the issue. One question about the rf serial lookup is would I figure out the mask for motion sensors and smoke detectors?

Dilbert66, 2023-02-19:
Only way is to trigger then reset them and see what the value changes to in the RFX line in the log.  It's a guessing game as I don't know the logic that determines it.  I don't have enough data from other users to compare. I also do not have any rf sensors set up so can't test myself. I'm only working off a bench test system as far as the vista is concerned.

travisrroy, 2023-02-19:
Ok, I tested the smoke detector. B0 when triggered and 00 when disarmed. Not sure how to determine the mask for that. Still waiting/trying to trigger the motion detector.

travisrroy, 2023-02-19:
Ok, so the PIR motion produces an 80 when triggered and 00 when cleared.

Dilbert66, 2023-02-19:
ok, Great. From what I see you can use 0x80 mask for the PIR and smoke detectors as they all set bit 7  (0x80) on and off.   For the door sensors, you use 0x20 for the mask as bit 7 (0x80) of the status seems to always be on but bit 5 (0x20) is the one that toggles to indicate open/close status. Interesting though that the smoke sensor set bit 7, 5 and 4 on for active and 0 for idle.  You only need to test for one of those bits so might as well use bit 7(0x80)

travisrroy, 2023-02-22:
So setting the proper mask worked great. When I updated to the latest commit (75d02a6), I got a warning in the compiler and the doors were no longer working in HA. I downgraded back to 21fa8b5 and it works great again.

Dilbert66, 2023-11-03:
FYI, i've changed the way the RF devices are entered in the yaml to make it easier.  It will use whatever info you have setup from program *56 for loop# and zone#.  No need to worry about masks.  Just enter the serial:loop#:zone#. You can see this document for default loop numbers for various devices: 
https://advancedsecurityllc.com/wp-content/uploads/5800%20Wireless%20Device%20List.pdf

closed 2024-11-01 by Dilbert66