Dilbert66/esphome-vistaECP #100: Cannot get zones to report status when using dev branch.

XD36JD2015, 2023-03-30:
When using the master branch code, I am able to get statuses (O or C) for all of my zones.  When using the dev branch (binary or single partition text) all of the statuses will pull in from my panel other than the zone statuses.  I don't know if I am missing something in setup or if there is another problem.  Let me know if you need more info from me.  Thanks.

Info
OS: Home Assistant OS
Home Assistant version: 2023.3.6
ESP Home version: 2023.3.2
ESP device: ESP8266
Alarm Panel Vista 20p with RF radio
[logs_vistaalarmnew_compile_text_dev.txt](https://github.com/Dilbert66/esphome-vistaECP/files/11106764/logs_vistaalarmnew_compile_text_dev.txt)
[logs_vistaalarmnew_run_text_dev.txt](https://github.com/Dilbert66/esphome-vistaECP/files/11106765/logs_vistaalarmnew_run_text_dev.txt)
[logs_vista_run_master.txt](https://github.com/Dilbert66/esphome-vistaECP/files/11106766/logs_vista_run_master.txt)
[vistaalarmnew.yaml.txt](https://github.com/Dilbert66/esphome-vistaECP/files/11106769/vistaalarmnew.yaml.txt)

Dilbert66, 2023-03-30:
Odd, ok, check a couple things.  Make sure "paneltext.h" has this line:     const char * FAULT = "FAULT";    
Also,  set debug to 2 in your yaml.
  VistaECP->debug=2;
I've pushed an update to vistaalarm.h to remove some warnings.   Use that.

XD36JD2015, 2023-03-31:
Here is a log with debug set to 2 and the newest versions of the files in the repository,
[logs_vistaalarmtextesp8266_logs.txt](https://github.com/Dilbert66/esphome-vistaECP/files/11117784/logs_vistaalarmtextesp8266_logs.txt)

Dilbert66, 2023-03-31:
I don't see the debug print of "The prompt  %s was matched" so it means that it's not seeing the "FAULT" message.  Did you look at the paneltext.txt file to make sure that this line:  const char * FAULT = "FAULT";
I don't get why it's not finding it on your system since it looks ok.

XD36JD2015, 2023-03-31:
I can confirm that const char * FAULT = "FAULT"; is indeed in paneltext.txt.

Dilbert66, 2023-03-31:
Then I have no idea as it works fine on my end and on yours its not even finding the prompt.  Something is different on your config somehow.

Dilbert66, 2023-03-31:
I did find an issue with how I was handling the const char value in the prompt check. Not sure if that was the issue but I've pushed an update to vistaalarm.h

XD36JD2015, 2023-03-31:
Your changes fixed the issue.  Thanks again!

closed 2023-03-31 by XD36JD2015