Dilbert66/esphome-vistaECP #104: polymer depreciated

jakejakejakejakejakejake, 2023-05-05:
As an FYI, polymer was depreciated. This breaks the custom keypad.

https://developers.home-assistant.io/blog/2023/04/04/deprecating_polymer/

Just wanted to others know.

You will see `Uncaught ReferenceError: Polymer is not defined` in browser console.

Dilbert66, 2023-05-05:
> I just ran into this myself. I downgraded by running ha core update
> --version 2023.4.5 from the shell.

Use the latest version in dev branch of the script

jakejakejakejakejakejake, 2023-05-05:
I will try that tonight thank you!

MihaT2021, 2023-05-23:
I get in HA dashbord
Error from custom card
r.setConfig  is not  a funcion

Dilbert66, 2023-05-23:
Read my comment above.

jakejakejakejakejakejake, 2023-07-30:
> I will try that tonight thank you!

Many nights later, I tried, and it worked. Many more nights later I am here to close this issue.

closed 2023-07-30 by jakejakejakejakejakejake