Dilbert66/esphome-vistaECP #105: Panels do report close events

sirvictory444, 2023-05-09:
"The panel only sends fault messages when a zone is faulted or alarmed and does not send data when the zone is restored".   

They can report closes, both wired and wireless.  The Big Blue company always nerfed their firmware, perhaps what you have is.

 I have several 20p, various firmware versions, bought through a Honeywell dealer and they all immediately report zone closings.   As I understand it, this is not accounted for because your panel doesn't report closes and instead a TTL was implemented, is that correct? 
 Would it be hard to add reading those events from the ecp bus?

Dilbert66, 2023-06-05:
Vista's do not report closing to the display that I know of. Perhaps you can post logs of your panel messsages and we can see if yours is reporting different then others.  There is no way for me to know about those supposed events unless I see packet logs.

rcmurphy, 2023-06-05:
Oooh, let me know what to look for; would love to see if 250s are
different. The alarm company I have is amazing about explaining the
internals; I'll ask them too.

Dilbert66, 2023-06-05:
Ok, I should clarify.  The display really only displays what the panel sends it.  When a fault occurs, the panel sends a "fault" message to the display.  When the fault is restored, there is no "restored" message. It simply stops sending the fault message.  My firmware basically only emulates a display panel.  Now having said that, I also added the ability to capture messages from the RF module to the panel since those are sent on the same bus.  With those I can capture open/close/battery status messages    direct from the device . That's the exception.  With wired zones that is not happening since the panel detects those open/close events directly and nothing is sent on the bus so we have to depend on what the alarm panel sends out to the display. 

Now there is a way to detect close messages from the internal 8 wired zones but that requires the convoluted use of emulated relay followers, etc which I wont' get into here.   So I don't see why Vista would have firmware that would send "restore" events to the panel for no reason since it doesnt display them anyhow.  If your panel does display those restore, I'd be suprised, but hey, I havent seen everthing.  Post logs  of a fault/restore of a zone and we will see if the data supports what you are telling me.

ChrisLizon, 2024-01-10:
@Dilbert66  Based on your response there I'm curious if Zones on Zone Expanders such as the 4219 would be faster to report the closed events since they're operating on the ECP bus? I would be tempted to add a zone expander and move my "important" zones to the zone expander if they're going to update quicker.

I'm also tempted to attempt to build a board that acts as a zone expander with a analog to digital chip (I don't really awnt to cut out the EOL resistors) on the i2c bus using your library. I just haven't quite figured that out yet. :D

Dilbert66, 2024-05-14:
You have to realize the door switches are connected directly to the panel and not to the bus.  The panel will detect the closure immediately because of that circuit, not because of a bus message.   The panel will not send out a close message on bus.  It only sends out Fault messages.  The only way that a close event will be seen on the bus is if the door sensor is an RF module so those messages will be captured on the bus. This is why the ESP code relies on an absense of a fault message for a period of time to determine a close event.

rwoldberg, 2024-05-14:
Sorry I was not clear with that message, I have two keypads attached to the main controller, when I open and close a door the keypads show both the open and close immediately, there is no delay as there is with the ESPHome integration.  The keypads are connected exactly the same as the ESP32 module so they should receive the same data correct?  And yes all of my switch are wired I have 16, 8 wired to the panel and another 8 wired via a zone expander.  I was testing with the front door which is wired to the panel no the zone expander.

Dilbert66, 2024-05-14:
You'll need to be more specific here as to what keypads you have. With 6160 keypads, all you see are the fault/alarm messages. You do not see closure messages.  The ESP code is emulating a 6160 type LCD keypad.

Dilbert66, 2024-05-14:
I re-read your message.  The ready light turning on/off with the zone action is a different message altogether.   When an open zone is closed, and if there are no other open zones on that partition, a ready message will be sent on the bus and the ESP code will show that as soon as it gets it which would be the same time as your physical keypad.   If your implementation is not showing the ready signal quickly then there is something else at play than the esp code since there should not be any delay showing the ready light on.

rwoldberg, 2024-05-14:
I was looking at that and I think the wired keypads when they get the READY message they also clear the code for the open switches.  I was thinking maybe the integration could do the same thing, when it sees the READY message then trigger the close event for any opened switches, since they would have to be closed for the system to ready.  Also I have to key pads an ADEMCO 6150 and a Honeywell 6271c graphical keypad.  The graphical keypad shows the open zones, when one closes it is removed from the keypad in 2 seconds, but if I set the timeout in the ESPHome integration to 2 seconds, any open zone will continually trigger open and close.  Why do you think there would be a difference?

Dilbert66, 2024-05-14:
ah.. ok,  I misunderstood what you were saying.  Yes, I had mean to implement the reset of open zones when the ready flag is active but forgot about it since I got busy with other aspects. I'll do that. Thanks for reminding me.  The graphical keypad works differently as it has a different protocol then the 6160 keypads.  It has more info provided to it from the panel.  As to the TTL timer, if you set it too short, it will timeout and think the zone is closed but then an updated Fault message comes in and retriggers it as open again.

rwoldberg, 2024-05-14:
Oh, interesting.  I get setting the TTL too low causes issue, I was just wondering why my keypad was behaving differently, now I know.  Thanks.

Dilbert66, 2024-05-15:
Actually, if you can, can you post esphome logs of the traffic you get when a zone is opened or closed? You would need to be using the dev  branch version of the code though.  I'm particularely interested in the F2 packet messages as those are the ones that hold the traffic between the panel and the 6170  keypad. I could potentially use this traffic to speed up zone status updates.  It can also be used for other functions as retrieving panel status data such as partitions, attached devices, active zones, etc.

  I've also updated the dev branch to clear zone's when the partition is in status ready as discussed.

rwoldberg, 2024-05-16:
Here is some data from opening and closing a door a couple times:
[20:29:15][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 54 FE FE EC 03 01 01 01 BC 
[20:29:18][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[20:29:18][I][CHK:572]: 2024-05-15 20:29  61 00 00 00 00 00 00 00 00 00 00 00 90 
[20:29:18][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 55 FE FE EC 03 01 01 01 BB 
[20:29:19][I][CMD:572]: 2024-05-15 20:29  F7 02 00 1D 10 08 00 1C 28 02 00 00 48 
[20:29:19][I][vista_alarm:947]: Partition: 01
[20:29:19][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'HOUSE           '
[20:29:19][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'CHIME,Rdy to Arm'
[20:29:19][I][vista_alarm:971]: Prompt: HOUSE           
[20:29:19][I][vista_alarm:972]: Prompt: CHIME,Rdy to Arm
[20:29:19][I][vista_alarm:973]: Beeps: 0
[20:29:22][I][CMD:572]: 2024-05-15 20:29  F2 12 66 00 00 00 00 66 6C 02 45 6C F5 EC 01 01 01 00 00 2D 
[20:29:22][I][CMD:572]: 2024-05-15 20:29  F2 16 66 00 00 00 00 67 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 7F 
[20:29:22][I][CMD:572]: 2024-05-15 20:29  F7 02 00 1D 10 01 03 00 28 02 00 00 46 
[20:29:22][I][vista_alarm:947]: Partition: 01
[20:29:22][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 01 FRONT  '
[20:29:22][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[20:29:22][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '3'
[20:29:22][I][vista_alarm:971]: Prompt: FAULT 01 FRONT  
[20:29:22][I][vista_alarm:972]: Prompt: DOOR            
[20:29:22][I][vista_alarm:973]: Beeps: 3
[20:29:22][D][binary_sensor:036]: 'Front Door': Sending state ON
[20:29:22][D][binary_sensor:036]: 'Ready (rdy_1)': Sending state OFF
[20:29:22][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:1'
[20:29:22][W][component:237]: Component vista_alarm_panel took a long time for an operation (64 ms).
[20:29:22][W][component:238]: Components should block for at most 30 ms.
[20:29:22][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[20:29:22][I][CHK:572]: 2024-05-15 20:29  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[20:29:22][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 56 FE FE EC 03 02 01 06 B4 
[20:29:25][I][CMD:572]: 2024-05-15 20:29  F2 12 66 00 00 00 00 60 6C 02 45 6C F5 EC 01 01 01 00 00 33 
[20:29:25][I][CMD:572]: 2024-05-15 20:29  F2 16 66 00 00 00 00 61 63 02 45 43 F5 31 FB 45 6C F5 EC 03 01 01 01 8B 
[20:29:25][I][CMD:572]: 2024-05-15 20:29  F7 02 00 1D 10 08 00 1C 28 02 00 00 48 
[20:29:25][I][vista_alarm:947]: Partition: 01
[20:29:25][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'HOUSE           '
[20:29:25][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'CHIME,Rdy to Arm'
[20:29:25][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[20:29:25][I][vista_alarm:971]: Prompt: HOUSE           
[20:29:25][I][vista_alarm:972]: Prompt: CHIME,Rdy to Arm
[20:29:25][I][vista_alarm:973]: Beeps: 0
[20:29:25][D][binary_sensor:036]: 'Ready (rdy_1)': Sending state ON
[20:29:25][D][binary_sensor:036]: 'Front Door': Sending state OFF
[20:29:25][D][text_sensor:064]: 'Zone Status (zs)': Sending state ''
[20:29:25][W][component:237]: Component vista_alarm_panel took a long time for an operation (67 ms).
[20:29:25][W][component:238]: Components should block for at most 30 ms.
[20:29:25][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[20:29:25][I][CHK:572]: 2024-05-15 20:29  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[20:29:25][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 57 FE FE EC 03 01 01 01 B9 
[20:29:27][I][CMD:572]: 2024-05-15 20:29  F7 02 00 1D 10 01 03 00 28 02 00 00 46 
[20:29:27][I][vista_alarm:947]: Partition: 01
[20:29:27][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 01 FRONT  '
[20:29:27][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[20:29:27][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '3'
[20:29:27][I][vista_alarm:971]: Prompt: FAULT 01 FRONT  
[20:29:27][I][vista_alarm:972]: Prompt: DOOR            
[20:29:27][I][vista_alarm:973]: Beeps: 3
[20:29:27][D][binary_sensor:036]: 'Front Door': Sending state ON
[20:29:27][D][binary_sensor:036]: 'Ready (rdy_1)': Sending state OFF
[20:29:27][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:1'
[20:29:27][W][component:237]: Component vista_alarm_panel took a long time for an operation (108 ms).
[20:29:27][W][component:238]: Components should block for at most 30 ms.
[20:29:27][I][CMD:572]: 2024-05-15 20:29  F2 12 66 00 00 00 00 62 6C 02 45 6C F5 EC 01 01 01 00 00 31 
[20:29:28][I][CMD:572]: 2024-05-15 20:29  F2 16 66 00 00 00 00 63 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 83 
[20:29:28][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[20:29:28][I][CHK:572]: 2024-05-15 20:29  21 00 00 00 00 00 00 00 00 00 00 00 90 
[20:29:28][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 50 FE FE EC 03 02 01 06 BA 
[20:29:28][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[20:29:28][I][CHK:572]: 2024-05-15 20:29  61 00 00 00 00 00 00 00 00 00 00 00 90 
[20:29:28][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 51 FE FE EC 03 02 01 06 B9 
[20:29:30][I][CMD:572]: 2024-05-15 20:29  F2 12 66 00 00 00 00 64 6C 02 45 6C F5 EC 01 01 01 00 00 2F 
[... 74 lines cut ...]
[20:29:37][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
[20:29:39][W][component:238]: Components should block for at most 30 ms.
[20:29:39][I][CMD:572]: 2024-05-15 20:29  F2 16 66 00 00 00 00 65 63 02 45 43 F5 31 FB 45 6C F5 EC 03 01 01 01 87 
[20:29:39][I][CMD:572]: 2024-05-15 20:29  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[20:29:39][I][CHK:572]: 2024-05-15 20:29  21 00 00 00 00 00 00 00 00 00 00 00 90 
[20:29:39][I][CMD:572]: 2024-05-15 20:29  F2 0E 02 00 00 00 00 50 FE FE EC 03 01 01 01 C0 
[20:29:43][I][CMD:572]: 2024-05-15 20:29  F7 02 00 1D 10 08 00 1C 28 02 00 00 48

Dilbert66, 2024-05-16:
Thank you ! It looks like you are missing the module data side of it.  Do you have the monitor line hooked up on your circuit? This is the "monitorpin" config in the yaml.  It is not showing data there. For instance it should show data sent from the keypads/expanders/etc to the panel.  This is very useful for capturing expander open/close events to immediately close zones that are hooked up to the expanders or RF modules.   You should be seeing EXT:xxx messages when the line is connected.

rwoldberg, 2024-05-16:
I do have it connected, but maybe it is not connected correctly, I will take a look at my board.

rwoldberg, 2024-05-17:
I had the wire on the wrong pin, oops, here is the output from opening and closing the front door multiple times: 

[07:42:25][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:42:25][I][vista_alarm:971]: Prompt: FAULT 06 SOUTH  
[07:42:25][I][vista_alarm:972]: Prompt: GUEST ROOM      
[07:42:25][I][vista_alarm:973]: Beeps: 0
[07:42:25][I][CMD:572]: 2024-05-17 07:42  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:42:25][I][EXT:572]: 2024-05-17 07:42  F6 01 A1 0C 00 6E 62 02 45 43 F5 31 FB 
[07:42:25][I][CHK:572]: 2024-05-17 07:42  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:42:25][I][CMD:572]: 2024-05-17 07:42  F2 0E 02 00 00 00 00 56 FE FE EC 03 02 01 06 B4 
[07:42:28][I][CMD:572]: 2024-05-17 07:42  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:42:28][I][CHK:572]: 2024-05-17 07:42  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:42:28][I][EXT:572]: 2024-05-17 07:42  F6 01 E1 07 00 6F 05 02 43 43 1C 00 00 
[07:42:28][W][component:237]: Component vista_alarm_panel took a long time for an operation (53 ms).
[07:42:28][W][component:238]: Components should block for at most 30 ms.
[07:42:28][I][CMD:572]: 2024-05-17 07:42  F2 16 02 00 00 00 00 57 FE EC 32 34 30 35 31 37 30 37 34 32 30 32 35 1E 
[07:42:29][I][CMD:572]: 2024-05-17 07:42  F2 12 66 00 00 00 00 60 6C 02 45 6C F5 EC 01 01 01 00 00 33 
[07:42:29][I][CMD:572]: 2024-05-17 07:42  F2 16 66 00 00 00 00 61 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 85 
[07:42:29][I][CMD:572]: 2024-05-17 07:42  F7 02 00 1D 10 03 00 00 28 02 00 00 46 
[07:42:29][I][vista_alarm:947]: Partition: 01
[07:42:29][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 03 MOTION '
[07:42:29][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DETECTOR        '
[07:42:29][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:42:29][I][vista_alarm:971]: Prompt: FAULT 03 MOTION 
[07:42:29][I][vista_alarm:972]: Prompt: DETECTOR        
[07:42:29][I][vista_alarm:973]: Beeps: 0
[07:42:29][I][CMD:572]: 2024-05-17 07:42  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:42:29][I][EXT:572]: 2024-05-17 07:42  F6 01 21 00 68 62 02 45 43 F5 31 FB 43 
[07:42:29][I][CHK:572]: 2024-05-17 07:42  21 00 00 00 00 00 00 00 00 00 00 00 90 
[07:42:29][I][CMD:572]: 2024-05-17 07:42  F2 0E 02 00 00 00 00 50 FE FE EC 03 02 01 06 BA 
[07:42:32][I][CMD:572]: 2024-05-17 07:42  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:42:32][I][CHK:572]: 2024-05-17 07:42  61 00 00 00 00 00 00 00 00 00 00 00 90 
[07:42:32][I][EXT:572]: 2024-05-17 07:42  F6 01 61 07 00 69 6B 02 43 43 3C 00 00 
[07:42:32][I][CMD:572]: 2024-05-17 07:42  F2 0E 02 00 00 00 00 51 FE EC 01 01 01 00 00 C0 
[07:42:33][I][CMD:572]: 2024-05-17 07:42  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[07:42:33][I][vista_alarm:947]: Partition: 01
[07:42:33][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[07:42:33][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[07:42:33][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:42:33][I][vista_alarm:971]: Prompt: FAULT 06 SOUTH  
[07:42:33][I][vista_alarm:972]: Prompt: GUEST ROOM      
[07:42:33][I][vista_alarm:973]: Beeps: 0
[07:42:33][D][binary_sensor:036]: 'Deck Sliding Door': Sending state OFF
[07:42:33][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:3,OP:6,OP:10,OP:11'
[07:42:33][W][component:237]: Component vista_alarm_panel took a long time for an operation (61 ms).
[07:42:33][W][component:238]: Components should block for at most 30 ms.
[07:42:33][I][CMD:572]: 2024-05-17 07:42  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:42:33][I][CHK:572]: 2024-05-17 07:42  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:42:33][I][EXT:572]: 2024-05-17 07:42  F6 01 A1 0C 00 6A 62 02 45 43 F5 31 FB 
[07:42:33][I][CMD:572]: 2024-05-17 07:42  F2 0E 00 00 00 00 52 FE FE EC 03 02 77 06 B8 00 
[07:42:33][I][CMD:572]: 2024-05-17 07:42  F2 12 66 00 00 00 00 62 6C 02 45 6C F5 EC 01 01 01 00 00 31 
[07:42:33][I][CMD:572]: 2024-05-17 07:42  F2 16 66 00 00 00 00 63 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 83 
[07:42:33][I][CMD:572]: 2024-05-17 07:42  F7 02 00 1D 10 10 00 00 28 02 00 00 46 
[07:42:33][I][vista_alarm:947]: Partition: 01
[07:42:33][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 10 MASTER '
[07:42:33][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'BEDROOM         '
[07:42:33][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:42:33][I][vista_alarm:971]: Prompt: FAULT 10 MASTER 
[07:42:33][I][vista_alarm:972]: Prompt: BEDROOM         
[07:42:33][I][vista_alarm:973]: Beeps: 0
[07:42:34][I][CMD:572]: 2024-05-17 07:42  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:42:34][I][CHK:572]: 2024-05-17 07:42  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:42:34][I][EXT:572]: 2024-05-17 07:42  F6 01 E1 00 6B 62 02 45 43 F5 31 FB 43 
[07:42:34][I][CMD:572]: 2024-05-17 07:42  F2 0E 02 00 00 00 00 53 FE FE EC 03 02 01 06 B7 
[07:42:36][I][CMD:572]: 2024-05-17 07:42  F2 12 66 00 00 00 00 64 6C 02 45 6C F5 EC 01 01 01 00 00 2F 
[07:42:36][I][CMD:572]: 2024-05-17 07:42  F7 02 00 1D 10 01 03 00 28 02 00 00 46 
[07:42:36][I][vista_alarm:947]: Partition: 01
[07:42:36][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 01 FRONT  '
[07:42:36][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[07:42:36][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '3'
[... 160 lines cut ...]
[07:43:01][D][vista_alarm:797]: Force refresh....
[07:43:01][I][CMD:572]: 2024-05-17 07:43  F7 02 00 1D 10 10 00 00 28 02 00 00 46 
[... cut ...]
[07:43:02][I][CMD:572]: 2024-05-17 07:43  F2 0E 02 00 00 00 00 51 FE EC 01 01 01 00 00 C0 
[07:43:02][I][CMD:572]: 2024-05-17 07:43  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:43:03][I][CHK:572]: 2024-05-17 07:43  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:43:03][I][EXT:572]: 2024-05-17 07:43  F6 01 A1 0C 00 6A 62 02 45 43 F5 31 FB 
[07:43:03][I][CMD:572]: 2024-05-17 07:43  F2 0E 02 00 00 00 00 52 FE FE EC 03 02 01 06 B8 
[07:43:05][I][CMD:572]: 2024-05-17 07:43  F7 02 00 1D 10 11 00 00 28 02 00 00 46

Dilbert66, 2024-05-17:
Awesome tks! Can I ask you another favor? I see that my f6 cmd log size is too small. I've updated the code to fully display the cmd so that we can capture the data sent from the display fully.  If you look  in your log, the EXT f6 cmds are what the keypad sends to the panel and the F2's are the resultant responses back.  Can you recompile to pick up the latest update and do this log again? Much appreciated.  I'm hoping to capture the cmds related to zone status.

rwoldberg, 2024-05-18:
Here is the log with the latest code:
[11:53:03][W][component:237]: Component vista_alarm_panel took a long time for an operation (106 ms).
[11:53:03][W][component:238]: Components should block for at most 30 ms.
[11:53:03][I][CMD:588]: 2024-05-18 11:53  F2 0E 02 00 00 00 50 FE FE EC 03 01 01 01 C0 00 
[11:53:05][I][CMD:588]: 2024-05-18 11:53  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[11:53:05][I][EXT:588]: 2024-05-18 11:53  F6 01 61 07 00 69 6B 02 43 43 3C 00 00 
[11:53:05][I][CHK:588]: 2024-05-18 11:53  61 00 00 00 00 00 00 00 00 00 00 00 90 
[11:53:06][I][CMD:588]: 2024-05-18 11:53  F2 0E 02 00 00 00 00 51 FE EC 01 01 01 00 00 C0 
[11:53:06][I][CMD:588]: 2024-05-18 11:53  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[11:53:07][I][CHK:588]: 2024-05-18 11:53  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[11:53:07][I][EXT:588]: 2024-05-18 11:53  F6 01 A1 0C 00 6A 62 02 45 43 F5 31 FB 
[11:53:07][I][CMD:588]: 2024-05-18 11:53  F2 0E 02 00 00 00 00 52 FE FE EC 03 01 01 01 BE 
[11:53:07][I][CMD:588]: 2024-05-18 11:53  F7 02 00 1D 10 08 00 1C 28 02 00 00 48 
[11:53:07][I][vista_alarm:963]: Partition: 01
[11:53:07][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'HOUSE           '
[11:53:07][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'CHIME,Rdy to Arm'
[11:53:07][I][vista_alarm:987]: Prompt: HOUSE           
[11:53:07][I][vista_alarm:988]: Prompt: CHIME,Rdy to Arm
[11:53:07][I][vista_alarm:989]: Beeps: 0
[11:53:11][I][CMD:588]: 2024-05-18 11:53  F2 12 66 00 00 00 00 62 6C 02 45 6C F5 EC 01 01 01 00 00 31 
[11:53:11][I][CMD:588]: 2024-05-18 11:53  F2 16 66 00 00 00 00 63 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 83 
[11:53:11][I][CMD:588]: 2024-05-18 11:53  F7 02 00 1D 10 03 00 00 28 02 00 00 46 
[11:53:11][I][vista_alarm:963]: Partition: 01
[11:53:11][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 03 MOTION '
[11:53:11][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DETECTOR        '
[11:53:11][I][vista_alarm:987]: Prompt: FAULT 03 MOTION 
[11:53:11][I][vista_alarm:988]: Prompt: DETECTOR        
[11:53:11][I][vista_alarm:989]: Beeps: 0
[11:53:11][D][binary_sensor:036]: 'Living Room Motion Detector': Sending state ON
[11:53:11][D][binary_sensor:036]: 'Ready (rdy_1)': Sending state OFF
[11:53:11][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:3'
[11:53:11][W][component:237]: Component vista_alarm_panel took a long time for an operation (64 ms).
[11:53:11][W][component:238]: Components should block for at most 30 ms.
[11:53:11][I][CMD:588]: 2024-05-18 11:53  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[11:53:11][I][EXT:588]: 2024-05-18 11:53  F6 01 E1 00 6B 62 02 45 43 F5 31 FB 43 
[11:53:11][I][CHK:588]: 2024-05-18 11:53  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[11:53:11][I][CMD:588]: 2024-05-18 11:53  F2 0E 02 00 00 00 00 53 FE FE EC 03 02 01 06 B7 
[11:53:15][I][CMD:588]: 2024-05-18 11:53  F7 02 00 1D 10 03 00 00 28 02 00 00 46 
[11:53:15][I][vista_alarm:963]: Partition: 01
[11:53:15][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 03 MOTION '
[11:53:15][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DETECTOR        '
[11:53:15][I][vista_alarm:987]: Prompt: FAULT 03 MOTION 
[11:53:15][I][vista_alarm:988]: Prompt: DETECTOR        
[11:53:15][I][vista_alarm:989]: Beeps: 0
[11:53:16][D][vista_alarm:764][cmdQueueTask]: Taskupdates free memory:  1164

[11:53:16][I][CMD:588]: 2024-05-18 11:53  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[11:53:16][I][CHK:588]: 2024-05-18 11:53  21 00 00 00 00 00 00 00 00 00 00 00 90 
[11:53:16][I][EXT:588]: 2024-05-18 11:53  F6 01 21 0C 00 6C 62 02 45 43 F5 31 FB 
[11:53:16][I][CMD:588]: 2024-05-18 11:53  F2 0E 02 00 00 00 00 54 FE FE EC 03 02 01 06 B6 
[11:53:17][I][CMD:588]: 2024-05-18 11:53  F7 02 00 1D 10 01 03 00 28 02 00 00 46 
[11:53:17][I][vista_alarm:963]: Partition: 01
[11:53:17][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 01 FRONT  '
[11:53:17][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[11:53:17][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '3'
[11:53:17][I][vista_alarm:987]: Prompt: FAULT 01 FRONT  
[11:53:17][I][vista_alarm:988]: Prompt: DOOR            
[11:53:17][I][vista_alarm:989]: Beeps: 3
[11:53:17][D][binary_sensor:036]: 'Front Door': Sending state ON
[11:53:17][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:1,OP:3'
[11:53:18][W][component:237]: Component vista_alarm_panel took a long time for an operation (62 ms).
[11:53:18][W][component:238]: Components should block for at most 30 ms.
[11:53:18][I][CMD:588]: 2024-05-18 11:53  F2 12 66 00 00 00 00 64 6C 02 45 6C F5 EC 01 01 01 00 00 2F 
[11:53:18][I][CMD:588]: 2024-05-18 11:53  F2 16 66 00 00 00 00 65 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 81 
[11:53:18][I][CMD:588]: 2024-05-18 11:53  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[11:53:18][I][EXT:588]: 2024-05-18 11:53  F6 01 61 00 6D 62 02 45 43 F5 31 FB 43 
[11:53:18][I][CHK:588]: 2024-05-18 11:53  61 00 00 00 00 00 00 00 00 00 00 00 90 
[11:53:18][I][CMD:588]: 2024-05-18 11:53  F7 02 00 1D 10 01 00 00 28 02 00 00 46 
[11:53:18][I][vista_alarm:963]: Partition: 01
[11:53:18][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 01 FRONT  '
[... 294 lines cut ...]
[11:54:04][I][vista_alarm:963]: Partition: 01
[11:54:04][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 03 MOTION '
[11:54:04][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DETECTOR        '
[11:54:04][I][vista_alarm:987]: Prompt: FAULT 03 MOTION 
[... cut ...]
[11:54:05][I][CMD:588]: 2024-05-18 11:54  F2 0E 02 00 00 00 00 50 FE FE EC 03 02 01 06 BA 
[11:54:05][I][CMD:588]: 2024-05-18 11:54  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[11:54:05][I][EXT:588]: 2024-05-18 11:54  F6 01 61 0C 00 69 62 02 45 43 F5 31 FB 
[11:54:05][I][CHK:588]: 2024-05-18 11:54  61 00 00 00 00 00 00 00 00 00 00 00 90 
[11:54:06][I][CMD:588]: 2024-05-18 11:54  F2 0E 02 00 00 00 00 51 FE FE EC 03 02 01 06 B9 
[11:54:08][I][CMD:588]: 2024-05-18 11:54  F7 02 00 1D 10 03 00 00 28 02 00 00 46 
[11:54:08][I][vista_alarm:963]: Partition: 01

Dilbert66, 2024-05-18:
Tks again! Well, from what I see, there are no queries for specific zone statuses.  All I see are periodic system status queries.  I guess the 6270's don't get zone close events either.   I'll look into it more.

rwoldberg, 2024-05-18:
Interesting, I wonder how it can get away with showing the zone as closed in a little less than two seconds.  Maybe the controller sends updates to it faster than other keypads?

Dilbert66, 2024-05-19:
Are you saying for example that you have two zones open and one closes, you get instant notification that this zone is closed? If there is only one zone open then closes, that's different as the system will send a ready flag and the keypad will close the zone immediately.  With two zones open and one closes, the system will not send any status change as it is still not ready so  as far as I can tell there is no individual zone notification.  I've never seen a 6270 so not aware of how it displays individual zone statuses. 
From what I see in the logs there is no zone info as far as I can see in the f2 cmds. Only system status which puzzles me.

Dilbert66, 2024-05-19:
Now, I do know that the keypad can retrieve zone statuses when asked via menu but that is a different process as it will send a cmd to the panel for it to send a zone update. It does not do that on it's own as far as I know.

rwoldberg, 2024-05-19:
This is the screen on the keypad that shows the open zones.
[image]
So maybe it is requesting the zone statuses, I could get a log of the output with this screen up on the keypad.  Would it be possible to do the same thing from the emulated keypad?  Maybe have an option for how often to request the status?

Dilbert66, 2024-05-19:
Yes if you could log that it would be great!

Dilbert66, 2024-05-19:
Fyi, whenever a zone or system status updates, there are 2 F2 system status messages sent out. Nothing zone related that I can see. Mostly system flags.   But those F2 cmds can be used to trigger a zone status update request cmd to do exactly what is needed to update any zone status immediately.  All I need is that zone status request cmd.

rwoldberg, 2024-05-19:
Here is the log with the status page open on the keypad, hopefully this has the data you need:
[17:11:42][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[17:11:42][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[17:11:42][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[17:11:42][I][vista_alarm:987]: Prompt: FAULT 06 SOUTH  
[17:11:42][I][vista_alarm:988]: Prompt: GUEST ROOM      
[17:11:42][I][vista_alarm:989]: Beeps: 0
[17:11:43][I][CMD:588]: 2024-05-19 17:11  FA 01 02 25 F7 E7 00 00 00 00 00 00 00 
[17:11:43][I][EXT:588]: 2024-05-19 17:11  FA 07 F7 00 70 00 00 00 00 00 00 00 00 
[17:11:43][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:43][I][CHK:588]: 2024-05-19 17:11  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:43][I][EXT:588]: 2024-05-19 17:11  F6 01 A1 0C 00 6E 62 02 45 43 F5 31 FB 
[17:11:43][I][CMD:588]: 2024-05-19 17:11  F2 0E 02 00 00 00 00 56 FE FE EC 03 02 01 06 B4 
[17:11:45][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:46][I][EXT:588]: 2024-05-19 17:11  F6 01 E1 16 00 6F 62 31 45 49 F5 31 FB 
[17:11:46][I][CHK:588]: 2024-05-19 17:11  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:46][I][CMD:588]: 2024-05-19 17:11  F2 13 02 00 00 00 00 57 FE FE FE FE EC 36 00 31 30 2D 31 32 97 
[17:11:46][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:46][I][CHK:588]: 2024-05-19 17:11  21 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:46][I][EXT:588]: 2024-05-19 17:11  F6 01 21 16 00 68 62 31 45 49 F5 31 FB 
[17:11:46][I][CMD:588]: 2024-05-19 17:11  F2 0B 02 00 00 00 00 50 FE FE FE FE B9 
[17:11:46][I][CMD:588]: 2024-05-19 17:11  F7 02 00 1D 10 10 00 00 28 02 00 00 46 
[17:11:46][I][vista_alarm:963]: Partition: 01
[17:11:46][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 10 MASTER '
[17:11:46][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'BEDROOM         '
[17:11:46][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[17:11:46][I][vista_alarm:987]: Prompt: FAULT 10 MASTER 
[17:11:46][I][vista_alarm:988]: Prompt: BEDROOM         
[17:11:46][I][vista_alarm:989]: Beeps: 0
[17:11:48][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:48][I][EXT:588]: 2024-05-19 17:11  F6 01 61 16 00 69 62 31 45 49 F5 31 FB 
[17:11:48][I][CHK:588]: 2024-05-19 17:11  61 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:48][I][CMD:588]: 2024-05-19 17:11  F2 13 02 00 00 00 00 51 FE FE FE FE EC 36 00 31 30 2D 31 32 9D 
[17:11:48][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:48][I][CHK:588]: 2024-05-19 17:11  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:49][I][EXT:588]: 2024-05-19 17:11  F6 01 A1 16 00 6A 62 31 45 49 F5 31 FB 
[17:11:49][I][CMD:588]: 2024-05-19 17:11  F2 0B 02 00 00 00 00 52 FE FE FE FE B7 
[17:11:49][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:49][I][CHK:588]: 2024-05-19 17:11  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:49][I][EXT:588]: 2024-05-19 17:11  F6 01 E1 0C 00 6B 62 02 45 43 F5 31 FB 
[17:11:49][I][CMD:588]: 2024-05-19 17:11  F2 0E 02 00 00 00 00 53 FE FE EC 03 02 01 06 B7 
[17:11:50][I][CMD:588]: 2024-05-19 17:11  F7 02 00 1D 10 11 00 00 28 02 00 00 46 
[17:11:50][I][vista_alarm:963]: Partition: 01
[17:11:50][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 11        '
[17:11:50][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'BASEMENT SLIDING'
[17:11:50][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[17:11:50][I][vista_alarm:987]: Prompt: FAULT 11        
[17:11:50][I][vista_alarm:988]: Prompt: BASEMENT SLIDING
[17:11:50][I][vista_alarm:989]: Beeps: 0
[17:11:51][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:51][I][CHK:588]: 2024-05-19 17:11  21 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:51][I][EXT:588]: 2024-05-19 17:11  F6 01 21 16 00 6C 62 31 45 49 F5 31 FB 
[17:11:51][I][CMD:588]: 2024-05-19 17:11  F2 13 02 00 00 00 00 54 FE FE FE FE EC 36 00 31 30 2D 31 32 9A 
[17:11:51][I][CMD:588]: 2024-05-19 17:11  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[17:11:51][I][CHK:588]: 2024-05-19 17:11  61 00 00 00 00 00 00 00 00 00 00 00 90 
[17:11:51][I][EXT:588]: 2024-05-19 17:11  F6 01 61 16 00 6D 62 31 45 49 F5 31 FB 
[17:11:52][I][CMD:588]: 2024-05-19 17:11  F2 0B 02 00 00 00 00 55 FE FE FE FE B4 
[17:11:54][I][CMD:588]: 2024-05-19 17:11  F7 02 00 1D 10 02 03 00 28 02 00 00 46 
[17:11:54][I][vista_alarm:963]: Partition: 01
[17:11:54][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 02 GARAGE '
[17:11:54][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[17:11:54][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '3'
[17:11:54][I][vista_alarm:987]: Prompt: FAULT 02 GARAGE 
[17:11:54][I][vista_alarm:988]: Prompt: DOOR            
[17:11:54][I][vista_alarm:989]: Beeps: 3
[17:11:54][D][binary_sensor:036]: 'Kitchen Door': Sending state ON
[17:11:54][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:2,OP:6,OP:10,OP:11,OP:12'
[17:11:54][W][component:237]: Component vista_alarm_panel took a long time for an operation (61 ms).
[17:11:54][W][component:238]: Components should block for at most 30 ms.
[17:11:54][I][CMD:588]: 2024-05-19 17:11  F2 12 66 00 00 00 00 67 6C 02 45 6C F5 EC 01 01 01 00 00 2C 
[... 242 lines cut ...]
[17:12:24][I][vista_alarm:987]: Prompt: FAULT 12 SLIDING
[17:12:24][I][vista_alarm:988]: Prompt: DOOR            
[... cut ...]
[17:12:27][I][EXT:588]: 2024-05-19 17:12  F6 01 21 16 00 68 62 31 45 49 F5 31 FB 
[17:12:27][I][CMD:588]: 2024-05-19 17:12  F2 0B 02 00 00 00 00 50 FE FE FE FE B9 
[17:12:27][I][CMD:588]: 2024-05-19 17:12  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[17:12:27][I][vista_alarm:963]: Partition: 01
[17:12:27][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[17:12:27][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[17:12:27][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[17:12:28][I][vista_alarm:987]: Prompt: FAULT 06 SOUTH

Dilbert66, 2024-05-20:
Tks again. Did you do a full recompile and re-upload? The code is still not showing the F6 length changes for the F2 cmds so the new code has not been picked up on your system..  They are still truncated at 13 characters.

rwoldberg, 2024-05-20:
I did a clean and then a reinstall, that should have downloaded the new code.  I will try again.

rwoldberg, 2024-05-20:
I reinstalled and here is the output:
[07:51:18][I][CMD:589]: 2024-05-20 07:51  FA 01 02 20 F7 EC 00 00 00 00 00 00 00 
[07:51:18][I][EXT:589]: 2024-05-20 07:51  FA 07 F7 00 70 00 00 00 00 00 00 00 00 
[07:51:20][I][CMD:589]: 2024-05-20 07:51  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:51:20][I][CHK:589]: 2024-05-20 07:51  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:51:20][I][EXT:589]: 2024-05-20 07:51  F6 01 A1 0C 00 6E 62 02 45 43 F5 31 FB 43 6C 29 
[07:51:20][I][CMD:589]: 2024-05-20 07:51  F2 0E 02 00 00 00 00 56 FE FE EC 03 02 01 06 B4 
[07:51:20][I][CMD:589]: 2024-05-20 07:51  F7 02 00 1D 10 12 00 00 28 02 00 00 46 
[07:51:20][I][vista_alarm:967]: Partition: 01
[07:51:20][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 12 SLIDING'
[07:51:20][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[07:51:20][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:51:20][I][vista_alarm:991]: Prompt: FAULT 12 SLIDING
[07:51:20][I][vista_alarm:992]: Prompt: DOOR            
[07:51:20][I][vista_alarm:993]: Beeps: 0
[07:51:24][I][CMD:589]: 2024-05-20 07:51  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[07:51:24][I][vista_alarm:967]: Partition: 01
[07:51:24][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[07:51:24][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[07:51:24][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:51:24][I][vista_alarm:991]: Prompt: FAULT 06 SOUTH  
[07:51:24][I][vista_alarm:992]: Prompt: GUEST ROOM      
[07:51:24][I][vista_alarm:993]: Beeps: 0
[07:51:28][I][CMD:589]: 2024-05-20 07:51  F7 02 00 1D 10 10 00 00 28 02 00 00 46 
[07:51:28][I][vista_alarm:967]: Partition: 01
[07:51:28][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 10 MASTER '
[07:51:28][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'BEDROOM         '
[07:51:28][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:51:28][I][vista_alarm:991]: Prompt: FAULT 10 MASTER 
[07:51:28][I][vista_alarm:992]: Prompt: BEDROOM         
[07:51:28][I][vista_alarm:993]: Beeps: 0
[07:51:30][D][vista_alarm:765][cmdQueueTask]: Taskupdates free memory:  1472

[07:51:30][I][CMD:589]: 2024-05-20 07:51  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:51:30][I][CHK:589]: 2024-05-20 07:51  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[07:51:30][I][EXT:589]: 2024-05-20 07:51  F6 01 E1 0C 00 6F 62 02 45 43 F5 31 FB 43 6C E8 
[07:51:30][I][CMD:589]: 2024-05-20 07:51  F2 0E 02 00 00 00 00 57 FE FE EC 03 02 01 06 B3 
[07:51:32][I][CMD:589]: 2024-05-20 07:51  F7 02 00 1D 10 11 00 00 28 02 00 00 46 
[07:51:32][I][vista_alarm:967]: Partition: 01
[07:51:32][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 11        '
[07:51:32][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'BASEMENT SLIDING'
[07:51:32][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:51:32][I][vista_alarm:991]: Prompt: FAULT 11        
[07:51:32][I][vista_alarm:992]: Prompt: BASEMENT SLIDING
[07:51:32][I][vista_alarm:993]: Beeps: 0
[07:51:36][I][CMD:589]: 2024-05-20 07:51  F7 02 00 1D 10 12 00 00 28 02 00 00 46 
[07:51:36][I][vista_alarm:967]: Partition: 01
[07:51:36][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 12 SLIDING'
[07:51:36][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DOOR            '
[07:51:36][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:51:36][I][vista_alarm:991]: Prompt: FAULT 12 SLIDING
[07:51:36][I][vista_alarm:992]: Prompt: DOOR            
[07:51:36][I][vista_alarm:993]: Beeps: 0
[07:51:39][I][CMD:589]: 2024-05-20 07:51  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:51:39][I][EXT:589]: 2024-05-20 07:51  F6 01 21 07 00 68 6B 02 43 43 7D 
[07:51:39][I][CHK:589]: 2024-05-20 07:51  21 00 00 00 00 00 00 00 00 00 00 00 90 
[07:51:39][I][CMD:589]: 2024-05-20 07:51  F2 0E 02 00 00 00 00 50 FE EC 01 01 01 00 00 C1 
[07:51:40][I][CMD:589]: 2024-05-20 07:51  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:51:40][I][CHK:589]: 2024-05-20 07:51  61 00 00 00 00 00 00 00 00 00 00 00 90 
[07:51:40][I][EXT:589]: 2024-05-20 07:51  F6 01 61 0C 00 69 62 02 45 43 F5 31 FB 43 6C 6E 
[07:51:40][I][CMD:589]: 2024-05-20 07:51  F2 0E 00 00 00 51 FE FE EC 03 02 01 77 B9 00 00 
[07:51:40][I][CMD:589]: 2024-05-20 07:51  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[07:51:40][I][vista_alarm:967]: Partition: 01
[07:51:40][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[07:51:40][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[07:51:40][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[07:51:40][I][vista_alarm:991]: Prompt: FAULT 06 SOUTH  
[07:51:40][I][vista_alarm:992]: Prompt: GUEST ROOM      
[07:51:40][I][vista_alarm:993]: Beeps: 0
[07:51:44][I][CMD:589]: 2024-05-20 07:51  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[... 218 lines cut ...]
[07:52:06][I][vista_alarm:991]: Prompt: FAULT 10 MASTER 
[07:52:06][I][vista_alarm:992]: Prompt: BEDROOM         
[07:52:06][I][vista_alarm:993]: Beeps: 0
[07:52:08][I][CMD:589]: 2024-05-20 07:52  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[07:52:08][I][EXT:589]: 2024-05-20 07:52  F6 01 61 07 00 6D 6B 02 43 43 38 
[07:52:08][I][CHK:589]: 2024-05-20 07:52  61 00 00 00 00 00 00 00 00 00 00 00 90 
[07:52:08][I][CMD:589]: 2024-05-20 07:52  F2 0E 02 00 00 00 00 55 FE EC 01 01 01 00 00 BC 
[07:52:09][I][CMD:589]: 2024-05-20 07:52  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
[07:52:09][I][CHK:589]: 2024-05-20 07:52  21 00 00 00 00 00 00 00 00 00 00 00 90 
[07:52:09][I][CMD:589]: 2024-05-20 07:52  F2 0E 02 00 00 00 00 50 FE FE EC 03 02 01 06 BA 
[07:52:10][D][vista_alarm:814]: Force refresh....
[07:52:10][I][CMD:589]: 2024-05-20 07:52  F7 02 00 1D 10 11 00 00 28 02 00 00 46 
[07:52:10][I][vista_alarm:967]: Partition: 01
[07:52:10][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 11        '
[07:52:10][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'BASEMENT SLIDING'
[07:52:10][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'

Dilbert66, 2024-05-20:
Perfect! Thank you. Ok I'll do some testing with this data and see what I come up with!

Dilbert66, 2024-05-21:
Ok, while going through your log, I saw that I had a bug in my F6 cmd handling and had the buffer too small so was loosing the last half of the data.  Unfortunately, this means I need you to recompile and rerun the log so we can capture this data again to get the missing info.   Sorry about that!  I appreciate you help on this! We are getting closer to the goal!

rwoldberg, 2024-05-21:
Try this one: 
[08:55:59][W][component:237]: Component vista_alarm_panel took a long time for an operation (65 ms).
[08:55:59][W][component:238]: Components should block for at most 30 ms.
[08:56:00][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:00][I][EXT:590]: 2024-05-21 08:56  F6 01 E1 07 00 6F 6B 02 43 43 B6 
[08:56:00][I][CHK:590]: 2024-05-21 08:56  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:00][I][CMD:590]: 2024-05-21 08:56  F2 0E 02 00 00 00 00 57 FE EC 01 01 01 00 00 BA 
[08:56:01][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:01][I][EXT:590]: 2024-05-21 08:56  F6 01 21 0C 00 68 62 02 45 43 F5 31 FB 43 6C AF 
[08:56:01][I][CHK:590]: 2024-05-21 08:56  21 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:01][I][CMD:590]: 2024-05-21 08:56  F2 0E 02 00 00 00 00 50 FE FE EC 03 02 01 06 BA 
[08:56:03][I][CMD:590]: 2024-05-21 08:56  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[08:56:03][I][vista_alarm:968]: Partition: 01
[08:56:03][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[08:56:03][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[08:56:03][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[08:56:03][I][vista_alarm:992]: Prompt: FAULT 06 SOUTH  
[08:56:03][I][vista_alarm:993]: Prompt: GUEST ROOM      
[08:56:03][I][vista_alarm:994]: Beeps: 0
[08:56:07][D][vista_alarm:766][cmdQueueTask]: Taskupdates free memory:  1248

[08:56:07][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[08:56:07][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[08:56:07][I][vista_alarm:992]: Prompt: FAULT 06 SOUTH  
[08:56:07][I][vista_alarm:993]: Prompt: GUEST ROOM      
[08:56:07][I][vista_alarm:994]: Beeps: 0
[08:56:08][I][CMD:590]: 2024-05-21 08:56  F2 12 66 00 00 00 00 67 6C 02 45 6C F5 EC 01 01 01 00 00 2C 
[08:56:09][I][CMD:590]: 2024-05-21 08:56  F2 16 66 00 00 00 00 60 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 86 
[08:56:09][I][CMD:590]: 2024-05-21 08:56  F7 02 00 1D 10 03 00 00 28 02 00 00 46 
[08:56:09][I][vista_alarm:968]: Partition: 01
[08:56:09][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 03 MOTION '
[08:56:09][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'DETECTOR        '
[08:56:09][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[08:56:09][I][vista_alarm:992]: Prompt: FAULT 03 MOTION 
[08:56:09][I][vista_alarm:993]: Prompt: DETECTOR        
[08:56:09][I][vista_alarm:994]: Beeps: 0
[08:56:09][D][binary_sensor:036]: 'Living Room Motion Detector': Sending state ON
[08:56:09][D][text_sensor:064]: 'Zone Status (zs)': Sending state 'OP:3,OP:6'
[08:56:09][W][component:237]: Component vista_alarm_panel took a long time for an operation (64 ms).
[08:56:09][W][component:238]: Components should block for at most 30 ms.
[08:56:09][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:09][I][EXT:590]: 2024-05-21 08:56  F6 01 61 0C 00 69 62 02 45 43 F5 31 FB 43 6C 6E 
[08:56:09][I][CHK:590]: 2024-05-21 08:56  61 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:09][I][CMD:590]: 2024-05-21 08:56  F2 0E 02 00 00 00 00 51 FE FE EC 03 02 01 06 B9 
[08:56:11][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:11][I][EXT:590]: 2024-05-21 08:56  F6 01 A1 0C 00 6A 62 02 45 43 F5 31 FB 43 6C 2D 
[08:56:11][I][CHK:590]: 2024-05-21 08:56  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:11][I][CMD:590]: 2024-05-21 08:56  F2 0E 02 00 00 00 00 52 FE FE EC 03 02 01 06 B8 
[08:56:11][I][CMD:590]: 2024-05-21 08:56  F2 12 66 00 00 00 00 61 6C 02 45 6C F5 EC 01 01 01 00 00 32 
[08:56:12][I][CMD:590]: 2024-05-21 08:56  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[08:56:12][I][vista_alarm:968]: Partition: 01
[08:56:12][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[08:56:12][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[08:56:12][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[08:56:12][I][vista_alarm:992]: Prompt: FAULT 06 SOUTH  
[08:56:12][I][vista_alarm:993]: Prompt: GUEST ROOM      
[08:56:12][I][vista_alarm:994]: Beeps: 0
[08:56:12][I][CMD:590]: 2024-05-21 08:56  F2 16 66 00 00 00 00 62 63 02 45 43 F5 31 FB 45 6C F5 EC 03 02 01 06 84 
[08:56:12][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:12][I][EXT:590]: 2024-05-21 08:56  F6 01 E1 0C 00 6B 62 02 45 43 F5 31 FB 43 6C EC 
[08:56:12][I][CHK:590]: 2024-05-21 08:56  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:12][I][CMD:590]: 2024-05-21 08:56  F2 0E 02 00 00 00 00 53 FE FE EC 03 02 01 06 B7 
[08:56:12][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:12][I][EXT:590]: 2024-05-21 08:56  F6 01 21 0C 00 6C 62 02 45 43 F5 31 FB 43 6C AB 
[08:56:12][I][CHK:590]: 2024-05-21 08:56  21 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:12][I][CMD:590]: 2024-05-21 08:56  F2 0E 02 00 00 00 00 54 FE FE EC 03 02 01 06 B6 
[08:56:13][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[08:56:13][I][EXT:590]: 2024-05-21 08:56  F6 01 61 16 00 6D 62 31 45 49 F5 31 FB 45 4A F5 32 FB 45 43 F5 31 FB 43 6C D1 
[08:56:13][I][CHK:590]: 2024-05-21 08:56  61 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:13][I][CMD:590]: 2024-05-21 08:56  F2 0D 02 00 00 00 00 55 FE FE FE FE EC 36 90 
[... 314 lines cut ...]
[08:56:54][I][CMD:590]: 2024-05-21 08:56  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
[08:56:57][I][CHK:590]: 2024-05-21 08:56  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[08:56:57][I][CMD:590]: 2024-05-21 08:56  F2 0B 02 00 00 00 00 52 FE FE FE FE B7 
[08:56:58][I][CMD:590]: 2024-05-21 08:56  F7 02 00 1D 10 06 00 00 28 02 00 00 46 
[08:56:58][I][vista_alarm:968]: Partition: 01
[08:56:58][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'FAULT 06 SOUTH  '
[08:56:58][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'GUEST ROOM      '
[08:56:58][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'

Dilbert66, 2024-05-21:
Awesome! Yes, that does the trick! I can use that cmd on my own system to retrieve open zones.

Dilbert66, 2024-05-23:
Ok, i've added initial code to support aui cmd/response capability.  Now on any zone status change that sends an F2 broadcast, the firmware will send a status update cmd to get any open/bypassed zones and immediately update the status.  You should see immediate responses to your zones.   You need to activate another AUI device in program field *189. Ensure it's not used else where.  For vista20's etc, there are only two addresses supported: 1 and 2, for newer/larger systems, you also have addresses 5 and 6 available.   Steps:
1. Enable a non used AUI address in program *189.  You can assign to any partition but I would choose partition 1.
2. Add this line to your yaml:
```
 auiaddr: "<AUI address from above>" #AUI address from program field *189 to use for zone status queries. Ensure it is not assigned to a real keypad.
```
Recompile and you should be good to do then.

rwoldberg, 2024-05-25:
I am getting compile errors: 
src/esphome/components/vista_alarm_panel/vistaalarm.cpp:534:10: error: no declaration matches 'void esphome::alarm_panel::vistaECPHome::send_cmd_bytes(int, std::__cxx11::string)'
     void vistaECPHome::send_cmd_bytes(int addr,std::string hexbytes) {
          ^~~~~~~~~~~~
In file included from src/esphome/components/vista_alarm_panel/vistaalarm.cpp:1:
src/esphome/components/vista_alarm_panel/vistaalarm.h:378:10: note: candidate is: 'void esphome::alarm_panel::vistaECPHome::send_cmd_bytes(uint8_t, std::__cxx11::string)'
     void send_cmd_bytes(uint8_t addr,std::string hexbytes);
          ^~~~~~~~~~~~~~
In file included from src/esphome/components/vista_alarm_panel/vistaalarm.cpp:1:
src/esphome/components/vista_alarm_panel/vistaalarm.h:97:8: note: 'class esphome::alarm_panel::vistaECPHome' defined here
  class vistaECPHome: public api::CustomAPIDevice, public time::RealTimeClock {
        ^~~~~~~~~~~~
Compiling .pioenvs/vista20p-bridge/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/vista20p-bridge/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/vista20p-bridge/src/esphome/components/wifi/wifi_component_libretiny.cpp.o
*** [.pioenvs/vista20p-bridge/src/esphome/components/vista_alarm_panel/vistaalarm.cpp.o] Error 1

Dilbert66, 2024-05-25:
I posted a corrected version

rwoldberg, 2024-05-25:
I am still getting an error:
src/esphome/components/vista_alarm_panel/vistaalarm.cpp:534:10: error: no declaration matches 'void esphome::alarm_panel::vistaECPHome::send_cmd_bytes(int, std::__cxx11::string)'
     void vistaECPHome::send_cmd_bytes(int addr,std::string hexbytes) {
          ^~~~~~~~~~~~
In file included from src/esphome/components/vista_alarm_panel/vistaalarm.cpp:1:
src/esphome/components/vista_alarm_panel/vistaalarm.h:378:10: note: candidate is: 'void esphome::alarm_panel::vistaECPHome::send_cmd_bytes(uint8_t, std::__cxx11::string)'
     void send_cmd_bytes(uint8_t addr,std::string hexbytes);
          ^~~~~~~~~~~~~~
In file included from src/esphome/components/vista_alarm_panel/vistaalarm.cpp:1:
src/esphome/components/vista_alarm_panel/vistaalarm.h:97:8: note: 'class esphome::alarm_panel::vistaECPHome' defined here
  class vistaECPHome: public api::CustomAPIDevice, public time::RealTimeClock {
        ^~~~~~~~~~~~

Dilbert66, 2024-05-25:
Doesnt look like it's picking up the new updates. What is your refresh: setting under the external_components: section?  You should be able to retry today and have it work now.

rwoldberg, 2024-05-25:
I did not have a refresh setting. I added one and it is compiling now.  Thanks.

rwoldberg, 2024-05-25:
This seems to be working very well.

Dilbert66, 2024-05-28:
Great.  Is there an option to show all enabled zones on the panel? Such as the menu "More choices" and "Show zones"? Is that the same option you used? I'm looking for the option that retrieves all enabled zones from the panel. Not just open or bypassed.  If there is, could you capture that log? It could be handing to have the ESP retrieve all that data on startup and use the data on startup.

rwoldberg, 2024-05-28:
Indeed there is, I will try and capture that info today.  Would you like a separate issue for that so this one could be closed?

rwoldberg, 2024-05-29:
Here is the output after a restart, does this have the data you need?
14:58:56][I][CMD:572]: 2024-05-29 14:58  FA 01 02 20 F7 EC 00 00 00 00 00 00 00 
[14:58:56][I][EXT:572]: 2024-05-29 14:58  FA 07 F7 00 00 00 00 00 00 00 00 00 00 
[14:59:01][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:01][I][EXT:572]: 2024-05-29 14:59  F6 01 21 07 00 68 6B 02 43 43 7D 
[14:59:01][I][CHK:572]: 2024-05-29 14:59  21 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:02][I][CMD:572]: 2024-05-29 14:59  F2 0E 02 00 00 00 00 50 FE EC 01 01 01 00 00 C1 
[14:59:02][D][vista_alarm:1006]: step = 0
[14:59:02][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:02][I][CHK:572]: 2024-05-29 14:59  61 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:02][I][EXT:572]: 2024-05-29 14:59  F6 01 61 08 00 69 6B 0B 45 6C 32 D5 
[14:59:02][I][CMD:572]: 2024-05-29 14:59  F2 09 02 00 00 00 00 51 FD 32 83 
[14:59:02][D][vista_alarm:1006]: step = 0
[14:59:02][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:02][I][EXT:572]: 2024-05-29 14:59  F6 01 A1 0C 00 6A 6B 0B 45 43 F5 32 FB 43 6C 1A 
[14:59:02][I][CHK:572]: 2024-05-29 14:59  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:02][I][CMD:572]: 2024-05-29 14:59  F2 15 02 00 00 00 00 52 FE FE EC 30 34 2E 30 32 20 30 34 2E 30 31 B6 
[14:59:02][D][vista_alarm:1006]: step = 0
[14:59:02][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:03][I][EXT:572]: 2024-05-29 14:59  F6 01 E1 0C 00 6B 6B 0B 45 43 F5 32 FB 43 6C D9 
[14:59:03][I][CHK:572]: 2024-05-29 14:59  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:03][I][CMD:572]: 2024-05-29 14:59  F2 15 02 00 00 00 00 53 FE FE EC 30 34 2E 30 32 20 30 34 2E 30 31 B5 
[14:59:03][D][vista_alarm:1006]: step = 0
[14:59:03][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:03][I][EXT:572]: 2024-05-29 14:59  F6 01 21 0C 00 6C 6B 0B 45 43 F5 33 FB 43 6C 97 
[14:59:03][I][CHK:572]: 2024-05-29 14:59  21 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:03][I][CMD:572]: 2024-05-29 14:59  F2 0A 02 00 00 00 00 54 FE FE 37 7B 
[14:59:03][D][vista_alarm:1006]: step = 0
[14:59:04][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:04][I][EXT:572]: 2024-05-29 14:59  F6 01 61 07 00 6D 6B 0D 43 43 2D 
[14:59:04][I][CHK:572]: 2024-05-29 14:59  61 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:04][I][CMD:572]: 2024-05-29 14:59  F2 09 02 00 00 00 00 55 FE 31 7F 
[14:59:04][D][vista_alarm:1006]: step = 0
[14:59:04][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:04][I][CHK:572]: 2024-05-29 14:59  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:04][I][EXT:572]: 2024-05-29 14:59  F6 01 A1 09 00 6E 6B 0E 45 43 F5 31 C1 
[14:59:04][I][CMD:572]: 2024-05-29 14:59  F2 08 02 00 00 00 00 56 FE B0 
[14:59:04][D][vista_alarm:1006]: step = 0
[14:59:04][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:04][I][CHK:572]: 2024-05-29 14:59  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:05][I][EXT:572]: 2024-05-29 14:59  F6 01 E1 0C 00 6F 6B 0B 45 43 F5 31 FB 43 6C D6 
[14:59:05][W][component:237]: Component vista_alarm_panel took a long time for an operation (74 ms).
[14:59:05][W][component:238]: Components should block for at most 30 ms.
[14:59:05][I][CMD:572]: 2024-05-29 14:59  F2 1D 02 00 00 00 00 57 FE FE EC 53 61 66 65 77 61 74 63 68 20 50 72 6F 20 33 30 30 30 69 7D 
[14:59:05][D][vista_alarm:1006]: step = 0
[14:59:05][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:05][I][EXT:572]: 2024-05-29 14:59  F6 01 21 07 00 68 6B 0C 43 61 55 
[14:59:05][I][CHK:572]: 2024-05-29 14:59  21 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:05][I][CMD:572]: 2024-05-29 14:59  F2 09 02 00 00 00 00 50 FE 33 82 
[14:59:05][D][vista_alarm:1006]: step = 0
[14:59:05][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:05][I][EXT:572]: 2024-05-29 14:59  F6 01 61 0C 00 69 6B 0C 45 43 F5 31 FB 43 6C 5B 
[14:59:05][I][CHK:572]: 2024-05-29 14:59  61 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:05][I][CMD:572]: 2024-05-29 14:59  F2 11 02 00 00 00 00 51 FE FE EC 31 00 48 4F 55 53 45 0D 
[14:59:05][D][vista_alarm:1006]: step = 0
[14:59:05][I][CMD:572]: 2024-05-29 14:59  F7 02 00 3D 10 08 00 1C 28 02 00 00 48 
[14:59:05][I][vista_alarm:1032]: Partition: 01
[14:59:05][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state 'HOUSE           '
[14:59:05][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state 'CHIME,Rdy to Arm'
[14:59:05][I][vista_alarm:1056]: Prompt: HOUSE           
[14:59:05][I][vista_alarm:1057]: Prompt: CHIME,Rdy to Arm
[14:59:05][I][vista_alarm:1058]: Beeps: 0
[14:59:06][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:06][I][CHK:572]: 2024-05-29 14:59  A1 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:06][I][EXT:572]: 2024-05-29 14:59  F6 01 A1 0C 00 6A 6B 0C 45 43 F5 32 FB 43 6C 19 
[14:59:06][I][CMD:572]: 2024-05-29 14:59  F2 0B 02 00 00 00 00 52 FE FE EC 32 95 
[14:59:06][D][vista_alarm:1006]: step = 0
[14:59:06][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:06][I][EXT:572]: 2024-05-29 14:59  F6 01 E1 0C 00 6B 6B 0C 45 43 F5 33 FB 43 6C D7 
[14:59:06][I][CHK:572]: 2024-05-29 14:59  E1 00 00 00 00 00 00 00 00 00 00 00 90 
[... 334 lines cut ...]
[14:59:35][I][vista_alarm:1056]: Prompt: HOUSE           
[14:59:35][I][vista_alarm:1057]: Prompt: CHIME,Rdy to Arm
[... cut ...]
[14:59:45][I][vista_alarm:1056]: Prompt: HOUSE           
[14:59:45][I][vista_alarm:1057]: Prompt: CHIME,Rdy to Arm
[14:59:45][I][vista_alarm:1058]: Beeps: 0
[14:59:46][I][CMD:572]: 2024-05-29 14:59  F6 01 00 00 00 00 00 00 00 00 00 00 00 
[14:59:46][I][EXT:572]: 2024-05-29 14:59  F6 01 61 0C 00 69 62 02 45 43 F5 31 FB 43 6C 6E 
[14:59:46][I][CHK:572]: 2024-05-29 14:59  61 00 00 00 00 00 00 00 00 00 00 00 90 
[14:59:46][I][CMD:572]: 2024-05-29 14:59  F2 0E 02 00 00 00 00 51 FE FE EC 03 01 01 01 BF 
[14:59:46][D][vista_alarm:1006]: step = 0

Dilbert66, 2024-05-30:
Excellent! Exactly what I needed! Thank you! I just need to analyze the cmds a bit and see what I can use.  So far I can retrieve the zone count per partition and enumerate each zone and identify the type and assigned name if it's there.

Dilbert66, 2024-11-15:
I have another request if you don't mind. Can you capture the logs of going through a time/date set sequence and have it send to the panel ? I want to see what cmds that this display uses to talk to the panel for setting date and time.

XD36JD2015, 2024-11-15:
Sure thing, I'll capture that this weekend.

Dilbert66, 2024-11-15:
That's for a 6170 or tuxedo keypad right? I need the AUI traffic.

Dilbert66, 2024-11-15:
Awesome! Tks for the logs again. FYI, I do have a set_panel_time() service existing on the esp code which just emulates the keypad cmds to actually set the date/time on the panel.  Try it out from the dev console on HA.  I want to also implement it using the AUI api if possible as it should be more consistent.

Dilbert66, 2024-11-16:
Great, I found the command in your log and will have a working service on the next update in a day or so.
```
11:41:59][I][EXT:693]: 2024-11-15 11:41 F6 01 61 16 00 69 05 02 45 43 F5 EC 32 34 31 31 31 35 31 31 34 31 30 35 35 21
```

Dilbert66, 2024-11-17:
I've update the "dev" branch.  The set_panel_time() service will use the AUI cmd for updating the time on the panel if you have an active aui address setup, otherwise, it will default to using the a key sequence to update the time as if it was entered from the panel.  You can use the service in the yaml by calling it with "$panelId->set_panel_time()" from an interval entry like this: 
```
interval:
  - interval: 300s
    then:
      - lambda: |-
          $panelId->set_panel_time();
```

rwoldberg, 2024-11-17:
Very cool.  I will give it a try.  Thanks.

rwoldberg, 2024-11-19:
The time set seems to be working.  I am assuming the seconds are not set by this API as the clock is always off by some seconds.  It seems to do the same thing even when setting from the actual panel.  I would guess the vista firmware just does not reset the seconds when setting the time.  No big deal.

Dilbert66, 2024-11-19:
My system was showing odd behaviour with certain settings of the seconds. It would always fail when seconds sent were for example, 48, 47 57, 39, etc.  The fields are there from what I see.  Not sure what the issue is there. It would always work fine when the seconds were sent as zero.   Your example looks like it is sending seconds but I need to see more examples.  Can you provide more logs of setting the time at various periods? 5 or 6 times should be fine.  I'd like to see if there is something I'm missing.

Dilbert66, 2024-11-19:
Ok, pushed a fix to "dev".  Now sets the seconds and day of week correctly and synced to the esp time.

rwoldberg, 2024-11-19:
I will give it a try.

rwoldberg, 2024-11-19:
I am getting an error now:

In file included from src/esphome/components/vista_alarm_panel/vistaalarm.cpp:2:
src/esphome/components/vista_alarm_panel/vistaalarm.h:9:10: fatal error: esphome/components/template_alarm/binary_sensor/template_binary_sensor.h: No such file or directory
 #include "esphome/components/template_alarm/binary_sensor/template_binary_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/vista20p-bridge/src/esphome/components/vista_alarm_panel/vistaalarm.cpp.o] Error 1

Dilbert66, 2024-11-19:
ahh.. ok, hold on. You are not using the template_alarm for your sensors and using the binary_sensors.  I'll push a quick change.

Dilbert66, 2024-11-19:
Ok, update pushed.

Dilbert66, 2024-11-19:
Ok, now you should be good.

rwoldberg, 2024-11-20:
Seems to be working.  My panel is still off by about 30 seconds, not a big deal, but kind of strange. The panel must not be resetting the seconds, but its close enough.

[2 more replies]

closed 2026-10-17 by Dilbert66