Dilbert66/esphome-vistaECP #106: Low battery on RF sensor causes unavailable state when partition is armed

philh30, 2023-07-09:
When the partition is armed and an RF zone is reporting low battery, the LOBAT reports are causing the partition status to flap between armed and unavailable:

```
[07:37:29][I][CMD:620]: 2023-07-02 07:37  F7 00 00 51 10 08 00 8C 08 02 00 00 41 
[07:37:29][I][INFO:912]: Display to partition: 01
[07:37:29][D][text_sensor:064]: 'Line1': Sending state 'ARMED ***STAY***'
[07:37:29][D][text_sensor:064]: 'Line2': Sending state 'You may exit now'
[07:37:29][I][INFO:939]: Prompt: ARMED ***STAY*** 
[07:37:29][I][INFO:940]: Prompt: You may exit now
[07:37:29][I][INFO:941]: Beeps: 0

[07:37:29][D][text_sensor:064]: 'System Status 1': Sending state 'armed_home'
[07:37:33][I][CMD:620]: 2023-07-02 07:37  F7 00 00 51 10 21 00 C0 08 02 00 00 4C 
[07:37:33][I][INFO:912]: Display to partition: 01
[07:37:33][D][text_sensor:064]: 'Line1': Sending state 'LOBAT 21 FRONT  '
[07:37:33][D][text_sensor:064]: 'Line2': Sending state 'DOOR            '
[07:37:33][I][INFO:939]: Prompt: LOBAT 21 FRONT   
[07:37:33][I][INFO:940]: Prompt: DOOR            
[07:37:33][I][INFO:941]: Beeps: 0

[07:37:33][D][text_sensor:064]: 'System Status 1': Sending state 'unavailable'
[07:37:33][D][text_sensor:064]: 'RF Msg': Sending state ''
[07:37:37][I][CMD:620]: 2023-07-02 07:37  F7 00 00 51 10 08 00 8C 08 02 00 00 41 
[07:37:37][I][INFO:912]: Display to partition: 01
[07:37:37][D][text_sensor:064]: 'Line1': Sending state 'ARMED ***STAY***'
[07:37:37][D][text_sensor:064]: 'Line2': Sending state 'You may exit now'
[07:37:37][I][INFO:939]: Prompt: ARMED ***STAY*** 
[07:37:37][I][INFO:940]: Prompt: You may exit now
[07:37:37][I][INFO:941]: Beeps: 0

[07:37:37][D][text_sensor:064]: 'System Status 1': Sending state 'armed_home'
```

When disarmed the partition is consistently reported as such:

```
[07:37:51][I][CMD:620]: 2023-07-02 07:37  F7 00 00 51 10 08 00 1C 08 02 00 00 2A 
[07:37:51][I][INFO:912]: Display to partition: 01
[07:37:51][D][text_sensor:064]: 'Line1': Sending state '****DISARMED****'
[07:37:51][D][text_sensor:064]: 'Line2': Sending state '  Ready to Arm  '
[07:37:51][I][INFO:939]: Prompt: ****DISARMED**** 
[07:37:51][I][INFO:940]: Prompt:   Ready to Arm  
[07:37:51][I][INFO:941]: Beeps: 0

[07:37:52][I][CMD:620]: 2023-07-02 07:37  FB 02 25 81 5D 00 00 00 00 00 00 00 00 
[07:37:53][I][EXT:620]: 2023-07-02 07:37  00 21 00 DF 00 00 00 00 00 00 00 00 74 
[07:37:53][I][CMD:620]: 2023-07-02 07:37  FB 02 20 82 61 00 00 00 00 00 00 00 00 
[07:37:54][I][EXT:620]: 2023-07-02 07:37  00 24 03 D9 00 00 00 00 00 00 00 00 74 
[07:37:55][I][CMD:620]: 2023-07-02 07:37  F7 00 00 51 10 21 00 50 08 02 00 00 4C 
[07:37:55][I][INFO:912]: Display to partition: 01
[07:37:55][D][text_sensor:064]: 'Line1': Sending state 'LOBAT 21 FRONT  '
[07:37:55][D][text_sensor:064]: 'Line2': Sending state 'DOOR            '
[07:37:55][D][text_sensor:064]: 'Beeps': Sending state '0'
[07:37:55][I][INFO:939]: Prompt: LOBAT 21 FRONT   
[07:37:55][I][INFO:940]: Prompt: DOOR            
[07:37:55][I][INFO:941]: Beeps: 0

[07:37:55][D][text_sensor:064]: 'System Status 1': Sending state 'disarmed'
[07:37:55][D][binary_sensor:036]: 'Stay': Sending state OFF
[07:37:55][D][binary_sensor:036]: 'Armed': Sending state OFF
[07:37:55][D][text_sensor:064]: 'zoneExtended': Sending state ''
[07:37:56][I][CMD:620]: 2023-07-02 07:37  F9 43 02 53 6F 00 00 00 00 00 00 00 00 
[07:37:56][I][CMD:620]: 2023-07-02 07:37  77 83 00 00 00 00 00 00 00 00 00 00 00 
[07:37:57][I][EXT:620]: 2023-07-02 07:37  F9 00 83 04 00 00 00 79 00 00 00 00 00 
[07:37:59][I][CMD:620]: 2023-07-02 07:37  F7 00 00 51 10 08 00 1C 08 02 00 00 2A 
[07:37:59][I][INFO:912]: Display to partition: 01
[07:37:59][D][text_sensor:064]: 'Line1': Sending state '****DISARMED****'
[07:37:59][D][text_sensor:064]: 'Line2': Sending state '  Ready to Arm  '
[07:37:59][I][INFO:939]: Prompt: ****DISARMED**** 
[07:37:59][I][INFO:940]: Prompt:   Ready to Arm  
[07:37:59][I][INFO:941]: Beeps: 0

[07:37:59][I][CMD:620]: 2023-07-02 07:37  F7 00 00 A0 00 08 00 1C 08 02 00 00 2A 
```

Dilbert66, 2023-07-19:
Ok, i see the issue. Will post a fix shortly.

Dilbert66, 2023-07-19:
I've pushed an update that should address this issue. Only vistaalarm.h needs to be updated.

closed 2023-11-02 by Dilbert66