Dilbert66/esphome-vistaECP #113: Custom Alarm Panel Card does not display properly in Google Chrome

crumbliness, 2023-11-02:
Custom Alarm Panel Card from this project does not display properly in Google Chrome Version 119.0.6045.106 (Official Build) (64-bit), Windows 11. Message: "Custom element doesn't exist: alarm-keypad-card." Sorry but I don't know when it stopped working.

[image]

Working fine in
- Brave [Version 1.59.124 Chromium: 118.0.5993.117 (Official Build) (64-bit)](https://brave.com/latest/)
- Microsoft Edge Version 118.0.2088.76 (Official build) (64-bit)
- Android 14 Chrome
- possibly others

Dilbert66, 2023-11-02:
What branch? If you are using master, I've pushed an updated version.

5hortCircuit, 2023-11-05:
I am having the same issue.  Appears to show fine with Edge, but chrome on my desktop/laptop and cellphone doesn't render the card.  Tried the new card and it didn't seem to resolve it for me.

Dilbert66, 2023-11-05:
Like I noted above you need to use the new card from dev or the new version on master.  You are still using the old polymer version.

crumbliness, 2023-11-06:
Tried the new card from mater and had the same display issues in Google Chrome Version 119.0.6045.106. Also tried to to install master and dev on a new ESP8266 NodeMCU device and could not compile/install either. Still trying to digest the errors reported.
ESPHome 2023.10.6

crumbliness, 2023-11-06:
Tried again using master and was able to compile/install with a handful of warnings ... no explanation for the change.

crumbliness, 2023-11-06:
The alarm-panel-card.js file in my /config/www directory of  homeassistant installation is using Lit ...
[image]

Dilbert66, 2023-11-06:
If you look at the dev readme, it explains what you need to do to refresh the javascript load. Due to browser caching, you need to make it look like a new file. The easiest is to add ?id=1 to the file name  where 1 where you will increase the =1 everytime you load to make it new again. 

eg:  /local/alarm-keypad-card.js?id=xx

crumbliness, 2023-11-07:
Thanks! It's working and I learned something.

5hortCircuit, 2023-11-09:
Just to clarify the new card requires the updated ESPHome installation?  Or can the new card be used with the old ESP software?

Dilbert66, 2023-11-09:
The card works with both new and old versions

5hortCircuit, 2023-11-09:
@Dilbert66 I think you have git merge leftovers in the dev branch

[image]

Dilbert66, 2023-11-09:
argh... Tks! Corrected and pushed

closed 2024-11-01 by Dilbert66