Dilbert66/esphome-vistaECP #115: Zones detected or Open in HomeAssistant after arming the system despite no changes

AlexandreUSA, 2023-11-13:
Hi, using the latest Master code as of 11/10, I am having all my zone changing status when I armed my Alarm. This is recurring. 
When I setup "Arm stay" for the night using either the HA app or the Vista 21ip keypad, my system arms but I do get most of my zones getting to another state (Open, Detected). For example, all my heat, fire, or smoke detectors show in HA as "Detected" for a few seconds and my doors and Windows as closed. Not all at the same time but in the seconds after the alarm is armed. They all go back to theu normal state. Nothing happens on the Vista 21ip side so no false alarms there and not fire trucks dispatched. I will need to find some time getting logs but I wanted this out as you may know what is going on.

Dilbert66, 2023-11-13:
Need to see logs otherwise I can't help.

AlexandreUSA, 2023-11-13:
I assume that you do not see that on your side so I will pull the logs. On my side:  "code + 3", System counts down for Arm Stay, once armed, the zones are showing in another states in HA then come back to normal.

Dilbert66, 2023-11-13:
I need to see logs so I can see what the panel  sends. No one else has reported the issue you have so it's impossible to know otherwise

Dilbert66, 2023-11-14:
What is the TTL set to in the yaml?

AlexandreUSA, 2023-11-14:
TTL is set to 30000 (default). I have also attached to this reply some portions of the log file in a txt format. I need to find a better way to copy paste the log as Ardunio IDE does not let me do it directly. I will try Putty next time. 

[Log.txt](https://github.com/Dilbert66/esphome-vistaECP/files/13343591/Log.txt)

It seems that during the Arm Stay count down, the fault corresponding to the "second left" is set at fault. Starting at zone 60 (but this one does not exist), then it hits my first real zone 23 where it shows as in fault, all the way to zone 1. 

> 20:06:08.836 -> [1;31m[E][INFO:1064]: Display to partition: 01[0m
> 20:06:08.836 -> [0;36m[D][text_sensor:064]: 'Line2': Sending state 'May Exit Now  24'[0m
> 20:06:08.836 -> [1;31m[E][INFO:1091]: Prompt: ARMED ***STAY*** [0m
> 20:06:08.836 -> [1;31m[E][INFO:1092]: Prompt: May Exit Now  24[0m
> 20:06:08.836 -> [1;31m[E][INFO:1093]: Beeps: 0
> 20:06:08.836 -> [0m
> 20:06:08.836 -> [0;36m[D][binary_sensor:036]: 'Heat Garage': Sending state ON[0m
> 20:06:08.868 -> [0;36m[D][test:1222]: fault found for zone 24,status=1[0m
> 20:06:08.868 -> [0;36m[D][text_sensor:064]: 'Zone Status': Sending state 'OP:15,OP:16,OP:24,OP:25,OP:26,OP:27,OP:28,OP:29,OP:30,OP:31,OP:32,OP:33,OP:34,OP:35,OP:36,OP:37,OP:38,OP:39,OP:40,OP:41,OP:42,OP:43,OP:44,OP:45,OP:46,OP:47,OP:48,OP:49,OP:50,OP:51,OP:52,OP:53,OP:54'[0m
> 20:06:08.904 -> [0;33m[W][component:214]: Component vista_alarm_panel took a long time for an operation (0.17 s).[0m
> 20:06:08.904 -> [0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
> 20:06:09.554 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  F9 83 02 53 2F 00 00 00 00 00 00 00 00 [0m
> 20:06:09.586 -> [0;36m[D][test:1222]: fault found for zone 24,status=1[0m
> 20:06:09.586 -> [0;36m[D][text_sensor:064]: 'Zone Status': Sending state 'OP:15,OP:16,OP:24,OP:25,OP:26,OP:27,OP:28,OP:29,OP:30,OP:31,OP:32,OP:33,OP:34,OP:35,OP:36,OP:37,OP:38,OP:39,OP:40,OP:41,OP:42,OP:43,OP:44,OP:45,OP:46,OP:47,OP:48,OP:49,OP:50,OP:51,OP:52,OP:53'[0m
> 20:06:09.624 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  78 C3 00 00 00 00 00 00 00 00 00 00 00 [0m
> 20:06:09.874 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  F7 00 00 0F 10 23 00 80 08 02 00 00 41 [0m
> 20:06:09.874 -> [1;31m[E][INFO:1064]: Display to partition: 01[0m
> 20:06:09.874 -> [0;36m[D][text_sensor:064]: 'Line2': Sending state 'May Exit Now  23'[0m
> 20:06:09.874 -> [1;31m[E][INFO:1091]: Prompt: ARMED ***STAY*** [0m
> 20:06:09.874 -> [1;31m[E][INFO:1092]: Prompt: May Exit Now  23[0m
> 20:06:09.906 -> [1;31m[E][INFO:1093]: Beeps: 0

...

> 20:06:31.645 -> [0;36m[D][binary_sensor:036]: 'Front Door': Sending state ON[0m
> 20:06:31.645 -> [0;36m[D][test:1222]: fault found for zone 2,status=1[0m
> 20:06:31.645 -> [0;36m[D][text_sensor:064]: 'Zone Status': Sending state 'OP:2,OP:3,OP:4,OP:5,OP:6,OP:7,OP:8,OP:15,OP:16,OP:17,OP:18,OP:19,OP:20,OP:21,OP:22,OP:23,OP:24,OP:25,OP:26,OP:27,OP:28,OP:29,OP:30,OP:31'[0m
> 20:06:31.681 -> [0;33m[W][component:214]: Component vista_alarm_panel took a long time for an operation (0.17 s).[0m
> 20:06:31.681 -> [0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
> 20:06:32.554 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  F7 00 00 0F 10 01 00 80 08 02 00 00 41 [0m
> 20:06:32.586 -> [1;31m[E][INFO:1064]: Display to partition: 01[0m
> 20:06:32.586 -> [0;36m[D][text_sensor:064]: 'Line1': Sending state 'ARMED ***STAY***'[0m
> 20:06:32.586 -> [0;36m[D][text_sensor:064]: 'Line2': Sending state 'May Exit Now  01'[0m
> 20:06:32.586 -> [0;36m[D][text_sensor:064]: 'Beeps': Sending state '0'[0m
> 20:06:32.586 -> [1;31m[E][INFO:1091]: Prompt: ARMED ***STAY*** [0m
> 20:06:32.637 -> [1;31m[E][INFO:1092]: Prompt: May Exit Now  01[0m
> 20:06:32.637 -> [1;31m[E][INFO:1093]: Beeps: 0
> 20:06:32.637 -> [0m
> 20:06:32.637 -> [0;36m[D][binary_sensor:036]: 'Smoke Basement': Sending state ON[0m
> 20:06:32.637 -> [0;36m[D][test:1222]: fault found for zone 1,status=1[0m
> 20:06:32.671 -> [0;36m[D][text_sensor:064]: 'Zone Status': Sending state 'OP:1,OP:2,OP:3,OP:4,OP:5,OP:6,OP:7,OP:8,OP:15,OP:16,OP:17,OP:18,OP:19,OP:20,OP:21,OP:22,OP:23,OP:24,OP:25,OP:26,OP:27,OP:28,OP:29,OP:30'[0m
> 20:06:32.671 -> [0;33m[W][component:214]: Component vista_alarm_panel took a long time for an operation (0.22 s).[0m
> 20:06:32.708 -> [0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
> 20:06:32.708 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  F2 12 06 00 00 00 00 67 6C 02 45 6C F5 [0m
> 20:06:32.708 -> [0;36m[D][test:1222]: fault found for zone 1,status=1[0m
> 20:06:32.863 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  F2 16 06 00 00 00 00 60 63 02 45 43 F5 [0m
> 20:06:32.863 -> [0;36m[D][test:1222]: fault found for zone 1,status=1[0m
> 20:06:32.915 -> [0;33m[W][component:214]: Component vista_alarm_panel took a long time for an operation (0.07 s).[0m
> 20:06:32.915 -> [0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
> 20:06:32.988 -> [1;31m[E][CMD:743]: 2023-11-13 20:06  F2 16 06 00 00 00 00 61 63 02 45 43 F5 [0m
> 20:06:32.988 -> [0;36m[D][test:1222]: fault found for zone 1,status=1[0m
> 20:06:32.988 -> [0;36m[D][text_sensor:064]: 'Zone Status': Sending state 'OP:1,OP:2,OP:3,OP:4,OP:5,OP:6,OP:7,OP:8,OP:15,OP:16,OP:17,OP:18,OP:19,OP:20,OP:21,OP:22,OP:23,OP:24,OP:25,OP:26,OP:27,OP:28,OP:29'[0m
[... cut ...]
> 20:07:12.472 -> [0;33m[W][component:214]: Component vista_alarm_panel took a long time for an operation (0.14 s).[0m
> 20:07:12.472 -> [0;33m[W][component:215]: Components should block for at most 20-30ms.[0m
> 20:07:22.285 -> [1;31m[E][CMD:743]: 2023-11-13 20:07  F7 00 00 0F 10 08 00 8C 08 02 00 00 41 [0m
> 20:07:22.285 -> [1;31m[E][INFO:1064]: Display to partition: 01[0m
> 20:07:22.285 -> [1;31m[E][INFO:1091]: Prompt: ARMED ***STAY*** [0m
> 20:07:22.285 -> [1;31m[E][INFO:1092]: Prompt:                 [0m
> 20:07:22.285 -> [1;31m[E][INFO:1093]: Beeps: 0
> 20:07:22.285 -> [0m

Dilbert66, 2023-11-14:
Excellent! I see the issue.  The system is seeing the exit timer as zones. I will fix that. My panel does not show that message.

Dilbert66, 2023-11-14:
I've pushed  a potential fix to main.  Your panel does not send a needed flag during the exit delay so my system was seeing it as a zone message.  I've added a check to look at the alarm flags.   Just recompile your yaml. It should pick up the new fix.

AlexandreUSA, 2023-11-15:
Hi, I recompiled the yaml file but no changes. Still the same issue. Was it pushed ? I do not see the change in the history. How can I check that the change that you pushed was included in my project? I did a clean file install and recompiled the project. I also see the github://Dilbert66/esphome-components@main in the installation log. Thanks again for your support.

Dilbert66, 2023-11-15:
The changes are now on the esphome-components repository. That's where the main library code resides. You should have seen it recompiling the main.cpp. The system pulls down any new version.

Dilbert66, 2023-11-15:
Ok, i've pushed another update to main. Just recompile.  If you have issues still, please post logs. Just stating it doesnt work doesnt help as I need to see what is going on.

Dilbert66, 2023-11-15:
Ok, i've pushed a potential fix. Recompile and try again.

AlexandreUSA, 2023-11-16:
Hi, thanks. The original issue is gone but a new one appeared. The "Stay" and "Armed" flags are getting triggered from on to off to on to off, etc while the system is in fact in Armed stay:

> [18:56:18][D][test:1273]: armed status/system,stay,away flag is: 1 , 0, 1 , 0
> [18:56:18][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.17 s).
> [18:56:18][W][component:215]: Components should block for at most 20-30ms.
> [18:56:19][E][CMD:802]: 2023-11-15 18:56  F7 00 00 0F 10 01 00 80 08 02 00 00 41 
> [18:56:19][E][INFO:1123]: Display to partition: 01
> [18:56:19][D][text_sensor:064]: 'Line1': Sending state 'ARMED ***STAY***'
> [18:56:19][D][text_sensor:064]: 'Line2': Sending state 'May Exit Now  01'
> [18:56:19][D][text_sensor:064]: 'Beeps': Sending state '0'
> [18:56:19][E][INFO:1150]: Prompt: ARMED ***STAY*** 
> [18:56:19][E][INFO:1151]: Prompt: May Exit Now  01
> [18:56:19][E][INFO:1152]: Beeps: 0
> 
> [18:56:19][D][test:1273]: armed status/system,stay,away flag is: 1 , 0, 1 , 0
> [18:56:19][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.17 s).
> [18:56:19][W][component:215]: Components should block for at most 20-30ms.
> [18:56:19][E][CMD:802]: 2023-11-15 18:56  F2 12 06 00 00 00 00 64 6C 02 45 6C F5 
> [18:56:20][E][CMD:802]: 2023-11-15 18:56  F7 00 00 0F 10 08 00 8C 08 02 00 00 41 
> [18:56:20][E][INFO:1123]: Display to partition: 01
> [18:56:20][D][text_sensor:064]: 'Line1': Sending state 'ARMED ***STAY***'
> [18:56:20][D][text_sensor:064]: 'Line2': Sending state '                '
> [18:56:20][D][text_sensor:064]: 'Beeps': Sending state '0'
> [18:56:20][E][INFO:1150]: Prompt: ARMED ***STAY*** 
> [18:56:20][E][INFO:1151]: Prompt:                 
> [18:56:20][E][INFO:1152]: Beeps: 0
> 
> [18:56:20][D][test:1273]: armed status/system,stay,away flag is: 1 , 1, 1 , 0
> [18:56:20][D][text_sensor:064]: 'System Status': Sending state 'armed_home'
**> [18:56:20][D][binary_sensor:036]: 'Stay': Sending state ON
> [18:56:20][D][binary_sensor:036]: 'Armed': Sending state ON**
> [18:56:20][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.20 s).
> [18:56:20][W][component:215]: Components should block for at most 20-30ms.
> [18:56:20][E][CMD:802]: 2023-11-15 18:56  F2 12 06 00 00 00 00 65 6C 02 45 6C F5 
> [18:56:20][E][CMD:802]: 2023-11-15 18:56  F2 16 06 00 00 00 00 66 63 02 45 43 F5 
> [18:56:20][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.06 s).
> [18:56:20][W][component:215]: Components should block for at most 20-30ms.
> [18:56:20][E][CMD:802]: 2023-11-15 18:56  F2 16 06 00 00 00 00 67 63 02 45 43 F5 
> [18:56:20][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.05 s).
> [18:56:20][W][component:215]: Components should block for at most 20-30ms.
> [18:56:29][E][CMD:802]: 2023-11-15 18:56  F7 00 00 0F 10 08 00 8C 08 02 00 00 41 
> [18:56:29][E][INFO:1123]: Display to partition: 01
> [18:56:29][E][INFO:1150]: Prompt: ARMED ***STAY*** 
> [18:56:29][E][INFO:1151]: Prompt:                 
> [18:56:29][E][INFO:1152]: Beeps: 0
> 
> [18:56:29][D][test:1273]: armed status/system,stay,away flag is: 1 , 1, 1 , 0
> [18:56:29][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.14 s).
> [18:56:29][W][component:215]: Components should block for at most 20-30ms.
> [18:56:33][E][CMD:802]: 2023-11-15 18:56  FB 02 20 81 62 00 00 00 00 00 00 00 00 
> [18:56:33][E][EXT:802]: 2023-11-15 18:56  00 24 00 DC 00 00 00 00 00 00 00 00 74 
> [18:56:33][E][CMD:802]: 2023-11-15 18:56  FB 02 25 60 7E 00 00 00 00 00 00 00 00 
> [18:56:34][E][EXT:802]: 2023-11-15 18:56  00 21 00 DF 00 00 00 00 00 00 00 00 74 
> [18:56:36][E][CMD:802]: 2023-11-15 18:56  F9 83 02 53 2F 00 00 00 00 00 00 00 00 
> [18:56:36][D][test:1273]: armed status/system,stay,away flag is: 1 , 1, 1 , 0
> [18:56:36][D][text_sensor:064]: 'System Status': Sending state 'not_ready'
**> [18:56:36][D][binary_sensor:036]: 'Stay': Sending state OFF
> [18:56:36][D][binary_sensor:036]: 'Armed': Sending state OFF**
> [18:56:36][D][text_sensor:064]: 'Lrr Msg': Sending state ''
> [18:56:37][E][CMD:802]: 2023-11-15 18:56  78 C3 00 00 00 00 00 00 00 00 00 00 00 
> [18:56:39][E][CMD:802]: 2023-11-15 18:56  F7 00 00 0F 10 08 00 8C 08 02 00 00 41 
> [18:56:39][E][INFO:1123]: Display to partition: 01
> [18:56:39][E][INFO:1150]: Prompt: ARMED ***STAY*** 
> [18:56:39][E][INFO:1151]: Prompt:                 
> [18:56:39][E][INFO:1152]: Beeps: 0
> 
> [18:56:39][D][test:1273]: armed status/system,stay,away flag is: 1 , 1, 1 , 0
> [18:56:39][D][text_sensor:064]: 'System Status': Sending state 'armed_home'
**> [18:56:39][D][binary_sensor:036]: 'Stay': Sending state ON
> [18:56:39][D][binary_sensor:036]: 'Armed': Sending state ON**
[... 91 lines cut ...]
> [18:57:41][D][binary_sensor:036]: 'Armed': Sending state OFF
> [18:57:41][E][CMD:802]: 2023-11-15 18:57  78 43 00 00 00 00 00 00 00 00 00 00 00 
> [18:57:44][E][CMD:802]: 2023-11-15 18:57  FA 01 02 25 F7 E7 00 00 00 00 00 00 00 
> [18:57:45][E][EXT:802]: 2023-11-15 18:57  FA 07 F7 00 03 00 00 00 00 00 00 00 00 
> [18:57:49][E][CMD:802]: 2023-11-15 18:57  F7 00 00 0F 10 08 00 8C 08 02 00 00 41 
> [18:57:49][E][INFO:1123]: Display to partition: 01
> [18:57:49][D][text_sensor:064]: 'Line1': Sending state 'ARMED ***STAY***'
> [18:57:49][D][text_sensor:064]: 'Line2': Sending state '                '
[... cut ...]
> [18:57:49][D][text_sensor:064]: 'Zone Status': Sending state 'OP:15,OP:16'
> [18:57:49][W][component:214]: Component vista_alarm_panel took a long time for an operation (0.21 s).
> [18:57:49][W][component:215]: Components should block for at most 20-30ms.
> [18:57:59][E][CMD:802]: 2023-11-15 18:57  F7 00 00 0F 10 08 00 8C 08 02 00 00 41 
> [18:57:59][E][INFO:1123]: Display to partition: 01
> [18:57:59][E][INFO:1150]: Prompt: ARMED ***STAY*** 
> [18:57:59][E][INFO:1151]: Prompt:                 
> [18:57:59][E][INFO:1152]: Beeps: 0

Dilbert66, 2023-11-16:
That's my fault I was testing some code and pushed to main instead of dev. I've pushed updated code that should fix that

AlexandreUSA, 2023-11-16:
Hi, all good now. Thank you! Quick question unrelated to my issue above: is there a reason why the messages like the ones below are now showing in red "Error"? They used to be just information if I recall correctly. It does not really matter, just wanted to know. 

> [18:57:41][E][CMD:802]: 2023-11-15 18:57 78 43 00 00 00 00 00 00 00 00 00 00 00
> [18:57:44][E][CMD:802]: 2023-11-15 18:57 FA 01 02 25 F7 E7 00 00 00 00 00 00 00
> [18:57:45][E][EXT:802]: 2023-11-15 [...]

Dilbert66, 2023-11-16:
That's just esphome displaying any esp_log print message that way. They differentiate from script generated log messages to esphome generated ones.

AlexandreUSA, 2023-11-16:
I recompiled again and all is good. Thank you!

Another question came to me while looking at the log about [11:01:46][D][text_sensor:064]: 'Zone Status': Sending state 'OP:15,OP:16'
These two zones are not used and disabled in my Vista21ip panel. I have the first 8 zones plus 8 more from an extender, plus a few more RF. Is there a way to disable these two zones in VistaAlarm as they are not used? I assume it does not hurt to keep them there either...

> [11:01:36][D][text_sensor:064]: 'Lrr Msg': Sending state ''
> [11:01:46][E][CMD:796]: 2023-11-16 11:01  F7 00 00 0F 10 08 00 1C 08 02 00 00 2A 
> [11:01:46][E][INFO:1111]: Display to partition: 01
> [11:01:46][D][text_sensor:064]: 'Line1': Sending state '****DISARMED****'
> [11:01:46][D][text_sensor:064]: 'Line2': Sending state '  Ready to Arm  '
> [11:01:46][D][text_sensor:064]: 'Beeps': Sending state '0'
> [11:01:46][E][INFO:1138]: Prompt: ****DISARMED**** 
> [11:01:46][E][INFO:1139]: Prompt:   Ready to Arm  
> [11:01:46][E][INFO:1140]: Beeps: 0
> 
> [11:01:46][D][text_sensor:064]: 'System Status': Sending state 'disarmed'
> **[11:01:46][D][text_sensor:064]: 'Zone Status': Sending state 'OP:15,OP:16'**
> [11:01:46][D][text_sensor:064]: 'RF Msg': Sending state ''
> [11:01:49][E][CMD:796]: 2023-11-16 11:01  FB 02 25 81 5D 00 00 00 00 00 00 00 00 
> [11:01:49][E][EXT:796]: 2023-11-16 11:01  00 21 00 DF 00 00 00 00 00 00 00 00 74 
> [11:01:50][E][CMD:796]: 2023-11-16 11:01  FB 02 20 60 83 00 00 00 00 00 00 00 00 
> [11:01:50][E][EXT:796]: 2023-11-16 11:01  00 24 00 DC 00 00 00 00 00 00 00 00 74 
> [11:01:53][E][CMD:796]: 2023-11-16 11:01  F9 43 02 53 6F 00 00 00 00 00 00 00 00 
> [11:01:53][E][CMD:796]: 2023-11-16 11:01  78 83 00 00 00 00 00 00 00 00 00 00 00 
> [11:01:56][E][CMD:796]: 2023-11-16 11:01  F7 00 00 0F 10 08 00 1C 08 02 00 00 2A 
> [11:01:56][E][INFO:1111]: Display to partition: 01
> [11:01:56][E][INFO:1138]: Prompt: ****DISARMED**** 
> [11:01:56][E][INFO:1139]: Prompt:   Ready to Arm  
> [11:01:56][E][INFO:1140]: Beeps: 0

Dilbert66, 2023-11-16:
I assume it's picking them up in the rf messages.  No worries, I've added a check to see if the zones are in defined as binary or text sensors and if not, do not display them.  It's in the latest code now.

AlexandreUSA, 2023-11-16:
Everything working perfectly. Thanks again.

closed 2023-11-16 by AlexandreUSA