Dilbert66/esphome-vistaECP #120: Motion sensors only trigger for a few seconds despite sustained motion

johnboiles, 2024-01-02:
After opening #119 I figured out how to get my motion sensors configured via the `rfseriallookup` field and they are reporting their state into HA 🎉 

But I'm noticing that when they trigger, they only stay in the 'Detected Motion' state for a few seconds, even if I keep jumping up and down in front of the sensor: 
[image]

If I walk away for a moment, then come back, I am able to get the sensor to trigger again for 3s. Is there any way to have the sensors only clear when they stop detecting motion? Is this an inherent limit of these sensors?

Dilbert66, 2024-01-03:
If your motion sensors are assigned to zones, you would need to setup some zone sensors in your config yaml for them to be reported normally.  You would see FAULT xx when those zones are triggered.   The rfseriallookup is normally there to capture close events quickly instead of waiting for the fault to expire as for other zones.  Also the refserialllookup will capture movement even when the system is armed.  As far as the trigger issue, that's not an issue with the code. That's just how your motions sensors work. I suggest you assign  zones in the yaml as binary sensors .

johnboiles, 2024-01-03:
> Also the rfseriallookup will capture movement even when the system is armed.

So normally when the system is not armed the system doesn't capture motion events? So motion detectors are handled differently than glass break sensors and door sensors? Those triggered just fine when disarmed, only using the zone sensors.

Dilbert66, 2024-11-01:
I realized i had not answered your question here.  Yes, correct when the system is armed, the Vista panel does not send out fault warnings on zones such as internal zones when armed stay.   The workaround that I've implemented uses the FB commands from the RF base to the panel to determine zone activity directly without needing the panel.  The only complication is that the RF base only sends out serial number to the panel so we need to do the serial number to zone mapping ourselves.   The other method I use is to emulate an AUI keypad to determine zone/bypass actity.  All this is on the "dev" branch.

closed 2024-11-01 by Dilbert66