Dilbert66/esphome-vistaECP #122: LRR Supervisor

rwoldberg, 2024-01-21:
I have Vista20P system and I had an envisalink 4 installed and so I had the LRR supervisor turned on in the panel config field *29.  I removed the envisalink and installed an esp32 with this component and had the lrrsupervisor flag set to true.  When I reset the vista with the esp32 attached, everything seems to work but I get a trouble code, with the LRR supervisor.  If I clear the*29 field and reset then the system appears to work and the esp32 appears to be working.   Is this expected behavior.  Am I missing updates from the panel with LRR supervisor turned off on the vista panel?

Dilbert66, 2024-01-23:
Without knowing what trouble code you are getting, i'm not sure what the issue with be.  Two things could be happening.  First, LRR only works if you have only 1 long rang device on the bus.  Either a real one or a virtual one.  If you have both, you will get an error.  The other issue could be LRR is on but no device is responding so the panel things it's defective.  If you have lrr to true in the yaml and it's still not being seen, it could be that your esp is not sending data back to the panel.   Are you able to send commands otherwise?   

Lrr is not really needed anyhow as it's only purpose is to provide extra  information in the lrr field.  If you do not use that info, then there is no point in having it on anyhow. Some of the info is  duplication of information available in other fields.
I would just leave it off.

rwoldberg, 2024-01-23:
The trouble code is for the LRR supervisor.  I belive the envisalink 4 emulated this as well, that module is now removed so the only one should be the emulated one from the esphome module.  I'll have to test that I can send commands.

closed 2024-04-11 by rwoldberg