Dilbert66/esphome-vistaECP #126: component vista_alarm_panel took a long time for operation (123ms)

jacmont, 2024-04-02:
Hello, thank's for the great project!

I am using ESP8266 on breadboard hooked to a Ademco3000/Vista20 security panel!
The ESP8266 circuitry i am using works with 3 optocouplers 4N35
Everything seem to work OK but I having warning from the device about long time operation on the component Vista_alarm_panel

I did copy them locally instead of using the your github versions of the files,
but I keep having ther same warning as the log file show!

And ESPHOME is running on Home Assistant mounted on a VM over Proxmox!

any ideas where to look ?

[12:19:57][W][component:232]: Component vista_alarm_panel took a long time for an operation (129 ms).
[12:19:57][W][component:233]: Components should block for at most 30 ms.
[12:20:06][E][CMD:843]: 2024-04-02 12:20  F7 00 00 03 10 08 00 1C 08 02 00 00 20 
[12:20:06][E][INFO:1158]: Display to partition: 01
[12:20:06][E][INFO:1185]: Prompt:   **** ADT ****  
[12:20:06][E][INFO:1186]: Prompt:   Ready to Arm  
[12:20:06][E][INFO:1187]: Beeps: 0

[12:20:06][W][component:232]: Component vista_alarm_panel took a long time for an operation (126 ms).
[12:20:06][W][component:233]: Components should block for at most 30 ms.
[12:20:16][E][CMD:843]: 2024-04-02 12:20  F7 00 00 03 10 08 00 1C 08 02 00 00 20 
[12:20:16][E][INFO:1158]: Display to partition: 01
[12:20:16][D][text_sensor:064]: 'Line1 (ln1_1)': Sending state '  **** ADT **** '
[12:20:16][D][text_sensor:064]: 'Line2 (ln2_1)': Sending state '  Ready to Arm  '
[12:20:16][D][text_sensor:064]: 'Beeps (bp_1)': Sending state '0'
[12:20:16][E][INFO:1185]: Prompt:   **** ADT ****  
[12:20:16][E][INFO:1186]: Prompt:   Ready to Arm  
[12:20:16][E][INFO:1187]: Beeps: 0

[12:20:16][D][text_sensor:064]: 'System Status (ss_1)': Sending state 'disarmed'
[12:20:16][D][text_sensor:064]: 'Zone Status (zs)': Sending state ''
[12:20:16][W][component:232]: Component vista_alarm_panel took a long time for an operation (178 ms).
[12:20:16][W][component:233]: Components should block for at most 30 ms.

Dilbert66, 2024-04-02:
That warning is not important. Esphome added this last year but it's totally useless and just a source of annoyance. It's just saying that the component is taking longer than what it wants.  Since this component is not really sharing resources with too many others, its not causing any issues.  The only way to not see it as by setting the log level to "ERROR" in the yaml.  A few people are asking the dev's to allow this to be controllable.

jacmont, 2024-04-02:
Thank you for your fast response!

closed 2024-04-02 by jacmont