Dilbert66/esphome-vistaECP #130: Compile error with Dev branch

bmwhitetx, 2024-05-17:
I see in the discussion that you changed the dev branch to now clear faults quicker instead of waiting for the TTL. So I decided to update to the Dev branch. To do that I duplicated the github source statement in the yaml and changed master to dev, like so:

```
external_components:
  #- source: github://Dilbert66/esphome-components@main
  - source: github://Dilbert66/esphome-components@dev
```
I then went to the ESPHome tab in HA, clicked the three dots and chose Install. It failed with the following log:
```
INFO ESPHome 2023.12.8
INFO Reading configuration /config/esphome/vistaalarm.yaml...
INFO Detected timezone 'America/Chicago'
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1028, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 407, in command_compile
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 192, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 204, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 679, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 184, in wrapped
    await coro(conf)
  File "/data/external_components/c9cfb228/components/vista_alarm_panel/__init__.py", line 153, in to_code
    variant = get_esp32_variant()
              ^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/components/esp32/__init__.py", line 88, in get_esp32_variant
    return (core_obj or CORE).data[KEY_ESP32][KEY_VARIANT]
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'esp32'

```
If I uncomment back to the master branch, no errors in the above process.
Maybe I'm not changing over the the Dev branch correctly?

I am using a Wemos D1 ESP8266.
```
#esp32: 
#  board: nodemcu-32s
#  framework: 
#    type: arduino
#    version: recommended

esp8266:    
  board: d1_mini
  framework:
    version: recommended

```
 I have updated HA to the most current versions of Core, System, and ESPHome.

Dilbert66, 2024-05-17:
Ah ok. It's a bug in the init. I forgot to add a check for an esp8266. I'll correct and update. I normally use an esp32

bmwhitetx, 2024-05-17:
Thank you. I have already built a custom board using schematic 1 around the Wemos D1. But I see they make a ESP32 that is pin compatible. Any advantage to moving to that for this project? The setup has been rock solid since install last November.

Dilbert66, 2024-05-17:
Ok, I've pushed a corrected file that compiles now.  The ESP32 has much more available ram so can run more complex firmware.  The code on dev branch will compile fine but you might hit some ram issues if you have a lot zones on your system.   If it works fine, then leave it alone. I don't normally test on the esp8266 anymore so can't comment on how well the dev code will work on it. I do try to make it compatible though.

bmwhitetx, 2024-05-18:
Update. The dev branch did compile with some notes. I was unable to get it to update OTA even with Connection turned off (saw the Readme on that issue). Tried many times and the best I could get was about 30% progress. So I uploaded it via USB. However, I could not get any response from the board in HA. I have 8 hardwired zones and nothing else.

So I jumpered in a ESP32 I had, switched boards in the yaml, uploaded dev and it worked straight away. It will even update OTA with no issue. Liking how fast it was to update OTA, I got a D1 Mini ESP32 overnight from Amazon which is pin-out compatible with the D1 Mini ESP8266 my custom PCB was based on. It's working great with that ESP32 board.

I have attached the log file of the ESP8266 compile on the dev branch if you're interested (warnings at line 108?). But I'm happy running the ESP32 because of the flawless OTA. Thanks for your support of this project!!
[logs_vistaalarm_run.txt](https://github.com/Dilbert66/esphome-vistaECP/files/15366867/logs_vistaalarm_run.txt)

Dilbert66, 2024-05-18:
Thank you for the update! I'll wire up my esp8266 and do some testing to see what the issue is and post back.

Dilbert66, 2024-10-30:
I"ve made some updates to the esp8266 so it should compile but yes, the esp32 is the platform I recommend.

closed 2024-11-01 by Dilbert66