Dilbert66/esphome-vistaECP #135: Compilation Errors After VistaAlarm.yaml Changes

kschlichter, 2024-10-14:
I'm trying to compile the dev branch for use on an ESP8266 with [ESP Home docker 2024.9.2](https://hub.docker.com/r/esphome/esphome/tags?name=2024.9.2) and I'm getting some compile errors.  It's possible I've caused these errors with the changes I made in `VistaAlarm.yaml` [issue/134](https://github.com/Dilbert66/esphome-vistaECP/issues/134) but another user seems to be hitting the same error in the [Home Assistant forum](https://community.home-assistant.io/t/ademco-honeywell-vista-esphome-custom-component-with-an-esp32-esp8266/230537/176) so maybe it's a separate issue.

I've attached the logs from my install attempt and I'm happy to share my VistaAlarm.yaml, if that's helpful.  
[logs_VistaAlarm_compile.txt](https://github.com/user-attachments/files/17357658/logs_VistaAlarm_compile.txt)

This is where it starts to go wrong:
```
In file included from src/esphome/core/log.h:8,
                 from src/esphome/components/sensor/sensor.h:3,
                 from src/esphome/core/application.h:16,
                 from src/esphome/components/vista_alarm_panel/vistaalarm.h:5,
                 from src/esphome/components/vista_alarm_panel/vistaalarm.cpp:2:
src/esphome/components/vista_alarm_panel/vistaalarm.cpp: In member function 'esphome::alarm_panel::vistaECPHome::zoneType* esphome::alarm_panel::vistaECPHome::createZone(uint16_t)':
src/esphome/components/vista_alarm_panel/vistaalarm.cpp:213:19: error: initializer fails to determine size of '__pstr__'
```

Dilbert66, 2024-10-14:
The esp8266 does not handle PSTR's like the esp32. I've corrected the code and pushed to dev. You can try recompiling now.

kschlichter, 2024-10-14:
That compiles for me now; thank you!

closed 2024-10-14 by Dilbert66