Dilbert66/esphome-vistaECP #136: Trouble Status Binary Sensor Not Working

XD36JD2015, 2024-10-15:
My alarm panel currently has 2 trouble faults, and I noticed that the trouble binary sensor doesn't indicate that the alarm panel has an issue.

[alarm-system.yaml.txt](https://github.com/user-attachments/files/17370647/alarm-system.yaml.txt)
[logs_alarm-system_logs(2).txt](https://github.com/user-attachments/files/17370648/logs_alarm-system_logs.2.txt)

Let me know if you need any more info from me, or need me to test changes.

Dilbert66, 2024-10-15:
The trouble binary sensor only activates currently ac and battery issues at this time.  For zone statuses, you will see the zone sensor be on for the zone and you will see the ck:<zone> fiels in the zone status (zs) field showing in HA. For example since you have 2 co2 detectors, they are currently showing trouble so you should see ck:15,ck:16 in the zone status field as well as showing zone 15 and zone 16 as on.

Are you saying you would also like to see the trouble sensor show a fault when individual zones are in check mode?

XD36JD2015, 2024-10-15:
Hi, I think it would be a nice feature to have a sensor with a trouble fault to either trigger the fault sensor, or perhaps the check sensor that  I remember seeing a while ago.  The two CO detectors are having issues, they have reached end of life. triggering the check status.  Thanks for your help!

Dilbert66, 2024-10-15:
Are you currently seeing the ck:15 and ck:16 in the zone status field on home assistant? You should be.  You should also be seeing the fault sensor being shown on for those two zones? If not then I also need to look at that.  I've enabled showing the trouble sensor for zone issues and pushed to dev.

Dilbert66, 2024-10-28:
Change the "return vista.keybusConnected" line to:
```
     return $panelId->connected();
```

XD36JD2015, 2024-10-28:
I fixed the .yaml error, still getting this:

[image]

Dilbert66, 2024-10-28:
You might want to check that you are compiling the dev version as well as that has a lot of fixes:
```
external_components:
  - source: github://Dilbert66/esphome-components@dev 
```

XD36JD2015, 2024-10-28:
I should be pulling from that repo:

[image]

Dilbert66, 2024-10-28:
Arghhhh. You are doing it right... My fault. I just check that I had not pushed an old fix to git so the errored version was still there.  I'll push a fix in couple minutes.  Sorry about that.

XD36JD2015, 2024-10-28:
No worries.  I recompiled using the newest version of what you have uploaded.  The trouble binary is still showing false, and the armed away and armed stay binaries are still showing false when the system is armed and the trouble faults are active.  I've also noticed now that when I arm my panel, I am getting two "armed stay" messages from my keypads.

[image]
[image]
[logs_alarm-system_logs(3).txt](https://github.com/user-attachments/files/17549497/logs_alarm-system_logs.3.txt)

Dilbert66, 2024-10-28:
What panel do you have? I don't have the same hardware available so never tested issues with CO sensors.  I see the issue as to why the armed sensors are not showing correctly as the flags are slightly different for those. I'll see if I can amend the code to support your version.

XD36JD2015, 2024-10-28:
Hi, I've got a Vista 20p.

Dilbert66, 2024-10-28:
Must be a later revision than mine. I have an old ADT panel so pretty old.  Your panel behaviour is slightly different then mine and others i've seen in how it sends out status.  Namely, it's missing a system flag that I was dependent on. I'll tweak the handling a bit and see what I can come up with.

Dilbert66, 2024-10-29:
Can you provide a log sample of  open/faulted zones as well. I need to see what flags it sends for those. Tks!

XD36JD2015, 2024-10-29:
Just installed the latest version, both the trouble binary and the armed stay and away binaries seem to be working now.  Thanks for your work on this!  I have wireless smoke/heat/freeze sensors, CO sensors, glass break sensors, flood sensors, door sensors and wireless windows sensors if you ever need some log data or need to try something out.

Dilbert66, 2024-10-29:
Cool. Tks for the offer!   I just updated one setting from zone fault capture that should address an issue I found with your version.

Dilbert66, 2024-10-30:
Actually, there is one message I would like to see.  A disarmed or armed status message that has no check messages associated. Just a normal status.  I need to see what the flags are for no check or fault statuses.

XD36JD2015, 2024-10-30:
I captured to unarmed/armed events for you with the trouble cleared.
[logs_alarm-system_logs(2).txt](https://github.com/user-attachments/files/17579802/logs_alarm-system_logs.2.txt)

Dilbert66, 2024-10-30:
You might want to remove your access codes from the logs. FYI

XD36JD2015, 2024-10-31:
Good point, I missed that it was in the logs.

Dilbert66, 2024-10-31:
Also remove the following "F6" cmds as the keys will show up in the data

Dilbert66, 2024-10-31:
You can remove the files from this issue. I have them downloaded.

XD36JD2015, 2024-10-31:
I was looking at my armed binary sensor (arm_1), and noticed some weird behavior when the trouble alert is active.  Doesn't seem to do it when I clear the trouble zone.
[image]

Dilbert66, 2024-10-31:
I've made a couple tweaks that should address your issue as well as some others.  Your panel is not behaving different than others. It's just that I've never seen CO sensors data with error faults and those are handled different by the panel. I've adjusted the code to handle those instances better without conflicting with the other logic.

XD36JD2015, 2024-10-31:
I'll give it a go and let you know how it goes.