Dilbert66/esphome-vistaECP #142: Is my AUI config working correctly?

hacsjalano, 2024-12-21:
I've added some logs from when my alarm system enters Arm Stay.  I then walked in front of my zone 3 wired motion detector a few times.  I see some AUI messages in the logs (F2 messages) but it does not change the status of z3 in Home Assistant.

I've cut off the first part of the logs which has a whole ton of junk including the countdown to Arm Stay.  I can add that back in if it's important.
[logs.txt](https://github.com/user-attachments/files/18219981/logs.txt)

hacsjalano, 2024-12-22:
Switched aui address to 1
[logs_aui1.txt](https://github.com/user-attachments/files/18220777/logs_aui1.txt)

hacsjalano, 2024-12-22:
Switched aui address to 2
[logs_aui2.txt](https://github.com/user-attachments/files/18222349/logs_aui2.txt)

Dilbert66, 2026-10-17:
This should now be working with the new update.

closed 2026-10-17 by Dilbert66