Dilbert66/esphome-vistaECP #144: HA-API connection drop

pletch, 2025-01-04:
Hello!  Great project.  I have been experimenting with a setup using an aithinker esp-s3-12k connected to a W5500 ethernet module as I look to move away from use of an ADUSB2.  

I figured since my alarm panel is in close proximity to my network panel that I would use this project as an opportunity to dig into use of ethernet with an ESP32 microcontroller.  In order to achieve stability, I had to move from initially using the simple non-isolated circuit to the more complex isolated-ground design shared by others and the board has been running for days without issue since doing so.  Prior to the change, i would sporadically experience spontaneous power-on reset events or ethernet module dropouts which were not eliminated through use of a separate linear voltage regulator for the W5500. I didn't investigate extensively but speculate that the ethernet module may be sensitive to ground noise???

Last night, I experienced about 10 seconds of an issue in which all Vista related sensors were fluttering between their valid state and unavailable.  The issue inexplicably stopped on its own after this period.  The ESP32 board did not reboot during this time according to the debug uptime sensor.  However, looking at the HA log, there was a protocol error that repeated showing the following:

```
Jan 04 00:02:13 hass hass[57966]:   File "aioesphomeapi/connection.py", line 902, in aioesphomeapi.connection.APIConnection.process_packet
Jan 04 00:02:13 hass hass[57966]: google.protobuf.message.DecodeError: Error parsing message with type 'TextSensorStateResponse'
Jan 04 00:02:13 hass hass[57966]: 2025-01-04 00:02:13.393 WARNING (MainThread) [aioesphomeapi.connection] vista-ecp @ 192.168.11.90: Connection error occurred: Invalid protobuf message: type=TextSensorStateResponse data=b'\r\xef\xf8(\x9a\x12\x10 Ready to Arm  \xfb': Error parsing message with type 'TextSensorStateResponse'
Jan 04 00:02:13 hass hass[57966]: 2025-01-04 00:02:13.396 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Jan 04 00:02:13 hass hass[57966]: Traceback (most recent call last):
Jan 04 00:02:13 hass hass[57966]:   File "/opt/python/3.13.1/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
Jan 04 00:02:13 hass hass[57966]:     self._protocol.data_received(data)
Jan 04 00:02:13 hass hass[57966]:     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
Jan 04 00:02:13 hass hass[57966]:   File "aioesphomeapi/_frame_helper/plain_text.py", line 65, in aioesphomeapi._frame_helper.plain_text.APIPlaintextFrameHelper.data_received
Jan 04 00:02:13 hass hass[57966]:   File "aioesphomeapi/_frame_helper/plain_text.py", line 93, in aioesphomeapi._frame_helper.plain_text.APIPlaintextFrameHelper.data_received
Jan 04 00:02:13 hass hass[57966]:   File "aioesphomeapi/connection.py", line 928, in aioesphomeapi.connection.APIConnection.process_packet
Jan 04 00:02:13 hass hass[57966]:   File "aioesphomeapi/connection.py", line 902, in aioesphomeapi.connection.APIConnection.process_packet
Jan 04 00:02:13 hass hass[57966]: google.protobuf.message.DecodeError: Error parsing message with type 'TextSensorStateResponse'
```

Note the extra left parenthesis in the referenced message as this seems potentially out of place:  
data=b'\r\xef\xf8(\x9a\x12\x10 Ready to Arm  \xfb'

I am running the latest dev branch of esphome-VistaECP and this occurred on HA 2024.12.4 core.  

Any idea on what might be going on here?  I want the connection to HA to be highly stable to ensure automations work as expected etc.  Thanks!
[systemd-log-ha.txt](https://github.com/user-attachments/files/18307094/systemd-log-ha.txt)
[vista-ecp.yaml.txt](https://github.com/user-attachments/files/18307117/vista-ecp.yaml.txt)

Dilbert66, 2025-01-04:
It looks to me like you have some noise somewhere. EIther the connection to the panel or to the api. Hard to know for sure as I've never seen the issue.  Truly though to know for sure I would need to see the Esphome logs as those would be a better indicator of any problem.

pletch, 2025-01-04:
Thanks for your thoughts!

Here is about 5 minutes of logging from esphome.  

When you say api socket connection noise, what does that mean exactly?  Errors on the network connection which could really only originate from the W5500?
[esphome-vista.log](https://github.com/user-attachments/files/18307442/esphome-vista.log)

Thanks!

closed 2025-01-20 by pletch