Dilbert66/esphome-vistaECP #148: Battery Status per Zone

nitej, 2025-01-23:
Is it possible to display the individual low battery status of wireless sensors in addition to the overall battery status of the system? This info is already available in the wireless “9E” messages. I didn’t see any options in the example yaml config for setting a zone sensor for battery display, and the text zone sensors do not indicate low wireless battery.

Dilbert66, 2025-01-23:
The zone low battery levels of RF zones are shown in the "zone status (zs)" field as LB:xx where xx is the zone number. In order for this to work, you need to enter the zones and associated loop and serial number data in the "rfseriallookup" yaml field .  This is needed since the FB data does not send zone info, only serial data so we need to map this info so it can be captured.  The LB:xx will only show when an RF zone sends a low battery flag for that serial.

nitej, 2025-01-24:
I was able to get it working (I was used the outdated format for the rfseriallookup parameter in the yaml file). However, I notice that the low battery signal only shows up when the zone isn’t faulted. Would it be possible to extract the battery state using the rf raw data since that data is sent on every rf signal from a sensor?

Dilbert66, 2025-01-24:
The code already grabs the status from the raw data.  If it's not working on your system, you'll need to post logs so I can see what's happening on your system because I don't have rf devices to test with on my end.