Dilbert66/esphome-vistaECP #150: LRR Issues

XD36JD2015, 2025-01-31:
The LRR functionality seems to not be working correctly on my setup.  The HomeAssistant sensor for LRR message stays at "ESP Restart", it never shows any LRR messages like my other system shows.  Also, with the report codes turned on for this alarm, when transitioning from armed to disarmed and vice-versa, all of the sensors go unavailable for a little bit, and the logs indicate that the connection dropped.  Other than that, the commands take, and the alarm does arm and disarm.  This is on a Vista 15p system and an ESP8266.

[logs_alarm-system-shop_run.txt](https://github.com/user-attachments/files/18611083/logs_alarm-system-shop_run.txt)

[alarm-system-shop(3).txt](https://github.com/user-attachments/files/18611091/alarm-system-shop.3.txt)

Dilbert66, 2025-01-31:
For the LRR to work, you panel has to be programmed to send  events.  I don't see any such events sent by this panel. 
As to the reboot, I can't say what it is. I would need to se the direct serial logs from the usb port and not the esphome web logs for that.  Those would indicate the crash details.

XD36JD2015, 2025-01-31:
I have the panel setup to send events, the settings match my Vista 20p panel (using the the same version of your firmware on an esp32), where the LRR functionality does work.  I'll get a serial capture this weekend.  Also wasn't sure if these issues had anything to do with the compile warnings at the beginning of the log.

XD36JD2015, 2025-01-31:
I managed to get some logs for you from the USB port.  It looks like the esp8266 is crashing.

[alarm-system-shop_logs (1).txt](https://github.com/user-attachments/files/18612361/alarm-system-shop_logs.1.txt)
[alarm-system-shop_logs (2).txt](https://github.com/user-attachments/files/18612360/alarm-system-shop_logs.2.txt)
[alarm-system-shop_logs (3).txt](https://github.com/user-attachments/files/18612362/alarm-system-shop_logs.3.txt)

Dilbert66, 2025-01-31:
Sorry, I can't read those logs.  They are supposed to be ascii text from the serial port.

XD36JD2015, 2025-01-31:
Ok, let me see what I can do.  I am seeing some F9 messages when I watch the log.  F9 is LRR right?

Dilbert66, 2025-01-31:
Yes, f9 is lrr but the messages that carry the needed info are type 0x58.  The ones in your log are 0x53 which are just normal polls but don't provide anything useful.

Dilbert66, 2025-02-02:
Ok, I fired up my esp8266 and can see the issue. It's related to some new code I added.  It works perfectly fine on my test esp32 platforms but the esp8266 is crashing with it. That explains why you are not seeing the f9's as that is what is causing it to crash.  I will fix it and push an update tomorow.  FYI, I also recommend you move up to an esp32.  Faster and much more reliable!

XD36JD2015, 2025-02-02:
Thanks for fixing the issue!  I do appreciate the effort you put into this project.  I do plan on upgrading to the ESP32 for this alarm panel at some point.  Just have had time to put a new board together yet.

Dilbert66, 2025-02-02:
No worries! Glad it works!

closed 2025-02-02 by Dilbert66