Dilbert66/esphome-vistaECP #152: Error update end ESP32-C6-zero

LalleNyberg, 2025-02-18:
Hi!

When I'm trying to install the project on my [Waveshare ESP32-C6-zero](https://www.waveshare.com/wiki/ESP32-C6-Zero) everything seems ok until the end of the update, then I get this error message: 
"INFO Upload took 4.29 seconds, waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information."

I've got no clue about what's wrong. But I have some questions in my head:
I changed the name of the vistaalarm.yaml, is that a problem?
I changed the three substitutions for name, friendlyName and panelID, can that disturb anything?
I commented away the accesscode row for now because I didn't want to put my code in to the secrets, at least not in the beginning, is that a proper way to turn that function off?
I commented away the autopoulate row because i got this error message: " [autopopulate] is an invalid option for [vista_alarm_panel]. Please check the indentation."
I choosed to set the board to esp32dev, is there any other option that is better? I used this on my standard setup to get the unit running on my wifi but that board doesn't work in this .yaml: "board: esp32-c6-devkitc-1
  variant: esp32c6"
Is the mayor problem that I bought an ESP32-C6 instead of an ESP32-C3? Thinking of the fact that i'ts not fully supported by esphome yet. If so, is it possible to come around it with some modifications?

I'll attatch the log from the install attempt.

[logs_ademcoecp_run.txt](https://github.com/user-attachments/files/18840683/logs_ademcoecp_run.txt)

LalleNyberg, 2025-02-18:
I did the installation over wifi. Is there a usb serial log anyway? Where can I find it?
If I choose to buy another esp32, like a C3 or so, can I buy any brand or is some brands/models tested out and preferred? I'm not so experienced with the esp's so I want to keep it simple to see if I can get the communications right with my vista 4140xmpt2 and 6139 keypads.

Dilbert66, 2025-02-19:
To log from the serial console, you need to have the esp connected to a pc via usb and either run a serial monitor to capture the usb output or run go to this url using chrome on the connected pc and connect to the device that way. You can then select the logs.  You can restart the device and capture the output that way.  I have a couple c6's on order but not sure when I'll get them to try out with.  Any other older esp32 such as c3's, s3's ,wroom, etc will work.  No specific brands really.

https://web.esphome.io/?dashboard_install

pletch, 2025-02-21:
Esphome arduino framework is based on ESP-IDF 4x and I believe it doesn't support the newer C6 variant.

If planning to use with EspHome API for connecting to Home Assistant, I have generated a fork of this project that uses ESP-IDF framework rather than Arduino.  Most functionality is implemented and it seems to work well enough though there are some distinct differences in the implementation and configuration.  The ESP32-C6 has two UARTs on it so it should work if pins are specified appropriately. 

Feel free to give it a go but look over the example yaml file carefully.

I think you will have to also update to the following in the configuration before compiling:
```
esp32:
  board: esp32-c6-devkitm-1
  variant: esp32c6
```

LalleNyberg, 2025-02-23:
@Dilbert66 Seems like I dont get any more detailed information from the USB log either. My guess is that @pletch is right that the C6 doesn't support arduino framework for EspHome yet. I'll try with the ESP-IDF fork and if I don't get it working properly I guess I'll buy a C3 or S3 and start over that way. (Right now I get some issue with the API or a boot loop when i flash the fork into the C6, but I think I'll discuss it in the fork's issue section) Thank you for your help this far!

Panko74, 2025-03-24:
@LalleNyberg you are right about the C6 version issue: it seems that it is not implemented in EspHome yet.
I bought a C6-Devkit, but having the same issue I had to buy a C3-Devkit, precisely this one:
https://www.amazon.it/dp/B0CB84WRZY
I managed to configure it successfully with the **dev branch** and the following parameters:
`esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino
    version: recommended`

and
`  ##esp32
  rxpin: "5" #GPIO pin to use for data receive (yellow line) 
  txpin: "4" #GPIO pin to use for data transmit (green line)
  monitorpin: "7" #GPIO pin to use for monitoring module traffic such as RF or Expanders . Set to -1 to disable`

I'm using the #1 scheme "Non-isolated simple version" and it's working fine, on a Vista120.

Dilbert66, 2025-11-17:
esp-idf is now supported . You can access it from the "dev" branch.  Use framework: esp-idf
```
source: github://Dilbert66/esphome-components@dev
```

closed 2026-10-17 by Dilbert66