Dilbert66/esphome-vistaECP #156: Vista 4140XMP

infinitytec, 2025-08-08:
Thanks for the project! I think I'm getting at least somewhat close to having it working, but not quite sure yet.

I have a Vista 4140XMP which also seems to be branded as a First Alert 1200C. It looks to have a lot in common with the supported models, but does not, at least at this time, seem to function for me.

I was getting some interesting lines in the logs:

```
[14:30:27][E][CHK:814]: [1969-12-31 21:21] E0 
[14:30:28][E][CHK:814]: [1969-12-31 21:21] 7E FE 
[14:30:28][E][CHK:814]: [1969-12-31 21:21] 9E 
[14:30:34][E][CHK:814]: [1969-12-31 21:21] 60 80 86 06 
[14:30:35][E][CHK:814]: [1969-12-31 21:21] 93 E8 
[14:30:35][E][CHK:814]: [1969-12-31 21:21] 3E 00 00 86 18 
[14:30:35][E][CHK:814]: [1969-12-31 21:21] 80 F8 00 
[14:30:36][E][CHK:814]: [1969-12-31 21:21] FE 
[14:30:36][E][CHK:814]: [1969-12-31 21:21] E0 80 
```

These may have been due to a lose ground wire as after I checked the wiring I started getting messages stating the panel may not be connected.

I guess just looking for some advice and wondering what the best way to proceed to see if this will work will be.

infinitytec, 2025-08-08:
OK, I will take a look!

Dilbert66, 2025-08-08:
No don't worry about that warning. It should be fine.

Dilbert66, 2026-10-17:
I'm pretty sure that panel is too old for the vista20 or se firmware but there is another project that might work for you as it's for the 4120xm

https://github.com/Stef86fal/esphome-ademco31

infinitytec, 2026-10-17:
Thanks, I actually probably should have closed this. I got a Konnected replacement board instead.

closed 2026-10-17 by infinitytec